openscad/openscad #5277: Proposal:  Use names or comments in CSG output

EmperorArthur, 2024-09-01:
# Rationale / Problem to solve
CSG as an intermediary file format looses so much information that it is difficult/impossible to perform some tasks.

# Examples:
* `children()` should be linked objects in FreeCAD, instead of separate objects that are repeated.
  * https://github.com/KeithSloan/OpenSCAD_Alt_Import/issues/29
* FreeCAD importers use generic names for the CSG nodes, instead of module names.
  * https://github.com/KeithSloan/OpenSCAD_Alt_Import/issues/30

# Proposed Solution
The [wiki](https://github.com/openscad/openscad/wiki/CSG-File-Format) seems to suggest that CSG files support named objects.  Therefore parsers **should** support this.  If this is incorrect, then the wiki needs to be clarified.

* When generating a CSG tree, generate a unique object name based on the module, number of times called, and calls to `children()`

### Example
```
module lineup(num, space) {
for (i = [0 : num-1])
translate([ space*i, 0, 0 ]) children();
}
lineup(2, 65){sphere(30);}
lineup(2, 65){sphere(30);}
```
Would produce:
```
lineup_0_children = group() {
				sphere($fn = 0, $fa = 12, $fs = 2, r = 30);
			}
lineup_0 = group() {
		multmatrix([[1, 0, 0, 0], [0, 1, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]]) {
			lineup_0_children()
		}
		multmatrix([[1, 0, 0, 65], [0, 1, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]]) {
			lineup_0_children()
		}
	}
lineup_1_children = group() {
				sphere($fn = 0, $fa = 12, $fs = 2, r = 30);
			}
lineup_1 = group() {
		multmatrix([[1, 0, 0, 0], [0, 1, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]]) {
			lineup_1_children()
		}
		multmatrix([[1, 0, 0, 65], [0, 1, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]]) {
			lineup_1_children()
		}
	}
```

# Alternatives

## CSG Tags
This is not my **preferred** solution, but maintains compatibility with CSG parsers which may not support named objects.

* When generating a CSG tree, add machine readable comments which tags with the module name where possible.
* When generating a CSG tree, add machine readable comments with unique IDs where possible.

### Example
```
module lineup(num, space) {
for (i = [0 : num-1])
translate([ space*i, 0, 0 ]) children();
}
lineup(2, 65){sphere(30);}
lineup(2, 65){sphere(30);}
```
Would produce:
```
group() { //lineup_0
		multmatrix([[1, 0, 0, 0], [0, 1, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]]) {
			group() {  //lineup_0_children
				sphere($fn = 0, $fa = 12, $fs = 2, r = 30);
			}
		}
		multmatrix([[1, 0, 0, 65], [0, 1, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]]) {
			group() {  //lineup_0_children
				sphere($fn = 0, $fa = 12, $fs = 2, r = 30);
			}
		}
	}
group() {  //lineup_1
		multmatrix([[1, 0, 0, 0], [0, 1, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]]) {
			group() {  //lineup_1_children
				sphere($fn = 0, $fa = 12, $fs = 2, r = 30);
			}
		}
		multmatrix([[1, 0, 0, 65], [0, 1, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]]) {
			group() {  //lineup_1_children
				sphere($fn = 0, $fa = 12, $fs = 2, r = 30);
			}
		}
	}
```

## JSON
Re - consider how CSG trees are exported in general.
JSON Reference format is commonly used, and allows for tagging individual items.

t-paul, 2024-09-01:
CSG files are supposed to be a valid subset of *.scad scripts, which makes the proposed solution impossible. While ok from technical side, I'm very much opposed using comments in that way.

Extending CSG is fine but that extension must be valid scad too.

One possible option are annotations or marker nodes (#1748) where I would hugely prefer annotations for multiple reasons discussed before.

KeithSloan, 2024-09-01:
"Extending CSG is fine but that extension must be valid scad too."

How are the comments not valid scad, does scad not support comments

t-paul, 2024-09-01:
Nobody said that is the case. I said the "_proposed solution_" is not possible, the "_alternative solution_" (the one talking about comments) I would strongly argue against while technically it would work.

EmperorArthur, 2024-09-01:
Updated the Proposed Solution, and proposed another alternative which addresses the issue with comments not being part of code, but still works with current OpenSCAD.  Albeit with a warning

This alternative solves the main issue with `marker()` nodes and the main proposal. In that both would be a major file version change instead of a minor increment.

kintel, 2024-09-01:
One of the challenges is also that CSG export is used for other things than FreeCAD import. If we want to support all use-cases, we kind of need to make the output configurable to include only what the downstream target importer expects.
Examples of pain points from the past which could be solved this ways are: hull and minkowski as polyhedrons, import nodes as polyhedron(3D) or polygon(2D).

jordanbrown0, 2024-09-01:
One could imagine having a CSG export emit module definitions that it would then invoke repeatedly.

However, note that a module name is not enough to identify the shape emitted by a module - you also need the arguments and the state of all $variables used by the module, and then you still have rands() and any other hypothetical future functions that can return different values on subsequent calls (e.g. current time).

EmperorArthur, 2024-09-01:
> If we want to support all use-cases, we kind of need to make the output configurable to include only what the downstream target importer expects.

True, this is why my alternative solution only adds two optional variables to the `group()` call.  That is a small enough change that most parsers could handle it right now, or could be considered a requirement for parsers to support the newer version of OpenSCAD.  The same applies to 2nd alternative solution.  Which is using comments.

> However, note that a module name is not enough to identify the shape emitted by a module - you also need the arguments and the state of all $variables used by the module, and then you still have [...]

That's **not** supported by this proposal, and would actually go against the entire point.  The idea of modules within CSG files is:
1. To consolidate **identical** code.
2. To provide a method for naming nodes.

I am editing the proposal to make this more clear.

jordanbrown0, 2024-09-01:
> To consolidate identical code.

Sure, but something has to figure out that it's identical.  Just the fact that it's the result of a module invocation doesn't make it identical to other invocations of that module.  From an "inputs" perspective, two module invocations would need to have matching parameters plus matching $ variables plus no use of rands().  From an "outputs" perspective, OpenSCAD could compare the resulting subtrees to see if they are identical.

OpenSCAD could generate a hash of the module name and arguments and $ variables (handwaving a bit about rands()) and use that as your unique ID.  (But: generating those hashes would add overhead to every OpenSCAD execution, not just the ones that will later be used for CSG export.)

Or at export time OpenSCAD could generate a hash of each subtree, and use that.  But then again, an importer could equally well do that.

Or perhaps a hybrid:  OpenSCAD could tag subtrees with the module name, with the understanding that there's no guarantee that two subtrees with the same module name are identical, and the importer could hash them and detect duplicates.  That gets you labelled nodes for human information, correctly detects duplicate subtrees, and puts the overhead in the export-import cycle where it belongs.

I would not focus on `children()`.  Most modules don't use it, and you would still want to detect identical subtrees in cases where it's not used.  You would, for instance, want to detect identical subtrees that results from `for()` loops, with no modules or `children()` in the picture.

Also note:
* Identical module names do not necessarily mean that they were generated by the same program snippet, because OpenSCAD allows nested modules to have duplicate names.  That's probably a non-issue here, because even if module names are unique you'd have to detect differences as discussed above.
* PR#4478, if it ever gets integrated, adds unnamed modules, directly analogous to the existing unnamed functions.  That too is probably a non-issue, because it could use the empty name and have the same issues as named modules.

It may sound like I'm against achieving this goal.  I'm not.  I'm dubious of it, because there's a ton of semantic information that would be lost and that would limit downstream editing, but if one's willing to accept those limits then more power to them.  But it needs to work correctly in all cases, and it needs to not add non-trivial overhead in cases that aren't using it.

nophead, 2024-09-02:
Why not just import the scad source code instead of the CSG?

EmperorArthur, 2024-09-08:
@jordanbrown0 

>  Just the fact that it's the result of a module invocation doesn't make it identical to other invocations of that module.

Not disagreeing at all.  Would you consider the approach of modifying the variables to add those two **optional** fields to the standard regardless?  Even having the module name can be useful for reading a CSG.  Then while we can debate on the ID field, it would exist in the standard for someone to use.  Even if it's a 3rd party program that does the hashing.

> But: generating those hashes would add overhead to every OpenSCAD execution, not just the ones that will later be used for CSG export.

As always, the first question is how much overhead compared to the benefit.  Both from a calculation perspective, and from a maintenance perspective.  I already mentioned the option of an external program, but that would require a standards change without going the comment approach.  One benefit of doing things within the code would be to allow for easier debugging when using the "Display CSG Tree" options.

The other part really comes down to me not knowing how OpenSCAD does things under the hood.  I had assumed that OpenSCAD already did such tagging as part of the caching process in order to avoid constantly having to re-compute the exact same geometry over and over again.  If that's not the case, then wow!  Here's a way to significantly reduce computation time of tiled calls.

@nophead
> Why not just import the scad source code instead of the CSG?

Because instead of writing a parser that's re-writing the entirety of OpenSCAD.  Sorry but your question reads more as "why does CSG exist at all?"

nophead, 2024-09-08:
CSG is basically a subset of the same syntax as OpenSCAD source but all the symbolic information is reduced to numbers. If you want the symbolic information as well as the geometry it would be better to work from the OpenSCAD source. The syntax is richer but the parser is built automatically from the grammar description and it is all open source, so it should be easy to use in another project.

jordanbrown0, 2024-09-09:
[ I should note that although I have opinions, I have no yes/no authority. ]

@EmperorArthur says:
>  Would you consider the approach of modifying the variables to add those two optional fields to the standard regardless? 

I think that it would be entirely appropriate and desirable to declare that CSG consumers must ignore any arguments that they do not understand.

>  Even having the module name can be useful for reading a CSG. 

I don't particularly have an opinion on whether to include a module-name parameter to the group() that's emitted for a module invocation.  (However, PR#4478, if it ever gets merged, introduces a couple of module-like constructs that don't have names - notably, functions that return geometry and module literals.)

> The other part really comes down to me not knowing how OpenSCAD does things under the hood. I had assumed that OpenSCAD already did such tagging as part of the caching process in order to avoid [...]

Yes and no.  OpenSCAD does not cache the results of module invocations.  To do so it would need to consider all of the stuff I mentioned - arguments, $ variables, use of non-pure functions like rands(), et cetera).  For most OpenSCAD programs, the execution of the program takes negligible time.  (Generally only programs generating large complex polyhedra take significant time.)  OpenSCAD's interpreter is very low-tech, but for most cases is plenty fast.

That execution produces a CSG tree - I believe it's exactly the CSG tree that it's willing to show you.

Generally, what takes time in F5 preview is triangle count.  I believe intersection() is quite expensive for preview, and difference() is more expensive than union().  But none of that is amenable to caching, because the results depend on where the various triangles are on the screen, how they overlap at the current view angle and distance, et cetera.  (And the computation is all done in the graphics card.)  You pay the performance cost every time you change the view.

The big expense is usually at the F6 rendering step.  That step consumes a CSG tree and emits polyhedra (or polygons, for 2D).  The rendering step *does* do caching.  I'm not sure of the details, but I believe it detects identical CSG subtrees and reuses previous results for those subtrees.  But that's from the CSG, deduplication that an external CSG consumer could do exactly the same way that OpenSCAD does it.

As an example, on my relatively ordinary system, randomly positioning 10,000 unit cubes in a 100x100x100 box takes about half a second to execute and preview, and rendering takes about three minutes with CGAL (old) and six seconds with Manifold (new).  Net, the execution time (vs render time) is only a tiny fraction of the time with CGAL, and is only a modest fraction (less than 10%) of the time with Manifold.

@nophead writes:
> CSG is basically a subset of the same syntax as OpenSCAD source but all the symbolic information is reduced to numbers. If you want the symbolic information as well as the geometry it would be [...]

CSG is a tiny subset of OpenSCAD semantics.  CSG is a text representation of a simple data structure; OpenSCAD is a programming language.  The difference is not just symbolic representation - that the radius of the circle is 47 rather than `r` - but what calculations led to that value and what flow control decisions led to generating a circle rather than a square or one circle rather than ten circles.  (For those that are familiar with them, comparing CSG to OpenSCAD is like comparing JSON to JavaScript.)  The OpenSCAD parser and lexer per se are about 1300 lines; the rest of the execution engine is something like another 8000, not including any geometry processing.  Yes, it's all open source and so available for inclusion by another project... if you want to either freeze the version you're using or commit to ongoing maintenance to integrate OpenSCAD changes into your project.  It is fairly easy to consume CSG; consuming OpenSCAD is a major project.

It would be quite cool if somebody could extract the OpenSCAD execution engine into a library that other projects could call, but so far nobody has stepped up to do the work to do that.

KeithSloan, 2024-09-10:
"The OpenSCAD parser and lexer per se are about 1300 lines"

For the FreeCAD CSG importer. I defined for CSG using ply.lex and ply.yacc

We are then able to create equivalent Brep objects for the CSG operatives apart from

- Hull
- minkowski

The other challenge is polyhedron's which can be created but currently not very efficiently and not great for
subsequent editing in FreeCAD.

This became a bigger problem as people started implementing like the BSOL and BSOL2 where functions for understandable reasons have been built up from a simple building block(s) based on  a polyhedron.

I know I could try and understand the OpenSCAD parser by looking at the source but this seems like a big task,
so I would like to know could you advise how much effort it might be to duplicated the scad parser and have it drive a different set of code i.e. more brep creative operations?

Similar question how much effort to parse a scad file for library references. 

What would be useful in our case would be to have a CSG file that has extra info that we could pick up on libraries and calls in the CSG so that on parsing if we found a library call with parameters, would could then stop interpreting the CSG invoke an equivalent to the library call and then drop back to CSG once the library call is over i.e. skip any CSG produced as a result of the library call.

Thanks

jordanbrown0, 2025-03-08:
See also #5126.