openscad/openscad #5278: roof(method="straight") fails with complex outlines

MethylBromide, 2024-09-02:
**Describe the bug**
WARNING: PolySet -> Manifold conversion failed: NotManifold
Trying to repair and reconstruct mesh..
ERROR: [manifold] Input mesh is not closed!
ERROR: [manifold] Surface_mesh -> Manifold conversion failed: NotManifold

-- and no object is created.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable the **roof** function in your preferences.
2. Execute the script below. This displays the word "plugh" with a 45° slope to its sides. So far, so good.
3. Now check the **growtext** checkbox in Customizer. This results in the error shown above, and no object is created.
4. Use customizer to change **method** to "voronoi" rather than "straight". This avoids the error and gives the intended output.
5. Use customizer to change **method** back to "straight", which causes the problem to recur (though the console error isn't always shown).
6. Now in Customizer, set **$fa** = 4. This avoids the error and gives the intended output.

**Expected behavior**
I would hope **roof** would work with no error even if the polygon segments are small.

**Code reproducing the issue**
```
growtext = false;
$fa = 1;
$fs = 0.1;

module chamfer(height=2, angle=45, method="straight", size = 1000, embiggen = false) {
	a = min(abs(angle),89);
	render()
	difference() {
		union() {
			scale([1,1,sin(45)/sin(a)])
			roof(method)
			offset(delta=embiggen ? height : 0, chamfer=true)
			children();
		}
		translate([-size/2, -size/2, height]) cube([size, size, size]);
	}
}

chamfer(.3, embiggen = growtext) text("plugh",  size=10, halign="center", valign="center");
```

**Screenshots**
n/a

**Environment and Version info (please complete the following information):**
 - OS: [e.g. Windows 10, macOS 12 Monterey, Ubuntu 22.04]
 - System: [e.g. Intel PC 64-bit, Raspberry PI 4]
 - OpenSCAD Version [e.g. 2021.01, 2023.04.16] and where it was installed from

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

**Additional context**
I'm not certain the problem is entirely with "roof" -- it might be "offset" is producing some nonsensical output which might contribute to the problem. But either way I'd like this sample code to work.

UBaer21, 2024-09-02:
version 2024.8.30 renders this without error
[image]

But the cube size has some severe impact on the result - using bigger cubes to cut will lower the precision and causing strange artifacts.  here with size=20000
[image]

MethylBromide, 2024-09-02:
I am using the same version as you. Sorry I didn't say the version. The "working" picture you posted is of the situation I already know works -- where we _don't_ enlarge the text before roofing it. The error occurs when you check the "growtext" box, which results in using **roof** on the output of **offset** (with delta > 0), which is supposed to look like this:
[image]
Notice how the letters are thicker than in your screenshot.

thehans, 2024-09-03:
> But the cube size has some severe impact on the result - using bigger cubes to cut will lower the precision and causing strange artifacts. here with size=20000 

Looks like another case of https://github.com/openscad/openscad/issues/4871

coryrc, 2025-11-13:
I use the Manifold backend and the problem does not repro at step 3. Recent build from HEAD.