openscad/openscad #5280: No python support available in features

MaierJuerg, 2024-09-03:
Installed [OpenSCAD](https://www.openscad.org/) version 2024.08.30.ai20307 (git dbd7d4f3d)
on my ubuntu 24.04 system

I can not activate python support in Features, it does not show up.

what are my options?

t-paul, 2024-09-03:
This is not available yet. See https://pythonscad.org/ for the current state of work.

closed 2024-09-03 by t-paul

MaierJuerg, 2024-09-04:
Hi Torsten
Thanks for your reply.
One of your examples shows a python script that starts with
import openscad
how can I make this library available in my python code?
Regards
Juerg

gsohler, 2024-09-04:
Hi Juerg,

I am  wondering, which python script you are referring to. Can you share a
source ?
"  import openscad" is not what i used to write.
PythonSCAD(Aka OpenSCAD with integrated python) comes with a built in
python library named "openscad" which you can import.

BTW: I have been to St Margrethen last Sunday.

MaierJuerg, 2024-09-04:
Hi
You might not have been in my home town (tiny) but in St. Margrethen, not
St. MargArethen, about 50 km away from there.

I see this import statement on https://pythonscad.org/ in the path extrude
example. (btw, some links on this page are broken for me)
Updated to latest pythonSCAD-2024-08-10.AppImage and enabled python in
features.

As I understand I have 2 main options:
run python and use libraries like solid or openpyscad
- this has the disadvantage of having to adjust to a new interface and
sequence rules

run openscad with calls to python code that returns scad-code
- I would rather like to use this approach

Running my example I first got "File is not trusted". If running the
AppImage from the console with --trust-python this error went away but I
get now "python not found".
I can run python from /usr/lib/python3 and I have also set PYTHONPATH to
this folder. Is there a way to find out why openscad can not find it or
where it is looking for it?

my scripts:
scad:

// kombination von openscad und python

$fn=100;

use </home/juerg/pcjmProjekte/drehTurmScad/drehTurm.py>

// len, xRot, yRot
dim = [[200,0,0,0],[100,90,0,0],[150,0,90]];

zeichneSegment(dim[0]);

python:
def zeichneSegment(laenge, rotX, rotY):
    stmt = "translate([0,0,0]"
    stmt += f"rotate([{rotX}, {rotY}, 0])"
    stmt += f"cylinder(d=40, h={laenge})"
    return stmt

or am I trying things that will not work?
Regards
Juerg

gsohler, 2024-09-04:
Hi Juerg,

Apparently i was mistaken about your home town, i was passing st.
margrethen when returning from switzerland and heading to germany.

Usually i write: 'from openscad import *'  and not 'import openscad'
please show me an exact broken link, i could not see any yet.

I tested the AppImage on Fedora28, i have /usr/bin/python and its version
is 3.11.6(but i doubt the version is so important),
also there is /ust/lib/python3.11/  in my system

you use pythonscad in the wrong way. you don't use python functions to
return scad code. you really do python like

def zeichneSegment(laenge, rotX, rotY):
  return cylinder(d=40,h=laenge).rotate([rotX, rotY, 0]).translate([0,0,0])

hope you get your python find issue sorted out soon

MaierJuerg, 2024-09-05:
Thanks for your explanations.
I would have loved to return scad statements from the python call...
Having to use the the dot notations in python instead of the scad way looks
ugly to me.

It a) reverses the sequence
scad:
color()
translate()
rotate()
cube()
python:
cube().rotate().translate().color() // reversed sequence!

and b) the statement need to be broken down by adding line breaks as the
lines get very long otherwise.

But that is just my personal opinion

The broken link I had was with the Tutorial link on the main page, but that
works for me now when I tried it again.

My main problem is that when I open a new python project (pycharm
community), create a venv, add my python3.11 interpreter and try to run the
first tutorial example code my project does not find the openscad library.
pip install openscad is also not working. What do I need to install into my
python to make the examples work?

Regards
Juerg

gsohler, 2024-09-05:
Hi Juerg,

You can still use scad syntax in pythonscad even though it's not the
intentional use.

solid = scad("shere(5);")
output(solid)

The scad function takes a string with valid scad language.

... and you are not forced to use dot notation, you can still so

b=translate(rotate(mycube,[1,1,1]), [2,2,2] )

which effectively fixes your desired command order

you will not be able to use the openscad library in the python3.11
interpreter.

The openscad library is only available INSIDE the PythonSCAD Program when
you put your code into the graphical editor on the left side of the program.
Reason is, that there needs to be a tight binding between the interpreter
and the openscad kernel, which is not possible with a standalone python
interpreter.

(of course you can use an external editor later)

Hope that helps