openscad/openscad #5281: Slow SVG extrude rendering

hackwin, 2024-09-04:
**Is your feature request related to a problem? Please describe.**
The renderer is too slow when extruding SVG shapes into 3d.  The "openscad-2020.10.11.ci6009" takes about 10 seconds while the 
"OpenSCAD-2024.08.30-x86-64" version takes about 2 minutes.

**Describe the solution you'd like**
Fast SVG extrusion.  Seems like this should be 1 second or less.  Maybe some command line argument?

> Newer version of clipper2 works fine and is really fast for this one, it took less than 0.2s in total. Not entirely sure why the older version of clipper2 and clipper1 are slow...
-Chum Kit LAM (from discuss@lists.openscad.org)

**Describe alternatives you've considered**
PyMesh, Inkscape, writing a script that outputs STL triangles from simple XY SVG points

**Additional context**

[image]
[image]

```
$fn=50;

color([45/255,252/255,132/255]){
    linear_extrude(1){
        resize([203.2+0.1,203.2+0.1,1]){
            offset(-100)offset(100){
                import("plain.svg",center=true);
            }
        }
    }
}
```

```
$fn=50;

color([245/255,203/255,15/255]){
    resize([203.2,203.2,3]){
        linear_extrude(10){
            import("plain.svg",center=true);
        }
    }
}
```

Sample files: https://mylabyrinthlab.com/data/1y4a2u/
Used here: https://mylabyrinthlab.com/

UBaer21, 2024-09-05:
the offset taking unexpected much time - use fill() instead

kintel, 2024-09-05:
Unrelated to the slowness, but the base could probably be constructed using `hull()` rather than two offsets.

kintel, 2024-09-05:
Interestingly, `openscad base.scad -o out.stl` causes an assertion failure for me:
```
Assertion failed: (s0 != s1 && s0 != s2 && s1 != s2), function append_stl, file export_stl.cc, line 161.
```

kintel, 2024-09-05:
The assert is likely caused by #4867 - we can deal with that separately.

kintel, 2024-09-05:
Some light-weight archeology:
* Last fast version I can find: 2021.11.15
* First slow version I can find: 2023.01.29

Let me know if anyone can reduce that span. I can try bisecting the source, but building old software isn't always fun..

UBaer21, 2024-09-05:
so i could test with 2021.10.19 was still fast 
2021.12.07 is already slow (git 28bf7c482)

test:
```
$fn=40;
offset(40)import("plain.svg");
```
7.4 sec vs 1.5 sec  -  the $fn and offset both will drastically increase the time used.

pca006132, 2024-09-05:
One guess: Maybe this is due to https://github.com/openscad/openscad/commit/f5816258db263408a7aa2feec1fafffe77644662 which draws the 2D polygons according to `$fn`, while the previous versions use some default value?

We can test this by checking the extruded polyhedron size (without offset) in the two versions, if their vertex count differs a lot, it means the slowness is caused by increase in vertex number, so this is not really a regression. Why clipper 1 is so much slower than clipper 2 is another issue.

pca006132, 2024-09-05:
Maybe we need to enlarge the thing a bit as well, as the current polyhedron may have vertices merged.

UBaer21, 2024-09-05:
Can't find a big time difference when using the same vertices count (not using $fn) 
old version seemed to ignore the $fn for the 2D import.

kintel, 2024-09-05:
> One guess: Maybe this is due to [f581625](https://github.com/openscad/openscad/commit/f5816258db263408a7aa2feec1fafffe77644662) which draws the 2D polygons according to `$fn`, while the previous versions use some default value?

Oh, good find! In that case, this is likely a feature, not a bug.
@hackwin Does that make sense to you? Did you specify a global `$fn` value with something specific in mind?

pca006132, 2024-09-05:
This is my test code for using clipper2 (with manifold), in case it is helpful:

```
  manifold::Polygons polygons;
  for (const auto &outline : poly.outlines()) {
    polygons.emplace_back();
    for (const auto &vert : outline.vertices) {
      polygons.back().emplace_back(vert.x(), vert.y());
    }
  }

  manifold::CrossSection cs(polygons, manifold::CrossSection::FillRule::EvenOdd);
  auto jt = manifold::CrossSection::JoinType::Square;
  if (joinType == ClipperLib::jtMiter)
    jt = manifold::CrossSection::JoinType::Miter;
  else if (joinType == ClipperLib::jtRound)
    jt = manifold::CrossSection::JoinType::Round;
  auto cs2 = cs.Offset(offset, jt, miter_limit);
  return std::make_unique<Polygon2d>(ManifoldUtils::polygonsToPolygon2d(cs2.ToPolygons()));
```

hackwin, 2024-09-07:
> Oh, good find! In that case, this is likely a feature, not a bug. @hackwin Does that make sense to you? Did you specify a global `$fn` value with something specific in mind?

Version "OpenSCAD-2024.08.30-x86-64" does get lower rendering time by lowering $fn.  If $fn=10, then the rendering take less than 10 seconds.  However, "openscad-2020.10.11.ci6009" takes about the same time even if $fn=100.

Maybe there should be a set of tests that check the run-time across the versions of OpenSCAD to find performance gains and losses.

t-paul, 2024-09-07:
The change in `$fX` behavior for SVG is documented in the Manual: https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/SVG_Import

hackwin, 2024-09-08:
> The change in `$fX` behavior for SVG is documented in the Manual: https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/SVG_Import

That worked and speed is high again.  The main factor that helped was setting $fs = 100.

> $fs
Double. The minimum segment length to use when converting circles and arcs to polygons. [Note: Requires version Development snapshot]

t-paul, 2024-09-08:
> The main factor that helped was setting $fs = 100.

I'm not sure this is a good idea. Right now the #3994 linked above notes:

>  Use the best estimates for ellipses (max radius, all 3 parameters), circles (all 3 parameters), curves (just obey $fn), and arcs (just obey $fn) - while preserving previous minimum defined steps.

Setting excessive values now may cause a similar issue like this one, but in the other direction in regard to resolution if the restriction of *not lowering* the resolution would be allowed.

kintel, 2024-09-08:
Ah, I didn't realize this was both well-known and documented. I guess we can close this and take any follow-up discussion back on the mailing list.

kintel, 2024-09-15:
Closing this as not an issue.

closed 2024-09-15 by kintel