openscad/openscad #5287: Mac OS Install - couple of missing dependencies.

MrBenGriffin, 2024-09-09:
Please first consider that general questions about using OpenSCAD or how to write code for specific models are better asked on the mailing list. See https://openscad.org/community.html for more details.

MacOS X native
Which environment is the issue about, e.g. Compiling with MSYS2 on Windows

zlib missing dependency.
meson - needs installing / updating
automake - needs installing / updating (for libxml2).
there was a broken dependency for meson also.

kintel, 2024-09-09:
Can you provide more info? It sounds like you're just installing build deps using homebrew and occasionally having to upgrade packages is normal as we don't control homebrew.

kintel, 2024-09-09:
How did you install? None of the mentioned dependencies should need resolution for a binary install.

MrBenGriffin, 2024-09-10:
compile from source

pca006132, 2024-09-10:
Can you show us the commands you use to compile from source? Are you using https://github.com/openscad/openscad/blob/master/scripts/macosx-build-dependencies.sh to build the dependencies?

MrBenGriffin, 2024-09-10:
Yeah, that’s exactly what I was using - my docs were based on what was missing from that shell.

kintel, 2024-09-10:
When building from source, you need to install meson yourself, typically with homebrew. Did that go wrong somehow?

zlib comes with the OS. In what way did you encounter that as a missing dependency?

MrBenGriffin, 2024-09-10:
Yeah, meson is already installed - I do quite a lot of python stuff, so both meson and ninja were there -  but their packages were not recent enough for one of the dependencies.
Again, the zlib was not being found during the dependency builds.  It seems that there was a trick missing with system libraries - but I didn’t seem to find it.  libzip still fails to download due to certificate errors - so I manually downloaded that, and put it into the right place. libxml2 needed to be compiled and that was failing internally - due to  not finding the correct python bindings.  It would be very useful to add (for macos build, at least)..

export PYTHON_CFLAGS=`python3-config  --cflags` && export PYTHON_LIBS=`python3-config  --libs`

Otherwise meson (or ninja) starts flailing around, especially if one has multiple python interpreters installed.

It took a good couple of hours to compile.  I was wondering about maybe forking some of that work out - while the actual compiles used all 20 cores (thanks to -j), a lot of the tests were single core/single threaded.  I did also upgrade cmake.  What also surprised me was that several of the libraries in dependencies are already installed - but they weren’t located.  

In fairness, I did all of this hoping that the CGAL Polygon Mesh rendering was going to run faster - but it does not. 

Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
Geometries in cache: 20
Geometry cache size in bytes: 3180880
CGAL Polyhedrons in cache: 7
CGAL cache size in bytes: 65751256
Total rendering time: 0:00:58.514.  <<———————— Just for an M5 bolt from ISOThread.
Top level object is a 3D object (Nef polyhedron):
   Simple:     yes
   Vertices:    10291
   Halfedges:   47160
   Edges:       23580
   Halffacets:  26588
   Facets:      13294
   Volumes:         3
Bounding box:
   Min:  -17.32, -15.00, 0.00
   Max:  57.32, 15.00, 50.00
   Size: 74.64, 30.00, 50.00
Rendering finished.

I would *love* to know how I might speed CGAL renders up.

kintel, 2024-09-10:
One note: The main purpose of macosx-build-dependencies.sh is to build fat binaries for distribution, so we cannot really rely on any libraries being picked up from the system, unless we can trust those to be available everywhere (like OS builtins). It works fine for dev purposes, as long as you're willing to take the one-time compilation hit. It's possible that we inadvertently run some unnecessary tests. If you spot any, feel free to open a pull request for that.
If you just want to build for your own system, using macosx-build-homebrew.sh is likely a much quicker solution.

For the individual failures, it's hard to address without breaking it down and troubleshooting each one separately. I think the cleanest way forward would be to open a separate ticket for each issue, then we can troubleshoot and fix those separately.

pkriens, 2024-09-30:
@MrBenGriffin could you share your Macos setup?

I've tried to build but gave up after a few hours. I've got an M1 Max mac. Would be very interested in a bit of help. You can contact me on peter.kriens@aQute.biz.

kintel, 2024-09-30:
Closing this as not actionable.
For individual issues, please open separate issues per problem.
Remember that you can always download nightly builds from https://openscad.org/downloads.html

closed 2024-09-30 by kintel