openscad/openscad #5289: difference adds module

dcherrera, 2024-09-10:
**Describe the bug**
when differencing a mudule from a square, if the module is built with another module as part of it. the module is unioned instead of differenced.

**To Reproduce**
Steps to reproduce the behavior:

ends();

module m3(){
    //make head slightly larger so it fits after printing
    cylinder(d=5.5,h=3,$fn=360);
    //make safe slightly smaller so threads have something to bite
    translate([0,0,3])cylinder(d=2.8,h=5,$fn=360);
}

module holder(){
//base
difference(){
    union(){
            translate([0,-2,0])linear_extrude(4)square([134,219]);
            translate([0,-2,4])linear_extrude(4)square([134,219]);
                    //bookends
                    translate([2,-2,0])linear_extrude(12)square([130,4]);
                    translate([2,213,0])linear_extrude(12)square([130,4]);
            }
            translate([33.5,-2,6])rotate([-90,0,0])m3();
            translate([100.5,-2,6])rotate([-90,0,0])m3();
            #translate([33.5,217,6])rotate([90,0,0])m3();
            #translate([100.5,217,6])rotate([90,0,0])m3();
            translate([3,2,4])linear_extrude(8)square([128,211]);
            translate([12,2,0])linear_extrude(8)square([110,211]);
            //slits
            translate([2,20,2])linear_extrude(12)square([130,2]);
            translate([2,40,2])linear_extrude(12)square([130,2]);
            translate([2,60,2])linear_extrude(12)square([130,2]);
            translate([2,80,2])linear_extrude(12)square([130,2]);
            translate([2,100,2])linear_extrude(12)square([130,2]);
            translate([2,120,2])linear_extrude(12)square([130,2]);
            translate([2,140,2])linear_extrude(12)square([130,2]);
            translate([2,160,2])linear_extrude(12)square([130,2]);
            translate([2,180,2])linear_extrude(12)square([130,2]);
            translate([2,200,2])linear_extrude(12)square([130,2]);  

            }
}
module ends(){
    difference(){

        translate([2,-2,0])linear_extrude(80)square([130,7]);

//work around
            //translate([33.5,-2,6])rotate([-90,0,0])m3();
            //translate([100.5,-2,6])rotate([-90,0,0])m3();
        holder();
    }
}

**Expected behavior**
render the code above. the m3 module is added to the ends module output when it should be difference with the rest of the holder module

**Code reproducing the issue**
If applicable, add trimmed down, self-contained code demonstrating the issue.

**Screenshots**
[image]

**Environment and Version info (please complete the following information):**
 - OS: [mac os 14.6.1]
 - System: [apple silicone]
 - OpenSCAD Version 2021.01 from website
**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

**Additional context**
also reproduces on openscad playground website

https://ochafik.com/openscad2/#H4sIAAAAAAAAE61W227iMBD9FSvdB5BCc+uFsurDqtI+rirtvrVRNSQTYuHErO2UUpR/33EoIXShN/BTPJ45Z3zGGXvpzEBBoZ3R0oHE8Ee8BZM7I8ebCVhMlKzK9FQnkDqukyGYSiH53jkFlDyTojGDNoNET+hTwPNiUJVcljQBrXkxow+FZYpqkEghlXZi19GyUkmDsyT6fXSJLA2WhhYpXvf63+/L+7KQaSWQFVGvv7wvGQ3PK2CKLEdImRZ8khuxYALUBBXTknHDMm40g8yQYaZ4aXg5WYUmC8Ftar30+vz03M2vI/dbVl5HF74l64BryHADrgsQYoVuckW8muXwSA6yQJMTOjOSjbnBFYZRUGoBBnt3vuu7Udzv8IanQ+I97/LWnX3mJDH52b163hg0QaY8y5AkTbCVoFG8na3HFu8gdP24T7QI6gGfjKpS7J319d8KFDkE0ZkbBlfxet/7MM6+gLEetAEpp7aWu9c7XOHufIOwQ+bbZPZQbUGFQfQVrHqvElFEh4USvIj7SprGNLiypSUWezD3axj4/qdCT16zhsFlN/aDoSvaj8d2Wd1dRR9utAuHBBy8dXACKsGOAnQwAn8nhufRL2f0XmDCpcD3Khu+lRtt7nCIi8MhhodDUJEPxziCoMERFA2OIGlwBE3Dj2rKmG3aXZy6aeQvbXx1gb3051cNfBP2fgMc+lvMl/HqUvS8uVRTBs31+fof+nrj2or9TOta31ovJiuFU8frh8BPqQqw1/pEjOmax6eZVOY/Y7NhdbN6M4zu4tp1HjnO7TtFyAmZyAE3jrc8maJyRhkIjfYhspCVsc5UASTYohKG28eKTCrdumHKjVRrLIuP7Yz8jCz4c4ta2yeJEDDTmN60i39g3OTXLFos5yS7Si/DhMh0Luc/nrBN1s5/55DKeWuyBf5VFWNU66zq+h8vsqrUlAkAAA==

jordanbrown0, 2024-09-10:
When you provide a reproduction case, please try to reduce it to the minimum necessary to demonstrate the problem.

I think this is working exactly as the program says it should.  holder() generates a sort of horizontal frame with two holes in the end, and subtracting that from the vertical cuboid means that there's material left in those holes.

holder() generates:
[image]

and then ends() subtracts that from the vertical cuboid.  The result is material anywhere that is in the cube, and is not in the holder object.

If you really intended to drill holes through *both* the vertical cube and the horizontal frame, you need to move those holes to the difference in ends(), as you have done in your "workaround".

Perhaps you were expecting the "negative" m3s in holder() to somehow remain "negative", and punch holes through the cuboid in ends().  That's not the way it works; there is no such thing as a negative object.  `difference() { A; B; }` results in the material that is in A and not in B.

Additional notes:
* When posting programs in github, wrap them in triple-accent graves so that indentation will be maintained and they will be set in a fixed-pitch font.
* Preview of this model is very difficult to understand because of all of the Z-fighting.  Best practice is to ensure that when you are differencing you ensure that there are no coincident faces; ensure that the object being subtracted extends beyond the side of the object being subtracted from.
* `linear_extrude(z) square([x, y]);` is the same as `cube([x,y,z]);`.

Simpler demonstration:
```
cube(10);
```
generates:
[image]

Subtracting a cylinder from that...
```
difference() {
    cube(10);
    translate([5,5,-1])
        cylinder(d=5, h=12);
}
```
yields
[image]

Subtracting *that* from another cube
```difference() {
    translate([-5,-5,-15]) cube(20);
    difference() {
        cube(10);
        translate([5,5,-1])
            cylinder(d=5, h=12);
    }
}
```
yields
[image]

The cylinder was not unioned with the large cube.  Rather, it was not present in the cube-with-a-hole object, and so not subtracted from the large cube.

dcherrera, 2024-09-10:
dang. sorry. this makes sense now. your right. thansk for taking the time to explain! you could have just closed it and said i was wrong. lol. have a good day.

jordanbrown0, 2024-09-10:
You're welcome!