openscad/openscad #5292: Axis Font Not Rendering Correctly Windows 11

UltimateCodeWarrior, 2024-09-14:
The number font looks like bad cursive. 

Take a look:

[OpenSCAD](https://www.openscad.org/) version 2024.08.30 (git dbd7d4f3d)

[image]

UBaer21, 2024-09-14:
obviously no line gaps between numbers/signs

hzeller, 2024-09-14:
Interesting, so `glEnd()` does not seem to behave the same on that platform (in `decodeMarkerValue()` in GLView.cc).

I don't have access to a Windows machine to test, but maybe someone knows the differences in their GL implementation ?

UBaer21, 2024-09-14:
Win10 is working fine though. Some change in Win11
[image]

hzeller, 2024-09-14:
Maybe the gl state machine does not like if it is called with multiple `glEnd()` without a `glBegin()` ? Can someone with Windows 11 try if  the following will fix it ?

```patch
--- a/src/glview/GLView.cc
+++ b/src/glview/GLView.cc
@@ -566,21 +566,25 @@ void GLView::decodeMarkerValue(double i, double l, int size_div_sm)
       glVertex3d(-y + (fh + bl), 0, x);  // z-label along z-axis; font below
     },
   };
+  bool needs_glend = false;
   for (const PlaneVertexDraw& axis_draw : axis_draw_planes) {
     // We get 'plot instructions', a sequence of vertices. Translate into gl ops
-    const auto plot_fun = [ = ](bool pen_down, float x, float y) {
+    const auto plot_fun = [&](bool pen_down, float x, float y) {
         if (!pen_down) { // Start a new line, coordinates just move not draw
-          glEnd();
+          if (needs_glend) glEnd();
           glBegin(GL_LINE_STRIP);
+          needs_glend = true;
         }
         axis_draw(x, y, font_size, baseline_offset);
       };
 
     hershey::DrawText(pos_number_str, i, 0,
                       hershey::TextAlign::kCenter, font_size, plot_fun);
-    glEnd();
+    if (needs_glend) glEnd();
+    needs_glend = false;
     hershey::DrawText(neg_number_str, -i - prefix_offset, 0,
                       hershey::TextAlign::kCenter, font_size, plot_fun);
-    glEnd();
+    if (needs_glend) glEnd();
+    needs_glend = false;
   }
 }
```

t-paul, 2024-09-14:
CI-Builds via #5293:
* Installer: [OpenSCAD-2024.09.14-x86-64_gl-font-fix-Installer.exe](https://output.circle-artifacts.com/output/job/b7b18e63-c91d-4de7-b034-693768a42d08/artifacts/0/64-bit/OpenSCAD-2024.09.14-x86-64_gl-font-fix-Installer.exe)
* ZIP: [OpenSCAD-2024.09.14-x86-64_gl-font-fix.zip](https://output.circle-artifacts.com/output/job/b7b18e63-c91d-4de7-b034-693768a42d08/artifacts/0/64-bit/OpenSCAD-2024.09.14-x86-64_gl-font-fix.zip)

commit 39bfcbcc (2024-09-17): Fix Axis Font Not Rendering Correctly Windows 11 (fixes #5292).
commit ccf5b328 (2024-09-17): Merge pull request #5293 from openscad/gl-font-fix

Fix Axis Font Not Rendering Correctly Windows 11 (fixes #5292).
closed 2024-09-17 by t-paul