openscad/openscad #5294: Manifold "zebra" triangulation issue is back.

kintel, 2024-09-15:
Using manifold's triangulator, triangulation sometimes fails. This was addressed earlier in https://github.com/elalish/manifold/issues/831, but seems to not fix all issues. See image below.

Design:
```
$fn=64;
text(text="R",font="naturePro",size=5);
```

Minimal cmd-line:
```
openscad zebra.scad -o out.png --enable=manifold
```

Cmd-line to reproduce image output:
```
openscad zebra.scad -o out.png --render --imgsize 1024,1024 --camera=0,0,100,0,0,0 --viewall --autocenter --projection=ortho  --enable=manifold
```
[image]

pca006132, 2024-09-20:
Weirdly, when using `-DUSE_MANIFOLD_TRIANGULATOR`, I don't get your output:

[image]

kintel, 2024-09-20:
After merging #5282, this works again!
It's odd, as I can swear I tested against that PR a week ago, but it works now, so let's move on :)

closed 2024-09-20 by kintel