openscad/openscad #5296: refined use of ! modifier

UBaer21, 2024-09-16:
currently ! will only work once and uses the first occurrence.

```
A(10);
!A(20);

module A(size){
!cube(size);
sphere(size);
}
```
It would be great if i could select a fork/path so  the second occurrence is used to display the bigger cube.

UBaer21, 2024-09-16:
At the moment it going along the calculation and stopping on the first ! .. which is the cube for the first example.  The rest is calculated but ignored to draw.

I thought about having hierarchy and a full inspection so if there is a ! on root level check if there is another that path. That would mean the second A module  and  then the sphere would be shown as the cube within B() is a deeper level.  And  the B(size) call is not !B(size)  else it show the difference of a cube(size) and a sphere (1).  
Or simply show everything from the last/deepest !  - if there are multiple - show the one with the most ! along the path

jordanbrown0, 2024-09-16:
I'm not immediately seeing what that algorithm would be.  (Which is not to say that there isn't one.)

My point is that you'll need to think through what the graph-walking algorithm would be and checking that the results are at least sort of what you're trying to achieve - and then trying to figure out how to explain it to a novice.

UBaer21, 2024-09-16:
It is a debug tool .. so  you start to "!" the part you need to debug ... and just place more "!" further down for what you need to see. 
The way it works at the moment is that you don't get the part you want - you first need to disable the other parts which is quite unfortunate.

nophead, 2024-09-17:
Perhaps allow multiple exclamation marks together and the one with the most
marks wins.