openscad/openscad #5308: Support externally built Manifold

kintel, 2024-09-21:
We recently added the `-DUSE_BUILTIN_MANIFOLD` CMake flag, defaulting to true, but the support for setting this flag to false is incomplete.

TODO:
* [ ] macOS: Build our own Manifold for distribution (macosx-build-dependencies.sh) and build OpenSCAD against it.
* [ ] Linux: Make sure OpenSCAD can build against a distro-packaged Manifold. Any suitable distros apart from nixOS? How is Clipper2 managed?
* [ ] Windows: msys2 + MXE, but may be out of scope for now
* [ ] Bonus: Also build Clipper2 outside of Manifold
* [ ] Bonus: Allow external Clipper2 but builtin Manifold

pca006132, 2024-09-22:
I think @cjmayo is packaging manifold for gentoo.

kintel, 2024-09-23:
@cjmayo Thanks! I'll ping you when/if I make changes so we don't break anything.

kintel, 2024-11-14:
@cjmayo How do you get around the manifold target name differences (`manifold` when using as submodule and `manifold::manifold` when using the installed version via `manifoldConfig.cmake`)?

hzeller, 2024-11-14:
FWIW, building using the [system-provided manifold](https://github.com/NixOS/nixpkgs/blob/master/pkgs/by-name/op/openscad-unstable/package.nix#L31) with [`-DUSE_BUILTIN_MANIFOLD=OFF`](https://github.com/NixOS/nixpkgs/blob/master/pkgs/by-name/op/openscad-unstable/package.nix#L110C6-L110C32) works fine out of the box in the Nixos packaging scripts.

kintel, 2024-11-15:
@hzeller Something doesn't feel right; is `manifoldConfig.cmake` being picked up on NixOS?
I tried installing manifold, from a nix installation on Ubuntu, but only got `manifold-2.5.1-unstable-2024-09-15`. I don't know enough nix magic to figure out how to get the latest one :/

pca006132, 2024-11-15:
We can just fix the target name differences in the upstream if needed.

pca006132, 2024-11-15:
@elalish

kintel, 2024-11-15:
@pca006132 Yeah, adding an ALIAS from manifold::manifold to the internal target might be a good thing.

At the moment I'm just curious how this works at all, as that might indicate a problem somewhere else.

pca006132, 2024-11-16:
I got some time this weekend, will look into this issue.

pca006132, 2024-11-16:
Looking at tbb, apparently they just do alias target. I guess we should do the same in the upstream.

hzeller, 2024-11-16:
In https://github.com/openscad/openscad/issues/5308#issuecomment-2477987237 I was confused to not see manifold in the library info.
Looks like even if it is linked to OpenSCAD, it is never printed -- filed #5432

hzeller, 2024-11-16:
@kintel the openscad-unstable version 2024-11-10 made it to the [nix unstable channel](https://search.nixos.org/packages?channel=unstable&from=0&size=50&sort=relevance&type=packages&query=openscad-unstable) if you want to try again.

kintel, 2024-11-16:
@hzeller Looks like CMake changes are coming in manifold, so I'm waiting for that to settle before trying any distro stuff again.

hzeller, 2024-11-17:
Ok, but I confirmed that apparently Manifold is compiled into the latest openscad-unstable, even without the cmake change. It is weird as the cmake does not emit a message when it finds manifold, but the final binary clearly uses manifold as rendered outputs retain color.

pca006132, 2024-11-17:
The cmake changes should be settled now, at least before another issue is found...

kintel, 2024-12-08:
Could someone double-check that Gentoo and NixOS can package the latest master (`8fb8c4865494bb536c0299fbfeb912109601df79` or newer) against Manifold V3?

pca006132, 2024-12-08:
Yes, NixOS can build with 8fb8c4865494bb536c0299fbfeb912109601df79 using external manifold v3.

Some of the logs:
```
-- Offscreen OpenGL Context - using Unix EGL GLX
-- TBB: 2021.11.0
-- mainfold: 3.0.0
```

closed 2024-12-20 by kintel