openscad/openscad #5311: Header includes: would be neat if relative to project-root

hzeller, 2024-09-22:
Right  now, all locally used project includes are included as local files, e.g.

```c++
#include "linalg.h"
#include "CSGNode.h"
#include "GeometryEvaluator.h"
```

Which makes it harder for someone new to the code base to see where things are and 'connect the dots'.

It would be good if this was relative to the project root to provide context at one glance; also reccommended by [styleguides](https://google.github.io/styleguide/cppguide.html#Names_and_Order_of_Includes)

```c++
#include "geometry/linalg.h"
#include "core/CsgOpNode.h"
#include "geometry/GeometryEvaluator.h"
```

If this sounds good, I have a script that can convert that.

kintel, 2024-09-22:
Agreed. I think this is just an artifact from back when we had a flat source tree.

kintel, 2024-09-22:
The only exception I can think of are includes isolated to sublibraries like the svg library, and perhaps .cc files assuming that the corresponding .h file can be included without a path.

hzeller, 2024-09-22:
I think for consistency, even the corresponding header to a `foo/bar/baz.cc` should include `foo/bar/baz.h` for consistency.

It should be the very first include in the file, but clang-format can take care of that once we can enable the header sorting.

PR #5312 (2024-09-25): Make headers include relative to project-root.
That is, relative to src/ and src/ext.

Fixes #5311

This was created with a script:

```bash
# Repair some project includes sometimes being included as system header.
sed -i 's/<Reindexer.h>/"Reindexer.h"/' $(find src -name "*.h" -o -name "*.cc")
sed -i 's/<PolySet.h>/"PolySet.h"/' $(find src -name "*.h" -o -name "*.cc")
sed -i 's/<PolySetBuilder.h>/"PolySetBuilder.h"/' $(find src -name "*.h" -o -name "*.cc")

sed -i 's/<json.hpp>/"json.hpp"/' $(find src -name "*.h" -o -name "*.cc")
sed -i 's|>FontListTableView.h<|>gui/FontListTableView.h<|' src/ [... cut]
closed 2024-09-25 by kintel