openscad/openscad #5313: Use Liberation Sans as fallback font when the specified font is not found

MethylBromide, 2024-09-22:
**Describe the bug**
When **text** is called with a **font** argument specifying a font that's not available on the Windows PC, an apparently random font is used instead.

**To Reproduce**
Steps to reproduce the behavior:
1. In preferences, enable fontmetrics.
2. Execute the script below.
3. Observe the characters appear in a font that is not the default text font used when **font** isn't specified. In my case, the font used is Goma Stencil, which is installed my system but doesn't have any status as a default font or anything. So it fails in a consistent way but it still fails.

**Expected behavior**
Specifying a nonexistent font should have the same result as specifying no font, i.e. the Liberation Sans font

**Code reproducing the issue**
```
text("abcd", font="NoSuchFont");
echo(fontmetrics());

```
**Screenshots**
[image]

**Environment and Version info (please complete the following information):**
 - OS: Windows 11 64-bit
 - System: Intel PC 64-bit
 - OpenSCAD Version 2024.09.17 (git 399551768)

**Additional context**
The output of fontmetrics() for me is:
`ECHO: { nominal = { ascent = 12.5733; descent = -2.9433; }; max = { ascent = 13.6109; descent = -4.2114; }; interline = 15.9709; font = { family = "Liberation Sans"; style = "Regular"; }; }`

jordanbrown0, 2024-09-22:
Font selection is handled by the fontconfig library, ref https://www.freedesktop.org/wiki/Software/fontconfig/ .

That library offers some configurability for how fonts are selected.  From OpenSCAD's point of view, it's a black box; we give them the font spec that you give, and they give us back a particular font.

I don't know what directories fontconfig searches to find its configuration files.  There are a couple in \Program Files\OpenSCAD\fonts, but I would have expected more somewhere.

Anyhow, it's unlikely that OpenSCAD's behavior will change; any behavior change would need to come from fontconfig's configuration or from changes to fontconfig.

jordanbrown0, 2024-09-22:
Fontconfig's documentation notes:
> Fontconfig performs matching by measuring the distance from a provided pattern to all of the available fonts in the system. The closest matching font is selected. This ensures that a font will always be returned, but doesn't ensure that it is anything like the requested pattern.

closed 2024-09-22 by jordanbrown0