openscad/openscad #5315: Compiling: No such file or directory: `manifold/manifold.h`

GoodClover, 2024-09-23:
**Environment**
Compiling on Void Linux, using "systems with older or missing dependencies", and then "Compiling" instructions.

**Issue**
When compiling an infinite loop of the following error is hit, and some form of memory leak occurs.
```
[ 40%] Building CXX object CMakeFiles/OpenSCAD.dir/src/core/DrawingCallback.cc.o
[ 41%] Building CXX object CMakeFiles/OpenSCAD.dir/src/core/SurfaceNode.cc.o
In file included from /home/olive/Downloads/Source/openscad/src/geometry/manifold/manifoldutils.h:5,
                 from /home/olive/Downloads/Source/openscad/src/geometry/manifold/manifoldutils.cc:2:
/home/olive/Downloads/Source/openscad/src/geometry/manifold/ManifoldGeometry.h:7:10: fatal error: manifold/manifold.h: No such file or directory
    7 | #include "manifold/manifold.h"
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[ 41%] Building CXX object CMakeFiles/OpenSCAD.dir/src/core/TextNode.cc.o
make[2]: *** [CMakeFiles/OpenSCAD.dir/build.make:300: CMakeFiles/OpenSCAD.dir/src/geometry/manifold/manifoldutils.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/olive/Downloads/Source/openscad/src/geometry/manifold/ManifoldGeometry.cc:2:
/home/olive/Downloads/Source/openscad/src/geometry/manifold/ManifoldGeometry.h:7:10: fatal error: manifold/manifold.h: No such file or directory
    7 | #include "manifold/manifold.h"
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/OpenSCAD.dir/build.make:286: CMakeFiles/OpenSCAD.dir/src/geometry/manifold/ManifoldGeometry.cc.o] Error 1
In file included from /home/olive/Downloads/Source/openscad/src/geometry/manifold/manifoldutils.h:5,
                 from /home/olive/Downloads/Source/openscad/src/geometry/manifold/manifold-applyops.cc:6:
/home/olive/Downloads/Source/openscad/src/geometry/manifold/ManifoldGeometry.h:7:10: fatal error: manifold/manifold.h: No such file or directory
    7 | #include "manifold/manifold.h"
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```
[image]

I would appreciate any help, thanks.

pca006132, 2024-09-23:
Did you update the submodules? Feels like you are using the old submodule.

No idea about the infinite loop issue. From my experience, openscad compiles faster and uses less memory when compiled using clang.

gsohler, 2024-09-23:
i believe that openscad has switched to a new version of manifold

type
git submodule update --init --recursive

to adapt

GoodClover, 2024-09-23:
haha, had just tried that and came back to report :P
fastest response I've ever got on GitHub, I'm impressed!
```
> git submodule update --init --recursive
warning: unable to rmdir 'bindings/python/third_party/nanobind': Directory not empty
warning: unable to rmdir 'src/third_party/quickhull': Directory not empty
Submodule path 'submodules/manifold': checked out '22c66051dfdbcefa2012e30dd12c9b5a20f89a01'
```
Guess I got unlukcy , this was a fresh clone yesterday.

However, the "memory leak" is still an issue.
```
[ 95%] Building CXX object CMakeFiles/OpenSCAD.dir/openscad_interface.cpp.o
[ 95%] Building CXX object CMakeFiles/OpenSCAD.dir/openscad_adaptor.cpp.o
[ 96%] Building CXX object CMakeFiles/OpenSCAD.dir/OpenSCAD_autogen/3YJK5W5UP7/qrc_common.cpp.o
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [CMakeFiles/OpenSCAD.dir/build.make:552: CMakeFiles/OpenSCAD.dir/src/geometry/cgal/cgalutils-project.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
^C
[olive@RedYeti openscad]$ make[2]: *** [CMakeFiles/OpenSCAD.dir/build.make:272: CMakeFiles/OpenSCAD.dir/OpenSCAD_autogen/mocs_compilation.cpp.o] Interrupt
make[2]: *** [CMakeFiles/OpenSCAD.dir/build.make:300: CMakeFiles/OpenSCAD.dir/src/geometry/manifold/manifoldutils.cc.o] Interrupt
make[2]: *** [CMakeFiles/OpenSCAD.dir/build.make:286: CMakeFiles/OpenSCAD.dir/src/geometry/manifold/ManifoldGeometry.cc.o] Interrupt
make[2]: *** [CMakeFiles/OpenSCAD.dir/build.make:328: CMakeFiles/OpenSCAD.dir/src/geometry/manifold/Polygon2d-manifold.cc.o] Interrupt
```

That first "Killed signal" is I assume Linux trying to keep itself alive.
The ^C is the Killed signal not doing much so I'm hoping it'll help to kill it :P
All 16GB of RAM and 32GB of swap had been full, system unresponsive, for a good couple minutes before that happened.
Whatever the generated `qrc_common` is, I assume is problematic?

t-paul, 2024-09-23:
For 16GB I would recommend an absolute maximum of -j3 for make/ninja/.. when using gcc. With clang it may allow the value a bit larger. This is mostly due to some of the libraries (CGAL) making heavy use of templates which requires lots of memory at compile time.

GoodClover, 2024-09-23:
Anything eating up memory and then swap in a constant growth, looks like a "leak" of some kind to me.
I was running in parallel, with unspecified `-j` as the README recommended.

Running with `-j3` now, is magnitudes slower, but memory seems to be kept in check.  It's constantly rising but, I assume when some form of atom is completed, it is also making quick sharp drops.

pca006132, 2024-09-23:
Oh, you probably never wants to do unspecified `-j`, iirc it just does the maximum parallelization possible, regardless of the number of cores in your machine. In this case, it probably spawns more than 10 processes compiling CGAL stuff and eat all your RAM for breakfast.

t-paul, 2024-09-23:
Where is that `-j`only in the docs? I fixed one place a while ago. If that is still written somewhere, that needs to be changed too.

GoodClover, 2024-09-23:
That is very good to know!  `-j$(nproc)` (8 for me), is what I usually see recommended.
[Line 225 of README.md](https://github.com/openscad/openscad/blob/a0522f56c0719aa50a5e66dababc31812ad3088c/README.md?plain=1#L255)

GoodClover, 2024-09-23:
It compiled!  :D  Thanks for your speedy & good help, all!

closed 2024-09-23 by GoodClover

t-paul, 2024-09-23:
> [Line 225 of README.md](https://github.com/openscad/openscad/blob/a0522f56c0719aa50a5e66dababc31812ad3088c/README.md?plain=1#L255)

Thanks, fixed via #5316.

jordanbrown0, 2024-09-23:
Appropriate levels or parallelization are tricky to determine analytically because there are multiple resources involved. I run a few test builds to find the -j value that yields the fastest build.