openscad/openscad #5318: Crash using right-click object information popup

MethylBromide, 2024-09-23:
**Describe the bug**
I right-click an object in the preview pane, then click on an entry in the resulting popup list. The client crashes. No error message observed.

**To Reproduce**
Steps to reproduce the behavior:
1. Install the BOSL2 library.
2. Enter the script shown below and execute it. This creates a model of a hex head bolt.
3. Right-click on the head of the bolt. This should give the screenshot shown below.
4. Left-click the final entry, "metric_bolt".
5. Program crashes.

**Expected behavior**
I wasn't sure what would happen but this wasn't it. I was thinking it might navigate to the relevant line of code.

**Code reproducing the issue**
```
$fa = 2; $fs = .3;
include <BOSL2/std.scad>
include <BOSL2/metric_screws.scad>
dPart = [40, 40, 20];

metric_bolt(headtype="hex", orient=UP, size=6, l=dPart.z, anchor="countersunk");
```

**Screenshots**
https://prnt.sc/TA_RVhh9JX5F

**Environment and Version info (please complete the following information):**
 - OS: Windows 11
 - System: Intel PC 64-bit
 - OpenSCAD Version 2024.09.17 (git 399551768)

**Library & Graphics card information**
```
OpenSCAD Version: 2024.09.17 (git 399551768)
System information: Microsoft Windows 11 (10.0.22631) x86_64 8 CPUs 7.78 GB RAM
User Agent: OpenSCAD/2024.09.17 (git 399551768) (Microsoft Windows 11 (10.0.22631) x86_64)
Compiler: GCC "13.2.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_81
Eigen version: 3.4.0
CGAL version, kernels: 5.5, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Qt version: 5.15.11
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.70.2
lodepng version: 20230410
libzip version: 1.5.2
fontconfig version: 2.14.2
freetype version: 2.13.2
harfbuzz version: 7.3.0
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: fast-csg, fast-csg-safer, fast-csg-debug, manifold*, roof*, input-driver-dbus, lazy-union, vertex-object-renderers-indexing, textmetrics*, import-function, predictible-output
Application Path: C:/Program Files/OpenSCAD
Documents Path: C:\Users\tyler\OneDrive\Documents
User Documents Path: C:\Users\tyler\OneDrive\Documents
Resource Path: C:/Program Files/OpenSCAD
User Library Path: C:/Users/tyler/OneDrive/Documents/OpenSCAD/libraries
User Config Path: C:\Users\tyler\AppData\Local/OpenSCAD
Backup Path: C:/Users/tyler/OneDrive/Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
C:/Users/tyler/OneDrive/Documents/OpenSCAD/libraries
C:/Program Files/OpenSCAD\libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
[... 230 lines cut ...]
GL_INTEL_coarse_fragment_shader
GL_INTEL_conservative_rasterization
GL_INTEL_fragment_shader_ordering
GL_INTEL_framebuffer_CMAA
GL_INTEL_map_texture
GL_INTEL_multi_rate_fragment_shader
GL_INTEL_performance_query
GL_KHR_blend_equation_advanced
GL_KHR_blend_equation_advanced_coherent
GL_KHR_context_flush_control
GL_KHR_debug
GL_KHR_no_error
GL_KHR_texture_compression_astc_ldr
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control

```

damienmarchal, 2024-10-03:
Hi @MethylBromide,

Thanks for reporting the problem with details. 
I will try to reproduce the bug ASAP.

damienmarchal, 2025-04-02:
Hi @MethylBromide, 

A fix has been merged recently the issue can probably be closed.

kintel, 2025-04-03:
Closing - please reopen if, for some reason, this didn't fix it.

closed 2025-04-03 by kintel