openscad/openscad #5319: Render with Manifold: transparent solids can look 'streaky'

hzeller, 2024-09-24:
I really like that Manifold renders retain the color information compared to CSG.

Of course, that results in new wishes, such as also rendering transparent objects nicely.

**Code**

Here a simple example with a difference of two disks:

```OpenSCAD
e=0.01;
color("#FFFFFF50") {
  difference() {
    cylinder(r=30, h=10);
    translate([0, 0, -e]) cylinder(r=5, h=10+2*e);
  }
}
```

With Manifold, they are rendered transparently, but they look 'streaky':

|   Preview  |  'Streaky' Render with Manifold |
-----------------|----------------------------------|
| [image] | [image] |

It looks like as if there are some triangle faces _inside_ the solid that modify the lighting. Maybe they are accidental triangles created by the `difference()` operation that should't be there ? Or it is working-as-intended, but still worthwhile to confirm by the developers.

**Environment and Version info (please complete the following information):**
 - OS: NixOS 
 -  System: Intel x86_64
 -  OpenSCAD Version 2024-08-17 based on
     * OpenSCAD git commit 1cf4e97ed488d606c823f107dcc361f218aa84ca
     * Manifold commit https://github.com/elalish/manifold/commit/22c66051dfdbcefa2012e30dd12c9b5a20f89a01
     
**Library & Graphics card information**
[library-info.txt](https://github.com/user-attachments/files/17117423/library-info.txt)

pca006132, 2024-09-24:
Oh, it may also be possible that we are getting wrong properties from vertex properties in manifold. Should dump those and check.

kintel, 2024-09-24:
I believe it's a renderer issue: We probably render using two-sided lighting without sorting the triangles back-to-front.

thehans, 2024-09-24:
The preview isn't drawing back faces at all.  Not sure that's necessarily desirable.
I vaguely remember looking into these sort of issues previously, but I think it was for PR which I never finalized.  #1485 

If it is desired for front and back faces are drawn, then I think a two pass approach is needed for translucent objects: drawing back faces first, then the front faces, in order for the blending to work out consistently.

thehans, 2024-09-24:
See also #1496

kintel, 2024-10-01:
Tagging as Manifold, not because it can be fixed in the manifold library, but because it's a bi-effect of Manifold outputting a "random" face order, exaggerating these artifacts.

hzeller, 2024-12-03:
In an animation it is easy to see that the location of the 'streaks' are not stable if looked at different directions which might support the not ordered back-to-front hypothesis https://youtu.be/yvGS4jcgmNw

jordanbrown0, 2026-09-21:
I suggest that because in 2021.01 render did not preserve color or transparency, what happens with Manifold is an improvement, not a regression, and so should not be considered a release blocker.  It's not ideal, but it's better than it was and it's not fatal.

jordanbrown0, 2026-09-21:
Demonstrated in 2026.08.29.

jordanbrown0, 2026-09-21:
@stefano2734, @pca006132 is one of the lead developers of Manifold.  If he says that it's probably not related to the Manifold version.... it is almost certainly not related to the Manifold version.

gsohler, 2026-09-21:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

especially for transparent solids its absolutely crucial to render them
from back to front.
Unfortunately this is very volatile depending on the viewing angle.
if we come up with a performant algorithm, to adapt the triangle order
on-the-fily, we can display transparent solids easily.

stefano2734, 2026-09-21:
Fine, not a release blocker, only an annoying  issue to solve in future.