openscad/openscad #5320: SIGSEGV with Minkowski operation using `manifold`

sesse, 2024-09-24:
This is similar to #5211, but with a different test case. I've simplified it down from a couple of hours to segfaulting in about five seconds; it is possible that it can be simplified even further, but it seems very sensitive to specific float values. To reproduce, run with

```
./OpenSCAD-2024.09.22.ai20484-x86_64.AppImage test.scad -o test.stl --enable manifold
```

The backtrace is not very informative:

```
#0  0x000004c9300d6e80 in ?? ()
#1  0x000055975ce3e9f5 in CGAL::Handle_for<CGAL::Nef_polyhedron_3_rep<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool>, std::allocator<CGAL::Nef_polyhedron_3_rep<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > >::~Handle_for() ()
#2  0x000055975cd36e6e in auto ManifoldUtils::applyMinkowskiManifold(std::__cxx11::list<std::pair<std::shared_ptr<AbstractNode const>, std::shared_ptr<Geometry const> >, std::allocator<std::pair<std::shared_ptr<AbstractNode const>, std::shared_ptr<Geometry const> > > > const&)::{lambda(auto:1 const&)#3}::operator()<std::shared_ptr<Geometry const> >(std::shared_ptr<Geometry const> const&) const [clone .isra.0] [clone .cold] ()
#3  0x000055975ce2fdeb in tbb::interface9::internal::start_for<tbb::blocked_range<__gnu_cxx::__normal_iterator<std::shared_ptr<Geometry const>*, std::vector<std::shared_ptr<Geometry const>, std::allocator<std::shared_ptr<Geometry const> > > > >, parallelizable_transform<__gnu_cxx::__normal_iterator<std::shared_ptr<Geometry const>*, std::vector<std::shared_ptr<Geometry const>, std::allocator<std::shared_ptr<Geometry const> > > >, __gnu_cxx::__normal_iterator<std::__cxx11::list<std::vector<CGAL::Point_3<CGAL::Epick>, std::allocator<CGAL::Point_3<CGAL::Epick> > >, std::allocator<std::vector<CGAL::Point_3<CGAL::Epick>, std::allocator<CGAL::Point_3<CGAL::Epick> > > > >*, std::vector<std::__cxx11::list<std::vector<CGAL::Point_3<CGAL::Epick>, std::allocator<CGAL::Point_3<CGAL::Epick> > >, std::allocator<std::vector<CGAL::Point_3<CGAL::Epick>, std::allocator<CGAL::Point_3<CGAL::Epick> > > > >, std::allocator<std::__cxx11::list<std::vector<CGAL::P--Type <RET> for more, q to quit, c to continue without paging--
tor<CGAL::Point_3<CGAL::Epick> > >, std::allocator<std::vector<CGAL::Point_3<CGAL::Epick>, std::allocator<CGAL::Point_3<CGAL::Epick> > > > > > > >, ManifoldUtils::applyMinkowskiManifold(std::__cxx11::list<std::pair<std::shared_ptr<AbstractNode const>, std::shared_ptr<Geometry const> >, std::allocator<std::pair<std::shared_ptr<AbstractNode const>, std::shared_ptr<Geometry const> > > > const&)::{lambda(auto:1 const&)#3}>(__gnu_cxx::__normal_iterator<std::shared_ptr<Geometry const>*, std::vector<std::shared_ptr<Geometry const>, std::allocator<std::shared_ptr<Geometry const> > > >, __gnu_cxx::__normal_iterator<std::shared_ptr<Geometry const>*, std::vector<std::shared_ptr<Geometry const>, std::allocator<std::shared_ptr<Geometry const> > > >, __gnu_cxx::__normal_iterator<std::__cxx11::list<std::vector<CGAL::Point_3<CGAL::Epick>, std::allocator<CGAL::Point_3<CGAL::Epick> > >, std::allocator<std::vector<CGAL::Point_3<CGAL::Epick>, std::allocator<CGAL::Point_3<CGAL::Epick> > > > >*, std::vector<std::__cxx11::list<std::vector<CGAL::Point_3<CGAL::Epick>, std::allocator<CGAL::Point_3<CGAL::Epick> > >, std::allocator<std::vector<CGAL::Point_3<CGAL::Epick>, std::allocator<CGAL::Point_3<CGAL::Epick> > > > >, std::allocator<std::__cxx11::list<std::vector<CGAL::Point_3<CGAL::Epick>, std::allocator<CGAL::Point_3<CGAL::Epick> > >, std::allocator<std::vector<CGAL::Point_3<CGAL::Epick>, std::allocator<CGAL::Point_3<CGAL::Epick> > > > > > > >, ManifoldUtils::applyMinkowskiManifold(std::__cxx11::list<std::pair<std::shared_ptr<AbstractNode const>, std::shared_ptr<Geometry const> >, std::allocator<std::pair<std::shared_ptr<AbstractNode const>, std::shared_ptr<Geometry const> > > > const&)::{lambda(auto:1 const&)#3} const&)::{lambda(auto:1)#1}, tbb::auto_partitioner const>::execute() ()
#
4  0x00007f656a5be545 in ?? ()
#5  0x0000000000000268 in ?? ()
#6  0x000004c92c990fd0 in ?? ()
#7  0x0000000000000000 in ?? ()
```

Valgrind is complaining, but I'm not sure if these are actually relevant to the crash:

```
==516050== Mismatched free() / delete / delete []
==516050==    at 0x48461EF: free (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==516050==    by 0x77506D4: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_mutate(unsigned long, unsigned long, char const*, unsigned long) (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.33)
==516050==    by 0x7751F3F: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_append(char const*, unsigned long) (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.33)
==516050==    by 0x26202D: CGAL::Failure_exception::Failure_exception(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) (in /tmp/.mount_OpenSCiSYy6I/usr/bin/openscad)
==516050==    by 0x262FA9: CGAL::Assertion_exception::Assertion_exception(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) (in /tmp/.mount_OpenSCiSYy6I/usr/bin/openscad)
==516050==    by 0x2631C1: CGAL::assertion_fail(char const*, char const*, int, char const*) (in /tmp/.mount_OpenSCiSYy6I/usr/bin/openscad)
==516050==    by 0x2633D4: CGAL::Interval_nt<false>::Test_runtime_rounding_modes::Test_runtime_rounding_modes() (in /tmp/.mount_OpenSCiSYy6I/usr/bin/openscad)
==516050==    by 0x22C334: _GLOBAL__sub_I_ManifoldGeometry.cc (in /tmp/.mount_OpenSCiSYy6I/usr/bin/openscad)
==516050==    by 0xCE7E7C: __libc_csu_init (in /tmp/.mount_OpenSCiSYy6I/usr/bin/openscad)
==516050==    by 0x79A1E4E: __libc_start_main@@GLIBC_2.34 (libc-start.c:343)
==516050==    by 0x24873D: (below main) (in /tmp/.mount_OpenSCiSYy6I/usr/bin/openscad)
==516050==  Address 0xad63d30 is 0 bytes inside a block of size 32 alloc'd
==516050==    at 0x4845263: operator new[](unsigned long) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==516050==    by 0x775065D: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_mutate(unsigned long, unsigned long, char const*, unsigned long) (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.33)
==516050==    by 0x7751F3F: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_append(char const*, unsigned long) (in /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.33)
==516050==    by 0x261F7D: CGAL::Failure_exception::Failure_exception(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) (in /tmp/.mount_OpenSCiSYy6I/usr/bin/openscad)
==516050==    by 0x262FA9: CGAL::Assertion_exception::Assertion_exception(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) (in /tmp/.mount_OpenSCiSYy6I/usr/bin/openscad)
==516050==    by 0x2631C1: CGAL::assertion_fail(char const*, char const*, int, char const*) (in /tmp/.mount_OpenSCiSYy6I/usr/bin/openscad)
==516050==    by 0x2633D4: CGAL::Interval_nt<false>::Test_runtime_rounding_modes::Test_runtime_rounding_modes() (in /tmp/.mount_OpenSCiSYy6I/usr/bin/openscad)
==516050==    by 0x22C334: _GLOBAL__sub_I_ManifoldGeometry.cc (in /tmp/.mount_OpenSCiSYy6I/usr/bin/openscad)
==516050==    by 0xCE7E7C: __libc_csu_init (in /tmp/.mount_OpenSCiSYy6I/usr/bin/openscad)
[... cut ...]
File: /usr/include/CGAL/Nef_3/SNC_external_structure.h
Line: 1150
```

Here's the actual segfaulting geometry:

```
[... 26 lines cut ...]
    }
  }
  cube(0.8);
}

function f(x) = 2.29736e-11*(x+11.5305)^6.41577;

module s(y1, y2, x1, x2) {
  translate([0, 0, 27.92-y2])
     linear_extrude(y2-y1, scale=(x1+54.5)/(x2+54.5))
       scale((x2+54.5)/119)
         t(221, 125);
}

module t(x, y) {
  hull() {
    for (i=[0:1]) {
      for (j=[0:1]) {
        translate([y/2-y*i,y/2-y*j,0])
          circle(r=(x - y)/2);
      }
    }
  }
}
```

kintel, 2024-09-25:
When I run this (on macOS), I get this on stderr:
```
CGAL error: assertion violation!
Expression : 
File       : /Users/kintel/code/OpenSCAD/libraries/install/include/CGAL/Nef_3/SNC_const_decorator.h
Line       : 276
Explanation: it is not possible to decide which one is a visible facet (if any)
```

..then it crashes in the destructor you mentioned, while unwinding the stack for the exception.
Sounds a bit like https://github.com/CGAL/cgal/issues/422, and potentially https://github.com/CGAL/cgal/issues/7271

I'll see if I can easily build against CGAL master and see if that helps.

kintel, 2024-09-25:
It looks like the problem boils down to that, if `CGAL::convex_decomposition_3()` throws, it leaves the object it's working on in a bad state, causing its destructor to crash in a construct like this:
```
CGAL_Nef_polyhedron3 decomposed_nef(*poly);
CGAL::convex_decomposition_3(decomposed_nef);
``

I tested CGAL master, and while it outputs the same error, it doesn't crash and seems to output a correct geometry.

sesse, 2024-09-25:
Let me know when it's reached the right builds, and I'll test with the non-minified example :-)

sesse, 2024-09-25:
FWIW, throwing exceptions from a C++ destructor is pretty much broken in general :-) You have zero guarantees from the standard library if that happens, for instance (and it doesn't make sense semantically either, how could possibly a destructor fail?).

kintel, 2024-09-25:
In this case, it's throwing an exception from a regular function, but some destructor crashes during stack unwinding due to a nullptr dereference. We won't make a build until the next CGAL release, but I'll keep my eyes open.

lgarron, 2024-11-18:
I've been on an OpenSCAD dev snapshot for a while, and the app crashes for me constantly on macOS. I live with it because CGAL is unusably slow for my projects, but I've started trying to look into the issue.

[This file](https://github.com/lgarron/cube-box/blob/0d0dbf015654a889a01c3ca34c1f66941e329e6c/cube-box-v0.1.4.scad) has an interesting effect:

- When the app is set to Manifold and the app is opened by clicking on the file, it renders just fine for about a dozen times until it crashes arbitrarily. I have yet to catch any pattern about what causes a crash vs. not.
- If I open the app by running `openscad` on the commandline and then opening the file in the app, it always crashes immediately.
- (Setting the app to CGAL doesn't crash but makes the app hang for a very long time because CGAL is slow.)

Here's the output when using `openscad` from the commandline:

```
╭─── /Users/lgarron ───────────────────────────────────────────────────
├─ lgarron@Germain
│ openscad
┴
Could not initialize localization (application path is '/opt/homebrew/bin').
qt.qpa.fonts: Populating font family aliases took 129 ms. Replace uses of missing font family "DejaVu Sans" with one that exists to avoid this cost.
Error reading examples.json: examples.json: cannot open file
WARNING: GLView: unknown colorscheme Tomorrow Night
ERROR: Cannot open shader source file: 'MouseSelector.vert'
ERROR: Cannot open shader source file: 'MouseSelector.frag'
/Users/distiller/project/src/gui/MouseSelector.cc: OpenGL Program Linker Error:
ERROR: Compiled vertex shader was corrupt.
ERROR: Compiled fragment shader was corrupt.

/Users/distiller/project/src/gui/MouseSelector.cc: OpenGL symbol retrieval went wrong, id is -1

FALLBACK (log once): Fallback to SW vertex for line stipple
FALLBACK (log once): Fallback to SW vertex processing, m_disable_code: 2000
FALLBACK (log once): Fallback to SW vertex processing in drawCore, m_disable_code: 2000
2024-11-17 18:46:30.413 openscad[90527:911189] +[IMKClient subclass]: chose IMKClient_Modern
2024-11-17 18:46:30.413 openscad[90527:911189] +[IMKInputSession subclass]: chose IMKInputSession_Modern

… (I open the file at this point.)

fish: Job 1, 'openscad' terminated by signal SIGSEGV (Address boundary error)
┬
├─ ❌ [SIGSEGV] command status
╰─── ⏱️ 3.857s ────────────────────────────────────────────────────────
```

kintel, 2024-11-18:
@lgarron this looks like a completely different issue - could you open a new issue for it?

stefano2734, 2026-09-20:
Actual available CGAL 6.2.1. last tests here with 6.0 or before.
And actual manifold is here in nightly implemented.