openscad/openscad #5324: difference with offset polygon cause preview error

UBaer21, 2024-09-25:
```
difference() {
  square(10);
  offset(1)difference() {
      square(21,true);
      square(10);
    }
}
```
[image]

version 2024.09.23 (also in older)

kintel, 2024-09-25:
A quick workaround for now is to wrap 2D nodes which evaluate geometry (like offset and text) with `render(convexity=2)`:
```
difference() { // or intersection()
  square(10);
  render(convexity=2) offset(1) difference() {
      square(21,true);
      square(10);
    }
}
```

jordanbrown0, 2024-09-25:
We really should render 2D objects before displaying them (as we render them before, e.g. linear extrusion), and then display them as zero-thickness objects (like incomplete polyhedra).  We shouldn't turn them into 1-unit-thick simulated 3D objects and then use the GL-based preview mechanisms.

jordanbrown0, 2026-05-07:
We either need these 2D modules to have `convexity` parameters, or we need to do something better to prevent convexity artifacts.

MichaelAtOz, 2026-05-10:
Presuming convexity in 2D would not be resource intensive, just use a 'big-ish' convexity internally??