openscad/openscad #5329: 3D Colour Export in .OFF using multiple primitives.

BK524, 2024-09-26:
Hi guys,
I'm doing some spacecraft modelling using OpenSCAD. I need to export in .off file format so I can maintain colour/material information in the mesh. This is a basic example of what I've been doing:

distance=0.71/2+1.21/2;
color("red") { rotate([ 0, 90, 0 ])
cylinder(d1=0.71, d2=0.71, h=3.2, center=true, $fn = 8);
}
translate([0,-distance+0.001,0])
rotate([0,0,90])
cube([1.21,3.199,0.01],center=true);
translate([0,distance-0.001,0])
rotate([0,0,90])
cube([1.21,3.199,0.01],center=true);

The issue is the colour is lost during render when I use more than one primitive (2 in this case). If I removed the Cube primitive and just rendered the cylinder it renders as expected but as soon as another primitive is added, the colour is lost.

Any Ideas?
Cheers

t-paul, 2024-09-26:
That feature is currently work in progress, but you match exactly the first step already done. Use the [latest snapshot version](https://openscad.org/downloads.html#snapshots), this supports render colors and export via OFF (only OFF at this point! more to come).

closed 2024-09-26 by t-paul

BK524, 2024-09-26:
@t-paul Does the most recent snapshot fix the issue when more than one primitive are combined?
Cheers

t-paul, 2024-09-26:
There is no fix, colors in render mode are a completely new feature via #5185, also upcoming in #5288 for exporting to PovRay (not yet in the snapshot builds) but you can see some color export examples there.

I'd recommend trying it out and open a specific issue with example and screenshot of the imported OFF (as OFF is not that widely supported, not everyone can easily test that).

kintel, 2024-09-26:
Are you sure this is not just a bug in your design? If you move your closing brace (`}`) from line 4 to the end of your file, does it do what you expect?