openscad/openscad #5330: Colour removed after render.

BK524, 2024-09-26:
Opening this as a separate issue as recommended by @t-paul 
Looking to take advantage of the new colour render capability to export spacecraft .scad models into (preferably) .obj or .off format. Used the newest developmental snapshot. 
I have created this basic example with one part coloured and the rest left uncoloured, this works fine in preview mode (see first image)
[image]
But when rendered this colour disappears (see second image):
[image]

The code is as follows:
distance=0.71/2+1.21/2;
color("red") { rotate([ 0, 90, 0 ])
cylinder(d1=0.71, d2=0.71, h=3.2, center=true, $fn = 8);
}
translate([0,-distance+0.001,0])
rotate([0,0,90])
cube([1.21,3.2,0.01],center=true);
translate([0,distance-0.001,0])
rotate([0,0,90])
cube([1.21,3.2,0.01],center=true);

Added the .off file exported here: [OFF file ex.docx](https://github.com/user-attachments/files/17151835/OFF.file.ex.docx)
Any ideas what is wrong?
Cheers

UBaer21, 2024-09-26:
Manifold is activated?  (preferences ↦ features ↦ manifold)

t-paul, 2024-09-26:
Yes, I was looking for a dedicated color switch but there's none. But "manifold" needs to be enabled for this to work. Sorry I missed that when closing the other issue. I tried locally and it works, but I have "manifold" enabled already.

BK524, 2024-09-26:
@UBaer21 @t-paul I hadn't clicked to activate manifold (silly me!)
I think it's working now.
Thanks for your help :)

closed 2024-09-26 by thehans