openscad/openscad #5338: Producing only triangular mesh elements when exporting into .off or .obj

BK524, 2024-09-29:
Hey guys,
Is there a way in OpenSCAD where we can specify that we only want triangular mesh elements? My program requires triangular mesh elements, so it would be great if I can restrict exports to that type. 
Cheers

kintel, 2024-09-29:
When using the manifold backend, all exported primitives should be triangles. The CGAL backend may yield polygons.

We're not planning to add more features to the CGAL backend.

closed 2024-09-29 by kintel

BK524, 2024-09-29:
I have manifold enabled in my preferences. I created a simple cube as an example, as shown below: 
[image]
I export that into OFF format and I get this:
OFF 8 6 0
-5 -5 -5 
5 -5 -5 
-5 5 -5 
5 5 -5 
-5 -5 5 
5 -5 5 
-5 5 5 
5 5 5 
4 4 5 7 6 255 0 0
4 2 3 1 0 255 0 0
4 0 1 5 4 255 0 0
4 1 3 7 5 255 0 0
4 3 2 6 7 255 0 0
4 2 0 4 6 255 0 0

t-paul, 2024-09-29:
That due to no geometry processing at all, and the cube primitive is created with quads.

kintel, 2024-09-29:
Right, that's an exception. We could consider creating internal primitives with triangles.

BK524, 2024-09-29:
Ahh ok, I see. I'm only ever going to be using examples with more than one primitive in my actual work anyway, I was just concerned when I got these from some basic tests. Thanks guys :)