openscad/openscad #5350: Bug in Large Polyhedrons

tgrundtvig, 2024-10-02:
**Describe the bug**
When rendering large Polyhedrons sometimes strange things happens to the polyhedrons.
In this example I have created two polyhedrons (generated with JavaCSG). They are in two separate modules and when they are rendered alone each of them works fine. But when they are rendered together, one of them changes to be the "hull" of the polyhedron.
The code is in the provided file:
[OpenSCAD_BUG.txt](https://github.com/user-attachments/files/17231330/OpenSCAD_BUG.txt)

**To Reproduce**
Steps to reproduce the behavior:
1. Download https://github.com/user-attachments/files/17231330/OpenSCAD_BUG.txt and rename file to "BUG.scad"
2. Open the file in OpenSCAD.
3. Press "render"
4. See error
5. Comment out first line.
6. Save
7. Press "Render"
8. Everything seems to be fine.

**Expected behavior**
I expected the two polyhedrons to render like they are shown in the preview.

**Code reproducing the issue**
In the file: (https://github.com/user-attachments/files/17231330/OpenSCAD_BUG.txt), File extension must be renamed to ".scad".

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info (please complete the following information):**
 - OS: Windows 11
 - System: Intel PC 64-bit
 - OpenSCAD Version: 2021.01

**Library & Graphics card information**
OpenSCAD Version: 2021.01
System information: Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833 20 CPUs 31.63 GB RAM
User Agent: OpenSCAD/2021.01 (Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833)
Compiler: GCC "7.4.0" 64bit
MinGW build: MingW64
Debug build: No
Boost version: 1_60
Eigen version: 3.3.7
CGAL version, kernels: 4.11, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.13.0
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.50.2
lodepng version: 20180910
libzip version: 1.5.2
fontconfig version: 2.13.1
freetype version: 2.10.1
harfbuzz version: 2.4.0
cairo version: 1.16.0
lib3mf version: 1.8.0
Application Path: C:/Program Files/OpenSCAD
Documents Path: C:\Users\tgrun\OneDrive\Documents
User Documents Path: C:\Users\tgrun\OneDrive\Documents
Resource Path: C:/Program Files/OpenSCAD
User Library Path: C:/Users/tgrun/OneDrive/Documents/OpenSCAD/libraries
User Config Path: C:\Users\tgrun\AppData\Local/OpenSCAD
Backup Path: C:/Users/tgrun/OneDrive/Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
C:/Users/tgrun/OneDrive/Documents/OpenSCAD/libraries
C:/Program Files/OpenSCAD\libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
C:/Windows/fonts
C:/Users/tgrun/.local/share/fonts
C:/Users/tgrun/.fonts
[... 233 lines cut ...]
GL_KHR_no_error
GL_KHR_shader_subgroup
GL_KHR_shader_subgroup_arithmetic
GL_KHR_shader_subgroup_ballot
GL_KHR_shader_subgroup_basic
GL_KHR_shader_subgroup_clustered
GL_KHR_shader_subgroup_quad
GL_KHR_shader_subgroup_shuffle
GL_KHR_shader_subgroup_shuffle_relative
GL_KHR_shader_subgroup_vote
GL_KHR_texture_compression_astc_ldr
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_OVR_multiview
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control

**Additional context**
The problem persists if I translate one polyhedra, so they do not touch each other.

jordanbrown0, 2024-10-02:
A cube(1) plus either of the polyhedra triggers the problem, so one of them can be eliminated.

jordanbrown0, 2024-10-02:
Fails only for CGAL.

jordanbrown0, 2024-10-02:
Not for Manifold, that is.

jordanbrown0, 2024-10-02:
Here's closer to a minimal demo.  Because the curvature is very small at this scale, I've set the $vp variables to focus in tightly on one edge.  Switching back and forth between F5 and F6 (or enabling and disabling the cube) will cause that edge to move a little.  Use orthogonal view.

Changing almost any of the coordinates here will change the behavior, either making it stop misbehaving or turning it into a CGAL error.

For instance, note that we're looking at the minimum-X, minimum-Z edge.  Changing the Z=9.1 in the second point - changing the *top* of the figure - will make it stop misbehaving.

Changing the X=1.9991 in the fourth point to 1.999 will trigger a CGAL error.

```
$vpt = [2.42, 1.31, 2.93 ];
$vpr = [83.7, 0, 220.5];
$vpd = 0.13;

cube(1);
M3806();
points = [
    [2, 0, 3],
    [2, 0, 9.1],
    [7, 0, 9.1],
    [7, 0, 3],
    [1.9991, 1, 3],
    [1.9991, 1, 9.1],
    [6.9991, 1, 9.1],
    [6.9991, 1, 3],
    [1.9966, 2, 3],
    [1.9966, 2, 9.1],
    [6.9966, 2, 9.1],
    [6.9966, 2, 3],
    [1.9924, 3, 3],
    [1.9924, 3, 9.1],
    [6.9923, 3, 9.1],
    [6.9923, 3, 3],
    [1.9865, 4, 3],
    [1.9865, 4, 9.1],
    [6.986, 4, 9.1],
    [6.986, 4, 3],
];

module M3806()
{
    polyhedron
    (
        points = points,
        faces =
        [
            [0, 1, 2, 3], 
            [0, 4, 5, 1], 
            [1, 5, 6, 2], 
            [2, 6, 7, 3], 
            [3, 7, 4, 0], 
            [4, 8, 9, 5], 
            [5, 9, 10, 6], 
            [6, 10, 11, 7], 
            [7, 11, 8, 4], 
            [8, 12, 13, 9], 
            [9, 13, 14, 10], 
            [10, 14, 15, 11], 
            [11, 15, 12, 8], 
            [12, 16, 17, 13], 
            [13, 17, 18, 14], 
            [14, 18, 19, 15], 
            [15, 19, 16, 12],
            [19, 18, 17, 16]
        ]
    );
}
```

That about reaches the limits of my diagnostic ability.  The fact that this problem doesn't manifest with Manifold might be reason to not worry about it.

tgrundtvig, 2024-10-04:
Thank you for your help. Everything works after I downloaded the nightly build instead of the stable version and enabled manifold :-)
But now I want to enable manifold from the command line. I found some references to the command line option:
--enable=manifold
but this does not seem to work. Anyone know how to enable manifold from the command line?

nophead, 2024-10-04:
No equals.  ```--enable manifold```

kintel, 2024-10-04:
`--backend manifold` or `--backend=manifold`

Manifold is no longer experimental.

tgrundtvig, 2024-10-05:
Thanks, that works :-)