openscad/openscad #5352: OpenSCAD crashes on elementaryOS 7.1

spotlesscoder, 2024-10-03:
**Describe the bug**
Got this error message

```
./OpenSCAD-2021.01-x86_64.AppImage 
QApplication: invalid style override 'adwaita' passed, ignoring it.
	Available styles: Windows, Fusion

(AppRun.wrapped:477612): Gtk-WARNING **: 11:50:44.968: Could not load a pixbuf from icon theme.
This may indicate that pixbuf loaders or the mime database could not be found.
**
Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/elementary/status/16/image-missing.svg: Unable to load image-loading module: /usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-svg.so: /lib/x86_64-linux-gnu/librsvg-2.so.2: undefined symbol: cairo_tag_end (gdk-pixbuf-error-quark, 5)
Bail out! Gtk:ERROR:../../../../gtk/gtkiconhelper.c:494:ensure_surface_for_gicon: assertion failed (error == NULL): Failed to load /usr/share/icons/elementary/status/16/image-missing.svg: Unable to load image-loading module: /usr/lib/x86_64-linux-gnu/gdk-pixbuf-2.0/2.10.0/loaders/libpixbufloader-svg.so: /lib/x86_64-linux-gnu/librsvg-2.so.2: undefined symbol: cairo_tag_end (gdk-pixbuf-error-quark, 5)
Aborted
```

**To Reproduce**
Steps to reproduce the behavior:
1. Start openSCAD
2. Click the green "open" button

**Expected behavior**
File open dialog appears

**Code reproducing the issue**
If applicable, add trimmed down, self-contained code demonstrating the issue.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info (please complete the following information):**
 - OS: elementaryOS 7.1 (all updates installed)
 - System: Thinkpad X270
 - OpenSCAD Version: 

**Library & Graphics card information**
```
QObject::startTimer: Timers can only be used with threads started with QThread
OpenSCAD Version: 2021.01 (git 41f58fe)
System information: Linux 6.8.0-45-generic #45~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Sep 11 15:25:05 UTC 2 x86_64 elementary OS 7.1 Horus 4 CPUs 15.37 GB RAM
User Agent: OpenSCAD/2021.01 (git 41f58fe) (Linux x86_64; elementary OS 7.1 Horus)
Compiler: GCC "5.4.0 20160609" 64bit
MinGW build: No
Debug build: No
Boost version: 1_58
Eigen version: 3.2.92
CGAL version, kernels: 4.11, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.14.2
QScintilla version: 2.9.1
InputDrivers: 
GLib version: 2.48.2
lodepng version: 20180910
libzip version: 1.0.1
fontconfig version: 2.11.94 (runtime: 2.13.1)
freetype version: 2.6.1 (runtime: 2.11.1)
harfbuzz version: 1.0.1 (runtime: 2.7.4)
cairo version: 1.14.6
lib3mf version: 1.8.1
Application Path: /tmp/.mount_OpenSCMOz9dU/usr/bin
Documents Path: /home/user/.local/share
User Documents Path: /home/user/Documents
Resource Path: /tmp/.mount_OpenSCMOz9dU/usr/share/openscad
User Library Path: /home/user/.local/share/OpenSCAD/libraries
User Config Path: /home/user/.config/OpenSCAD
Backup Path: /home/user/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /home/user/.local/share/OpenSCAD/libraries
  /tmp/.mount_OpenSCMOz9dU/usr/share/openscad/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  /usr/share/fonts
  /usr/local/share/fonts
  /home/user/.local/share/fonts
  /home/user/.fonts
  /usr/X11/lib/X11/fonts
  /System/Library/Fonts
  /Library/Fonts
  /home/user/Library/Fonts
  /usr/share/fonts/X11
  /usr/share/fonts/cMap
  /usr/share/fonts/cmap
  /usr/share/fonts/opentype
  /usr/share/fonts/truetype
  /usr/share/fonts/type1
  /usr/share/fonts/woff
  /usr/share/fonts/X11/Type1
  /usr/share/fonts/X11/encodings
  /usr/share/fonts/X11/misc
  /usr/share/fonts/X11/util
  /usr/share/fonts/cmap/adobe-cns1
  /usr/share/fonts/cmap/adobe-gb1
  /usr/share/fonts/cmap/adobe-japan1
  /usr/share/fonts/cmap/adobe-japan2
  /usr/share/fonts/cmap/adobe-korea1
  /usr/share/fonts/opentype/font-awesome
  /usr/share/fonts/opentype/inter
  /usr/share/fonts/opentype/linux-libertine
  /usr/share/fonts/opentype/mondulkiri
  /usr/share/fonts/opentype/noto
  /usr/share/fonts/opentype/opendyslexic
  /usr/share/fonts/opentype/urw-base35
  /usr/share/fonts/truetype/arphic
  /usr/share/fonts/truetype/capture-it-elementary
  /usr/share/fonts/truetype/croscore
  /usr/share/fonts/truetype/crosextra
  /usr/share/fonts/truetype/daniel-elementary
  /usr/share/fonts/truetype/dejavu
  /usr/share/fonts/truetype/droid
  /usr/share/fonts/truetype/font-awesome
  /usr/share/fonts/truetype/freefont
  /usr/share/fonts/truetype/gentium
  /usr/share/fonts/truetype/gentium-basic
  /usr/share/fonts/truetype/hvd-bodedo-elementary
  /usr/share/fonts/truetype/jenna-sue-elementary
  /usr/share/fonts/truetype/kacst
  /usr/share/fonts/truetype/kacst-one
  /usr/share/fonts/truetype/liberation
  /usr/share/fonts/truetype/liberation2
  /usr/share/fonts/truetype/libreoffice
  /usr/share/fonts/truetype/limelight-elementary
  /usr/share/fonts/truetype/lobster-elementary
  /usr/share/fonts/truetype/noto
  /usr/share/fonts/truetype/open-sans
  /usr/share/fonts/truetype/operating-instructions-elementary
  /usr/share/fonts/truetype/padauk
  /usr/share/fonts/truetype/plainblack-elementary
  /usr/share/fonts/truetype/raleway-elementary
  /usr/share/fonts/truetype/redacted-elementary
  /usr/share/fonts/truetype/roboto
  /usr/share/fonts/truetype/roboto-mono-elementary
  /usr/share/fonts/truetype/ubuntu
  /usr/share/fonts/type1/gsfonts
  /usr/share/fonts/type1/urw-base35
  /usr/share/fonts/woff/opendyslexic
  /usr/share/fonts/X11/encodings/large
  /usr/share/fonts/truetype/roboto/unhinted
  /usr/share/fonts/truetype/roboto/unhinted/RobotoTTF

GLEW version: 1.13.0
OpenGL Version: 4.6 (Compatibility Profile) Mesa 23.2.1-1ubuntu3.1~22.04.2
GL Renderer: Mesa Intel(R) HD Graphics 520 (SKL GT2)
GL Vendor: Intel
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
GL context creator: GLX
PNG generator: lodepng
GLX version: 1.4
OS info: Linux 6.8.0-45-generic #45~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Sep 11 15:25:05 UTC 2
Machine: x86_64
user@ThinkPad-X27
```

**Additional context**
Add any other context about the problem here.

pca006132, 2024-10-03:
Can you try the latest development snapshot?

hdfx, 2024-10-03:
sure, where can I download?

spotlesscoder, 2024-10-03:
sure, where can I download?

pca006132, 2024-10-03:
https://openscad.org/downloads.html#snapshots

spotlesscoder, 2024-10-03:
With ./OpenSCAD-2024.10.02.ai20769-x86_64.AppImage I had no crashes :+1:

spotlesscoder, 2024-10-03:
When will a new release be created? 
The one I downloaded first seems to be around 3 years old already

pca006132, 2024-10-03:
See https://github.com/orgs/openscad/projects/2

closed 2024-10-03 by spotlesscoder