openscad/openscad #5353: Differences between preview and render

Dapid, 2024-10-03:
**Describe the bug**
I am making some hollow cylinders, but when rendering one of the inner faces is flattened.

[image]

**To Reproduce**
Run the attached code. The problem is present in both CGAL and Manifold.

**Expected behavior**
The preview is correct:
[image]

**Code reproducing the issue**
Here is the code, annotating how some changes fix the problem:

```
n_sides = 1024 ; // Changing this to 512 works
$fn = n_sides;
inf = 1000; // Changing this to 100 works

module ground_plane() {
	translate([(-inf / 2), (-inf / 2), -inf]) cube([inf, inf, inf]);
}
module everything() {
    ground_plane();
    //union() // Commenting out this works if lazy-union is disabled; present works if lazy-union is enabled.
    {
	for(pos = [h3_position, h2_position]) translate(pos) difference() { // adding only one hole works
		cylinder(h = supports_height, r = supports_radius, $fn = n_sides);
		translate([0, 0, (supports_height - supports_hole_depth)]) cylinder(h = supports_hole_depth, r = supports_hole_radius,);
		translate([0, 0, supports_height]) cylinder(r = (supports_radius * 1.5), h = 0.5, center = true);
	}
    }
}

sink_depth = 1;
h3_position = [10, 40, -sink_depth];
h2_position = [h3_position[0], (h3_position[1] - 15), h3_position[2]];
supports_height = (14 + sink_depth);
supports_radius = 3;
supports_hole_radius = (3 / 2);
supports_hole_depth = 6;

render() everything();
```

**Library & Graphics card information**

Openscad installed through snap.

```
$ openscad-nightly --info
QObject::startTimer: Timers can only be used with threads started with QThread
OpenSCAD Version: 2024.09.10
System information: Linux 6.10.11-200.fc40.x86_64 #1 SMP PREEMPT_DYNAMIC Wed Sep 18 21:09:58 UTC 2024 x86_64 Ubuntu Core 20 8 CPUs 62.72 GB RAM
User Agent: OpenSCAD/2024.09.10 (Linux x86_64; Ubuntu Core 20)
Compiler: Clang "10.0.0 " 64bit
MinGW build: No
Debug build: No
Boost version: 1_71
Eigen version: 3.3.7
CGAL version, kernels: 5.5.2, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.1
Qt version: 5.12.8
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.64.6
lodepng version: 20230410
libzip version: 1.5.1
fontconfig version: 2.13.1
freetype version: 2.10.1
harfbuzz version: 2.6.4
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: fast-csg, fast-csg-safer, fast-csg-debug, manifold, roof, input-driver-dbus, lazy-union, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output
Application Path: /snap/openscad-nightly/4550/usr/bin
Documents Path: /home/david/snap/openscad-nightly/4550/.local/share
User Documents Path: /home/david/snap/openscad-nightly/4550
Resource Path: /snap/openscad-nightly/4550/usr/share/openscad-nightly
User Library Path: /home/david/snap/openscad-nightly/4550/.local/share/OpenSCAD/libraries
User Config Path: /home/david/snap/openscad-nightly/4550/.config/OpenSCAD
Backup Path: /home/david/snap/openscad-nightly/4550/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /home/david/snap/openscad-nightly/4550/.local/share/OpenSCAD/libraries
  /snap/openscad-nightly/4550/usr/share/openscad-nightly/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  /usr/share/fonts
  /usr/local/share/fonts
  /home/david/snap/openscad-nightly/4550/.local/share/fonts
  /home/david/snap/openscad-nightly/4550/.fonts
  /home/david/.fonts
  /snap/openscad-nightly/4550/usr/share/fonts
  /usr/X11/lib/X11/fonts
  /System/Library/Fonts
  /Library/Fonts
  /home/david/snap/openscad-nightly/4550/Library/Fonts
  /usr/share/fonts/aajohan-comfortaa-fonts
  /usr/share/fonts/abattis-cantarell-fonts
  /usr/share/fonts/abattis-cantarell-vf-fonts
  /usr/share/fonts/adf-gillius-2-fonts
  /usr/share/fonts/adf-gillius-fonts
  /usr/share/fonts/adobe-source-code-pro-fonts
  /usr/share/fonts/dejavu-sans-fonts
  /usr/share/fonts/dejavu-sans-mono-fonts
  /usr/share/fonts/dejavu-serif-fonts
  /usr/share/fonts/gdouros-symbola
  /usr/share/fonts/gnu-free
  /usr/share/fonts/google-carlito-fonts
  /usr/share/fonts/google-crosextra-caladea-fonts
  /usr/share/fonts/google-droid-sans-fonts
  /usr/share/fonts/google-noto
  /usr/share/fonts/google-noto-color-emoji-fonts
  /usr/share/fonts/google-noto-sans-cjk-vf-fonts
  /usr/share/fonts/google-noto-sans-mono-cjk-vf-fonts
  /usr/share/fonts/google-noto-serif-cjk-vf-fonts
  /usr/share/fonts/google-noto-vf
  /usr/share/fonts/jomolhari-fonts
  /usr/share/fonts/liberation-mono
  /usr/share/fonts/liberation-narrow
  /usr/share/fonts/liberation-sans
  /usr/share/fonts/liberation-serif
  /usr/share/fonts/lohit-assamese-fonts
  /usr/share/fonts/lohit-bengali-fonts
  /usr/share/fonts/lohit-devanagari-fonts
  /usr/share/fonts/lohit-gujarati-fonts
  /usr/share/fonts/lohit-kannada-fonts
  /usr/share/fonts/lohit-marathi-fonts
  /usr/share/fonts/lohit-odia-fonts
  /usr/share/fonts/lohit-tamil-fonts
  /usr/share/fonts/lohit-telugu-fonts
  /usr/share/fonts/madan-fonts
  /usr/share/fonts/mathjax
  /usr/share/fonts/open-sans
  /usr/share/fonts/opensymbol
  /usr/share/fonts/paktype-naskh-basic-fonts
  /usr/share/fonts/rit-meera-new-fonts
  /usr/share/fonts/rit-rachana-fonts
  /usr/share/fonts/sil-mingzat-fonts
  /usr/share/fonts/sil-nuosu-fonts
  /usr/share/fonts/sil-padauk-fonts
  /usr/share/fonts/stix-fonts
  /usr/share/fonts/urw-base35
  /usr/share/fonts/vazirmatn-vf-fonts
  /usr/share/fonts/wqy-microhei-fonts
  /usr/share/fonts/xscreensaver
  /home/david/.fonts/Bowerbird Display
  /home/david/.fonts/From the Internet
  /home/david/.fonts/Frutiger
  /home/david/.fonts/Rajdhani
  /snap/openscad-nightly/4550/usr/share/fonts/truetype
  /home/david/.fonts/Frutiger/Frutiger LT Std
  /snap/openscad-nightly/4550/usr/share/fonts/truetype/dejavu
  /snap/openscad-nightly/4550/usr/share/fonts/truetype/ubuntu

GL context creator: GLX (new)
GLX version: 1.4
PNG generator: lodepng

GLEW version: 2.1.0
OpenGL Version: 4.6.0 NVIDIA 560.35.03
GL Renderer: NVIDIA GeForce RTX 2070/PCIe/SSE2
GL Vendor: NVIDIA Corporation
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: ye
```

kintel, 2024-10-03:
This looks like a floating point precision issue, triggered by the large plane. We increased precision to double around 2024.09.19, so try a more recent snapshot

UBaer21, 2024-10-03:
can't reproduce with 2024.09.30 besides $fn 1024 is quite high.

Dapid, 2024-10-07:
Thanks. I think we can close it now.

My actual model is actually quite complex, but the floating point errors only appeared at the very end, once I added those holes.

closed 2024-10-07 by Dapid