openscad/openscad #5355: OpenSCAD SVG export sizes / coordinate system are wrong for projection of linear_extrude

dgatwood, 2024-10-04:
**Describe the bug**

On this model, OpenSCAD exports the SVG at about 12% larger than the expected size.  Simple geometric shapes work as expected, so this is specific to some feature in the model itself.

**To Reproduce**

1. Export the model below.

**Expected behavior**

The model is 24 units tall along the y axis, with the values in the linear_extrude call ranging from -12 to 12.  I expected the exported SVG to also go from -12 to +12 on the y axis.  Instead, it goes from -13.44 to +13.44.  The X range should be from 10 to 20.6121.  Instead, the range is from 11.2 to 20.6121.

So the portion of the cylinder that is shown has the correct dimensions, but the linear_extrude coordinate system is very wrong, resulting in too much of the cylinder remaining in both directions.  The only way I know to fix this is to hand edit the SVG.

**Code reproducing the issue**

projection(cut=true)  // Change to true for generating the other set of cuts.
    union() {
        difference() {
            cylinder(h=8, r = 20.6121, center=true, $fn=1000);
            translate([0, 0, -2]) linear_extrude(height = 20, center = true, scale=1.2) polygon(points = [ [15, -4], [15, 4], [10, 12], [40, 12], [10, 40], [-23, 12], [-23, -12], [10, -30], [40, -12], [10, -12]], convexity = 4);
        };
    }

**Environment and Version info (please complete the following information):**
 - OS: macOS 14 Sonoma
 - System: M1
 - OpenSCAD Version 2021.01 and 2024.09.30 (git 296b86735)

**Library & Graphics card information**
N/A

**Additional context**
N/A

UBaer21, 2024-10-04:
Maybe you importing add some scaling. 
If you open the SVG with an editor you find.
`<path d="
M 20.6121,-0 L 20.6117,-0…`
which is exactly the radius of your 20.6121 circle
You are using projection(cut=true) which will be the profile at Z=0 and not a projection of the full 3D Object but just a slice.
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/3D_to_2D_Projection

jordanbrown0, 2024-10-04:
Right, what Ulrich said.  Because of the scale, your linear extrusion is 24 units tall at the bottom and 28.8 units tall at the top.  Put a # in front of the linear_extrude and then rotate to look at the side view:

[image]

closed 2024-10-04 by jordanbrown0