openscad/openscad #5358: FAST-CSG is no longer fast starting with version 20240929

coder2999, 2024-10-07:
When I press F6, it outputs multiple lines of the same text:
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices.

It takes more time to complete the task compared to the old version.

What does it actually mean? How can I solve this problem?

coder2999, 2024-10-07:
I found that disable "fast-cfg-safer" can solve this problem.

coder2999, 2024-10-07:
I tested it in another project, I found that even if I disabled "fast-csg-safer", it still takes more time to complete the task compared to the old version.

t-paul, 2024-10-07:
fast-csg is obsolete, please use Manifold instead, see https://lists.openscad.org/empathy/thread/D6KV3ZLXHLBHSITSQ5GPUZUKHURU4ABE

closed 2024-10-07 by t-paul