openscad/openscad #5361: Crashing randomly and frequently

JeffreyOI, 2024-10-08:
**Describe the bug**
This is some kind of UI issue I'm guessing because the app will crash as I'm typing. Sometimes immediately, sometimes after a line. QoL breaking.

**To Reproduce**
1. Open OpenScad.
2. Begin composing code.

**Expected behavior**
No crashing 

**Code reproducing the issue**

**Screenshots**
https://github.com/user-attachments/assets/f4317ffe-8848-4f28-bb3c-9e2f1fbcb3cc

**Environment and Version info (please complete the following information):**
 - OS: macOS Sonoma 14.6.1
 - System: Intel 64-bit MacBook Pro
 - OpenSCAD Version: I've tried 2021.01 and 2024.10.3 installed from binary.

**Library & Graphics card information**
QObject::startTimer: Timers can only be used with threads started with QThread
OpenSCAD Version: 2024.10.06 (git fc912cb28)
System information: macOS 14.6.1 x86_64 MacBookPro16,1 8 CPUs 32.00 GB RAM 
User Agent: OpenSCAD/2024.10.06 (git fc912cb28) (macOS 14.6.1 x86_64 MacBookPro16,1)
Compiler: Clang "15.0.0 (clang-1500.3.9.4)" 64bit
MinGW build: No
Debug build: No
Boost version: 1_81
Eigen version: 3.4.0
CGAL version, kernels: 6.0, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.6.0
Qt version: 5.15.13
QScintilla version: 2.13.3
InputDrivers: 
GLib version: 2.81.0
lodepng version: 20230410
libzip version: 1.9.2
fontconfig version: 2.14.1
freetype version: 2.12.1
harfbuzz version: 6.0.0
cairo version: 1.18.0
lib3mf version: 2.3.1
Features: fast-csg, fast-csg-safer, fast-csg-debug, roof, input-driver-dbus, lazy-union, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output
Application Path: /Applications/OpenSCAD.app/Contents/MacOS
Documents Path: /Users/*****/Documents
User Documents Path: /Users/*****/Documents
Resource Path: /Applications/OpenSCAD.app/Contents/Resources
User Library Path: /Users/*****/Documents/OpenSCAD/libraries
User Config Path: /Users/*****/Library/Application Support/OpenSCAD
Backup Path: /Users/*****/Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /Users/*****/Documents/OpenSCAD/libraries
  /Applications/OpenSCAD.app/Contents/Resources/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  /System/Library/Assets/com_apple_MobileAsset_Font3
  /System/Library/Assets/com_apple_MobileAsset_Font4
  /Users/*****/.local/share/fonts
  /usr/local/share/fonts
  /usr/share/fonts
  /Users/*****/.fonts
  /usr/X11/lib/X11/fonts
  /System/Library/Fonts
  /Library/Fonts
  /Users/*****/Library/Fonts
  /System/Library/Fonts/Supplemental
  /Library/Fonts/Microsoft

GL context creator: CGL (new)
PNG generator: Core Foundation

GLAD version: 2.0.4
OpenGL Version: 2.1 ATI-5.5.17
GL Renderer: AMD Radeon Pro 5500M OpenGL Engine
GL Vendor: ATI Technologies Inc.
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

**Additional context**
Add any other context about the problem here.

kintel, 2024-10-08:
It is curious that OpenSCAD-2021.01 crashes, as that binary hasn't changed for years and has been running stable on many Macs. ..so I'm suspecting that either something is very odd about your system, or you have rare hardware and happens to be one of the first people running into and reporting this. Could be related to AMD GL drivers.

Could you attach the produced crash log? e.g. click "Show Details" or find it in the crash log in the Console app.

JeffreyOI, 2024-10-08:
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               OpenSCAD [8576]
Path:                  /Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD
Identifier:            org.openscad.OpenSCAD
Version:               2024.10 (2024.10.06)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-10-08 00:13:32.4368 -0400
OS Version:            macOS 14.6.1 (23G93)
Report Version:        12
Bridge OS Version:     8.6 (21P6074)
Anonymous UUID:        4757DB77-4AE9-CF8D-F2ED-00AAFE7DC54B

Sleep/Wake UUID:       206FF249-6B3F-40E6-87B4-FE3345542553

Time Awake Since Boot: 1000000 seconds
Time Since Wake:       65032 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 6 Abort trap: 6
Terminating Process:   OpenSCAD [8576]

Application Specific Information:
abort() called

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	    0x7ff816f24db6 __pthread_kill + 10
1   libsystem_pthread.dylib       	    0x7ff816f5debd pthread_kill + 262
2   libsystem_c.dylib             	    0x7ff816e83a79 abort + 126
3   libsystem_malloc.dylib        	    0x7ff816d883a9 malloc_vreport + 857
4   libsystem_malloc.dylib        	    0x7ff816dabc5b malloc_zone_error + 183
5   libsystem_malloc.dylib        	    0x7ff816d9d389 nanov2_guard_corruption_detected + 34
6   libsystem_malloc.dylib        	    0x7ff816d9c692 nanov2_allocate_outlined + 385
7   libsystem_malloc.dylib        	    0x7ff816d7d14a nanov2_calloc + 607
8   CoreFoundation                	    0x7ff8170020af __NSSetM_new + 213
9   CoreFoundation                	    0x7ff817046b9b _NSSetI_mutableCopyWithZone + 159
10  WindowManagement              	    0x7ffc27998a02 -[_WMWindow applyTags:mask:] + 514
11  AppKit                        	    0x7ff81a96c579 _NSSetHidesOnDeactivate + 81
12  AppKit                        	    0x7ff81a851ae7 -[NSPanel _initContent:styleMask:backing:defer:contentView:] + 61
13  AppKit                        	    0x7ff81a6bc9d0 -[NSWindow initWithContentRect:styleMask:backing:defer:] + 42
14  AppKit                        	    0x7ff81a851a95 -[NSPanel initWithContentRect:styleMask:backing:defer:] + 59
15  AppKit                        	    0x7ff81a97f361 -[NSWindow initWithContentRect:styleMask:backing:defer:screen:] + 50
16  libqcocoa.dylib               	       0x10bd44844 0x10bd0a000 + 239684
17  libqcocoa.dylib               	       0x10bd2edd8 0x10bd0a000 + 151000
18  libqcocoa.dylib               	       0x10bd27cf9 0x10bd0a000 + 122105
19  libqcocoa.dylib               	       0x10bd27767 0x10bd0a000 + 120679
20  QtGui                         	       0x10e5ece22 QWindowPrivate::create(bool, unsigned long long) + 146
21  QtWidgets                     	       0x10e041415 QWidgetPrivate::create() + 1189
22  QtWidgets                     	       0x10e040095 QWidget::create(unsigned long long, bool, bool) + 325
23  QtWidgets                     	       0x10e052959 QWidgetPrivate::setVisible(bool) + 169
24  libqscintilla2_qt5.dylib      	       0x10ce931c6 0x10cd4a000 + 1348038
25  libqscintilla2_qt5.dylib      	       0x10cf14947 0x10cd4a000 + 1878343
26  libqscintilla2_qt5.dylib      	       0x10cf15b7b 0x10cd4a000 + 1883003
27  libqscintilla2_qt5.dylib      	       0x10cd4fec9 QsciScintilla::startAutoCompletion(QsciScintilla::AutoCompletionSource, bool, bool) + 1065
28  QtCore                        	       0x10d6e4f9a 0x10d4e4000 + 2101146
29  libqscintilla2_qt5.dylib      	       0x10cf20b9e QsciScintillaBase::SCN_CHARADDED(int) + 62
30  libqscintilla2_qt5.dylib      	       0x10ced9a3d 0x10cd4a000 + 1636925
31  libqscintilla2_qt5.dylib      	       0x10cf137c4 0x10cd4a000 + 1873860
32  libqscintilla2_qt5.dylib      	       0x10cd5a9fb QsciScintillaBase::keyPressEvent(QKeyEvent*) + 459
[... 845 lines cut ...]
      "rolloutId" : "648cada15dbc71671bb3aa1b",
      "factorPackIds" : {
        "SIRI_UNDERSTANDING_ATTENTION_ASSETS" : "65a8118e05d9422724106749",
        "SIRI_UNDERSTANDING_ASR_ASSISTANT" : "65a8115446dfa0720068876d",
        "SIRI_UNDERSTANDING_NL" : "65a811c046dfa0720068876e",
        "SIRI_EXPERIENCE_CAM" : "65a81173096f6a1f1ba46525",
        "SIRI_UNDERSTANDING_MORPHUN" : "65a811a96c2f903aa921c246"
      },
      "deploymentId" : 240000116
    }
  ],
  "experiments" : [
    {
      "treatmentId" : "c47ab4cc-c9c3-4b5d-a87c-e2433ce02597",
      "experimentId" : "6639bc6ba73d460582162323",
      "deploymentId" : 400000006
    },
    {
      "treatmentId" : "45f4e2a5-551b-4bc2-a2dc-19c244dda8f8",
      "experimentId" : "6643969b3099cf28e049862f",
      "deploymentId" : 400000007
    }
  ]
}
}

kintel, 2024-10-09:
> Time Awake Since Boot: https://github.com/openscad/openscad/commit/1000000907bb30c11336942149fd0bfaeca2689c seconds

Just to be on the safe side, could you try rebooting?

kintel, 2024-11-28:
Closing due to lack of feedback

closed 2024-11-28 by kintel