openscad/openscad #5363: Do we need both Window/Editor and Window/Hide Editor?

jordanbrown0, 2024-10-09:
Like the Title says.  Do we need both of these menu items?  Why not just one "Editor" item, that's either checked or not?

t-paul, 2024-10-09:
First is "Select window and make visible if not" for keyboard navigation use
Second is "Toggle visibility" (at some point I'll change the "Hide" to "Show", as that negative check mark is annoying)

They overlap in functionality but I don't see how to get the same feature set with a single entry.

jordanbrown0, 2024-10-09:
Hmm.  Standard window-selection keyboard navigation on Windows is Alt-Tab, and Ctrl-Tab switches tabs (and, I think, between sub-windows in an MDI).  OpenSCAD's secondary windows are some sort of second-class window; they don't appear in the taskbar and you can't navigate to them with Alt-Tab.  I'd have to look around for other similar layouts and see how they're handled.  Promoting them to first-class windows would allow Alt-Tab (and allow minimize and maximize), but I don't know whether you want the taskbar clutter.

jordanbrown0, 2024-10-20:
Another nearby tidbit:  the Windows taskbar does both functions, using sort of a three-way toggle.  If the window is buried or minimized, clicking brings it to the top.  If the window is already on the top, clicking minimizes it.

But I don't think I'd like a three-way toggle for a menu checkbox.

damienmarchal, 2025-01-22:
My 2 cents:
 * I share the questioning of @jordanbrown0 about the double entries in the menu (as mentionned in #5613)
 * About toggling the "Hide" to "Show"...  for sure negative checks shouldn't be use, but "Show Editor" neither... because depending on the state it will "show" or "hide".  

To makes things clearer I think that:
    - toggle the visibility on/off: should use a check box with just the dock's name (eg: "Editor"), because the checkbox "contains" the information that it will hide or show. This is also the default behavior for such thing in QT. 
    - only make the dock visible:  there should be no check box, and both use of "Editor" and "Show Editor" are ok. Because it always show the editor.

Here is what Visual Studio Code is doing for the show only option: 
[image]

The interesting thing is that to closing a open view in Visual Studio Code is not done by using an additional/duplicated menu but by letting the user "closing" the dock using its closing "cross". 

I will give it a quick try.

damienmarchal, 2025-01-22:
Here is the result of a quick test. 

[image]

It removes the existing "Hide XXXX" and use the QT default action for such task. 
```
    std::vector<Dock*> docks = {
      editorDock,
      consoleDock,
      parameterDock,
      errorLogDock,
      animateDock,
      fontListDock,
      viewportControlDock
  };

  for(auto& dock : docks){
      menuWindow->addAction(dock->toggleViewAction());
  }
```
 
If extra things has to be done when the visibility is changed... it is possible to connect to the action triggered.

damienmarchal, 2025-01-22:
To handle the keyboard navigation from the different docks without the main window menu entries it could be possible to use a dedicated popup menu with the same content. 

In qtcreator, this is for exemple used to navigate from file to file:
[image]

But same principle could be used for navigation between windows.

t-paul, 2025-01-22:
I think the navigation actions are useful for keyboard focused people and essential for vision impaired users. If we can provide a reasonable and accessible way to close the windows, the checkbox action could probably be dropped without harm.

The file switcher might be a good idea too, it feels like this could be more accessible than the tabs. I have to try this out though, I'm not really experienced in using those tools.

jordanbrown0, 2025-01-23:
In a Windows multi-pane ([MDI](https://en.wikipedia.org/wiki/Multiple-document_interface)) application, I believe that Ctrl+F4 is the usual way to close the focused pane, and Ctrl+Tab is the usual way to move the focus from one pane to another.  However, those may have been co-opted by tabs, since tabs are the new MDI.  I don't know whether there are many examples of multi-pane multi-tab applications.

damienmarchal, 2025-01-27:
Hi, 

Here is  small proposal on how the remove the duplicated entries in the window menu still preserving keyboard navigation. 
Part 1: add a menu entry "Jump to..." and a submenu with the editor names.  This has a double advantage, as it state clearly what will happens (jump to...) and avoid the quasi duplicated name entries in the same menu 

[image]

Part 2: to make the keyboard navigation more fast, add a shortcut so this submenu can pop-up at any place. This allows to jump from docks to dock very conveniently.  

[image]

The implementation quite simple.

damienmarchal, 2025-01-29:
Hi,

I continued working on this with this small experiment in which I added a visual feedback when navigating from dock to dock. (see in video: https://youtu.be/XgBB8Lnk_qM)

[image]

damienmarchal, 2025-03-21:
The menu has been changed in PR #5630.
The nearly duplicated entries has been removed & there is no more Hide/Show with the checkbox. 

The issue can probably be close

kintel, 2025-03-21:
Let's close this as the Window menu was reworked to improve this. If issues remain, let's open new tickets

closed 2025-03-21 by kintel