openscad/openscad #5376: "The given mesh is not closed" rendering error when `rotate_extrude` rotates a shape with a point on the rotation axis.

lgarron, 2024-10-14:
**Describe the bug**

OpenSCAD prints the following error when `rotate_extrude` rotates a shape with a point on the rotation axis.

```
ERROR: The given mesh is not closed! Unable to convert to CGAL_Nef_Polyhedron.
```

**To Reproduce**

Use OpenSCAD to render on the following scene:

```
rotate_extrude(convexity = 10) translate([ 1, 0, 0 ]) circle(r = 1, $fn = 8);
translate([ 2, 0, 0 ]) cube([ 2, 2, 2 ], center = true);
```

This includes:

- A low-poly torus generated from a circle tangent to the origin, i.e. a torus with a hole that has contracted to size 0.
- Any other bogus object in the scene, which doesn't even have to overlap.

Previewing this scene works perfectly fine:

[image]

But rendering it results in the error above and removes the torus from the scene:

[image]

**Expected behavior**

Neither previewing nor rendering the scene causes an error, and all the objects are rendered.

**Code reproducing the issue**
See above.

**Screenshots**
See above.

**Environment and Version info (please complete the following information):**
 - OS: macOS 15.0.1
 - System: MacBook Pro
 - OpenSCAD Version: 2021.01 (latest [from Homebrew](https://github.com/Homebrew/homebrew-cask/blob/60cb5e6dc01db4b11cc87c52357a4011c5a0ad6e/Casks/o/openscad.rb))

**Library & Graphics card information**
N/A

**Additional context**
N/A

pca006132, 2024-10-14:
rotate extrude and linear extrude are known to produce non-manifold results... we should fix this at some point similar to manifold's extrude algorithms (adapt it to fit openscad).

kintel, 2024-10-14:
This was fixed for the Manifold backend back in April or so. If you use a recent development snapshot instead of 2021.01 and select Manifold from Preferences->Advanced

closed 2024-10-14 by kintel

lgarron, 2024-10-14:
> This was fixed for the Manifold backend back in April or so. If you use a recent development snapshot instead of 2021.01 and select Manifold from Preferences->Advanced

Woah! Not only does this resolve the issue for me, it also reduces rendering time from 9 minutes to 1 second when I crank up `$fn` for the project that was running into this! 😳

I'm not sure what to make of this output, but it doesn't seem to have affected the accuracy of the result:
```
   Top level object is a 3D object (manifold):
   Status:     NoError
   Genus:      -14
   Vertices:    57205
   Facets:     114350
```

kintel, 2024-10-14:
That output is perfectly fine. Manifold has some special interpretation of negative genus, it probably means you've got multiple objects. @pca006132 may be able to comment on that (and whether we should avoid printing it if it's potentially confusing)

pca006132, 2024-10-15:
Yeah it is due to having multiple objects. Maybe we should not display the information as it may scare users.

v3rtx, 2025-05-15:
@kintel huge thanks for the dev snapshot tip! Works like a charm, also the whole UI is much better than stable 2021 version. Just curious, taking into account all the goodness that has been added since 2021 why there wasn't a stable release?

rowlesmr, 2025-05-15:
I'm running into a bunch of these errors when using BOSTL and the planetary gear generator, as well as a bunch of other things I'm doing which use rotate_extrude. 

I may have to dive into trying to build my own verions for Windows...

kintel, 2025-05-15:
> [@kintel](https://github.com/kintel) huge thanks for the dev snapshot tip! Works like a charm, also the whole UI is much better than stable 2021 version. Just curious, taking into account all the goodness that has been added since 2021 why there wasn't a stable release?

It may be better, but not everything is stable enough: https://github.com/orgs/openscad/projects/2

kintel, 2025-05-15:
@rowlesmr Please discuss on the mailing list, or open a new issue if you have a specific problem with OpenSCAD itself. But chat with the BOSL folks first :)