openscad/openscad #5381: Flipped normals under certain conditions

UBaer21, 2024-10-17:
**Describe the bug**
normals flipped

**To Reproduce**
version 2024.10.15
run code export 3mf - check with 3rd party software as no error will show in open SCAD

**Expected behavior**
no flipped normals

**Code reproducing the issue**
```

$fs=.2;
$fa=1;

l2=50/6;
threadD=6;
bD=8;

// union()  // 3mf export error
for (fn=[85:5:95])translate([fn*2-160,0])Test(1.5,fn);

module Test(rund=2,fn)union(){
  rotate_extrude()
  Rund(0,rund){
  //offset(-rund,$fn=fn2)offset(rund,$fn=fn2){
    translate([0,-l2-.5])square([bD/2*sin(35),l2],center=0);
    intersection(){
      circle(d=bD,$fn=fn);
      translate([0,-25])square(50);
    }
  }
}

module Rund(or=+0,ir,chamfer=false,fn,fs=$fs,fa=$fa) {
    fs=is_list(fs)?fs:[fs,fs];
    fn=is_list(fn)?fn:[fn,fn];
    ir=is_undef(ir)?is_list(or)?or[1]:or:ir;
    or=is_list(or)?or[0]:or;
    chamfer=chamfer?true:false;
   if(!chamfer)
          offset(r = -ir,$fn=fn[1],$fs=fs[1],$fa=fa)offset(delta =  ir,$fn=fn[1],$fs=fs[1],$fa=fa) 
          offset(r =  or,$fn=fn[0],$fs=fs[0],$fa=fa)offset(delta = -or,$fn=fn[0],$fs=fs[0],$fa=fa)
       children();
}
```

**Screenshots**
[image]

**Environment and Version info (please complete the following information):**
 - OS:Windows 10
 - System: Intel PC 64-bit,
 - OpenSCAD Version 2024.10.15

**Additional context**

the sequence of offsets with certain settings play a role.
changing resolution or size will move the error to a different combination

UBaer21, 2024-10-17:
small changes also show that rotate extrude can't generate the surface.
```
$fs=.2;
$fa=1;

l2=49/6.0;
bD=7.3;

// union()  // 3mf export error
for (fn=[85:5:130])translate([fn*2-160,0])Test(1.5,fn);

module Test(rund=2,fn)union(){
  rotate_extrude()
  offset(-rund)offset(delta=rund){
    translate([0,-l2-.5])square([bD/2*sin(35),l2],center=0);
    intersection(){
      circle(d=bD,$fn=fn);
      translate([0,-25])square(50);
    }
  }
}
```
[image]

ERROR: all points for rotate_extrude() must have the same X coordinate sign (range is -0.00 -> 3.65)

pca006132, 2024-10-18:
OK, the issue is due to having negative floating point values. Basically, offset can cause 0 to offset to slightly negative (around 1e-16), and the rotate extrude function uses the min x sign to determine if things should be flipped, which in this case it flipped because the sign is negative. The second issue is exactly the same as the first one, but the code correctly catches the problem instead of silently giving a corrupted version.

```c++
  if ((max_x - min_x) > max_x && (max_x - min_x) > fabs(min_x)) {
```

`max_x - min_x == max_x` because `min_x`'s magnitude is too small comparing with `max_x`, so the code cannot correctly identify `max_x - min_x > max_x` in this case. In fact, I am not sure why we should do arithmetic at all... checking the signs of min and max x directly should fix the issue.

pca006132, 2024-10-18:
And this is a fix for your code, because fixing the rotate extrude function will only make your code produce error:

```scad
module Test(rund=2,fn)union(){
  rotate_extrude()
  intersection() {
    // just make things non-negative.
    translate([0,  -50]) square([1000, 1000]);
    offset(-rund)offset(delta=rund){
      translate([0,-l2-.5])square([bD/2*sin(35),l2],center=0);
      intersection(){
        circle(d=bD,$fn=fn);
        translate([0,-25])square(50);
      }
    }
  }
}
```

thehans, 2024-10-18:
I am starting to wonder if it would be better to just have rotate_extrude pre-process the 2d shape, basically union it with a horizontal mirror of itself, then clipping to the positive-X halfplane.

UBaer21, 2024-10-19:
An auto cut sounds a good idea.
Wouldn't it be more performant to just limit the points to min (0,x) or max(0,x) depending if  max (x) or -min(x) of the polygon is bigger. So we are not getting an additional difference or intersection although the performance impact in 2D is probably not so high.

thehans, 2024-10-19:
Simply limiting the x coordinate without a proper clipping operation could result in overlapping colinear line segments along the y-axis, which i doubt the geometry engine would be happy with.

UBaer21, 2024-10-19:
You are right about that, and it is for sure not elegant but seems it causes less problems than the current situation.

```
p=[
[10,10],
[10,-10],
[0,-10],
for ([0:100]) [0,rands(-10,10,1)[0]],
[0,10],
];

rotate_extrude()polygon(p);
```

Also the question if we should encourage people to design with polygons that are in + and - X, which is a questionable choice in parametric design to hope that the proper parts are cut. 

`rotate_extrude() rotate(45) square([50,5],true);`
Do we want this working or better not causing a warning/error ?

PR #5384 (2024-10-23): fix rotate extrude
Fixes #5381
closed 2024-10-23 by kintel