openscad/openscad #5382: $fonts

yahbluez, 2024-10-17:
I like to suggest to implement an internal static list that contains a list of all fonts visibly from openscad.

For example:

$fonts = ["Liberation Sans:style=Bold Italic", "font2", "font3", ...]

text("OpenSCAD", font = $fonts[0] );

With such a list openscad script / customizer has access to a font switcher.

The content of this $fonts list should be the same as visibly in the GUI.

jordanbrown0, 2024-10-17:
Customizer doesn't have access to dynamic values, including variables.  (It sets them, but can't read them.)  Customizer support would require some special markup.

I'd be inclined toward a function rather than a variable.

But I have to ask:  how would the OpenSCAD program use this information?  If it doesn't know what font it wants a priori, how would it use this information to pick one?  (Excepting demo programs that just emit samples of all fonts or just dump list of fonts to the console.)  Note that if you were thinking of matching parts of the font name (like "Bold Italic"), you can already use fontconfig font-selection expressions to select a font, e.g.:
```
text("Hello", font=":family=sans-serif:slant=italic");
```
The fontconfig documentation isn't very good, but it's a start: https://fontconfig.pages.freedesktop.org/fontconfig/fontconfig-user.html

yahbluez, 2024-10-22:
> But I have to ask: how would the OpenSCAD program use this information?

The idea is to let a user select in the Customizer which font he like to use.
In the same way a user gives any data or uploads a file he could select a font from the list of fonts.

Today there is no way to handel that inside openscad without let the user type in the FONT name wich is much top much for many.

Working and loving openscad i see more and more that there are build in limits that are unnecessary.

I  guess it is my lag of understanding but why is there no way to get the (any) dimensional information about an included STL file? (for example)

The idea in BOSL to have modules and same name functions that deliver data is very nice.

But that is another story.

Back to font, if the user types in a font path by hand, that eventually  exists,  he can chose the font.
But it would be much more easy if that would work with the existing font list.

t-paul, 2024-10-22:
There's not going to be a `$fonts` variable. This would allow assignments which would just be confusing.

A function returning the font list would be possible, but that would not be of any use for the customizer as mentioned above.

If the goal is to allow font selection in customizer, there's an issue for that already #4291. My understanding this is the main target, so closing as duplicate.

closed 2024-10-22 by t-paul

jordanbrown0, 2024-10-23:
I'm happy to explain why the current design doesn't allow for getting dimensional information about STL files, but this isn't the right place.  Ask on the mailing list or drop me a note at openscad (at) jordan.maileater.net.