openscad/openscad #5386: Always start a new scope with {}

jordanbrown0, 2024-10-20:
**Is your feature request related to a problem? Please describe.**
In most C-like languages, braces {} delimit scopes.  It's not good practice, but
```
int x;
{
    int x;
}
```
is absolutely legal C.

OpenSCAD looks similar.
```
x = 1;
if (something) {
    x = 2;
}
```
is legal OpenSCAD; the "if" block is a new scope.

But, unlike C, it is not the braces that cause the new scope; it is that the brace-enclosed block is a child of the "if".
```
x = 1;
{
    x = 2;
}
```
is *not* legal OpenSCAD; the braces do not start a new scope.

**Describe the solution you'd like**
Have braces always start a new scope, so that
```
x = 1;
{
    x = 2;
}
```
is legal.

**Additional context**
Why?  Consistency with other languages.  Ability to have variables that are local to one part of a larger sequence.  Avoid confusion in cases like
```
x = 1;
// Comment out the "if" to make the block unconditional
// if (something)
{
    x = 2;
}
```

Theoretically such a change would be backward-incompatible.  However, "naked" braces like this seem unlikely in the wild, given that they have zero(?) effect.  Cases that rely on values escaping from the inner scope
```
{
    x = 1;
}
echo(x);
```
seem even less likely.

nophead, 2024-10-20:
I think it has been mentioned before and people said they use naked brackets to be able to collapse code in the editor.

jordanbrown0, 2024-10-20:
Drat.  So that *is* a reason for them.  I'd still make the change, but it's more of a compatibility concern than I thought.

(I never use that "collapse" feature, so didn't remember it.)

pca006132, 2024-10-21:
Maybe we can make some comment pattern for collapsing? e.g. `//{` and `//}`.

EmperorArthur, 2024-11-01:
> I think it has been mentioned before and people said they use naked brackets to be able to collapse code in the editor.

Wow, that would confuse the heck out of me if I ever came across code which did that.

@jordanbrown0 unfortunately, variable handling in OpenSCAD is different from every other language I've ever used.  However, I believe I have a temporary solution to your issue.  The `group()` builtin should do it.

```scad
echo(a);
a = 1;
group(){
  echo(b);
  b = 2;
}
echo(b);
```
gives:
```
ECHO: 1
ECHO: 2
[WARNING: Ignoring unknown variable 'b' in file , line 7]
ECHO: undef
```