openscad/openscad #5387: Can't set font size for font Noto Emoji

microenh, 2024-10-21:
First, let me say, I just recently discovered OpenSCAD and I am really enjoying using it. It's become my main 3D modeling software. Thanks to all who make it possible.

On the Mac versions 2024.10.14 and 2024.10.20, OpenSCAD gets the error "Can't set font size for font Noto Emoji:Style = Regular."  under both preview and render.

font = "Noto Emoji:style=Regular";
text("\u263a",size=20,font=font);

If the font is something else i.e. "LiberationMono" the code executes. The given Unicode character is the smiley face emoji.

This is with the Manifold render engine. With the CGAL engine, there is no error, but nothing is displayed in either preview or render.

The code works without error and renders properly on Mac OpenSCAD version  2021.01 as well as Windows 2021.01 and 2024.10.17

This is a Mac Min M2 running OSX 14.7 Sonoma
The font Noto Emoji is properly installed. The font was downloaded from Google Fonts.

[https://fonts.google.com/noto/specimen/Noto+Emoji](url)

t-paul, 2024-10-21:
The Noto Emoji font family does not have regular fonts, they use a table of SVGs instead of the normal font glyphs.

This is not supported and out of scope as those images are usually based on color gradients, layered objects and other features not compatible with the CAD style use of SVG in OpenSCAD.

For example [emoji_u263a.svg](https://github.com/googlefonts/noto-emoji/blob/main/svg/emoji_u263a.svg) would just import as filled circle even if the font support code could read the SVG font data.

microenh, 2024-10-22:
Thanks for the reply. 

I don't know if this makes a difference, but there is a monochrome version of Noto Emoji. That is the one I installed and it does work on OpenSCAD 2021.01 and Windows releases, just not the Developer's version on the Mac.

t-paul, 2024-10-22:
Please upload the exact font file, this should be fine license wise, otherwise it's impossible to tell what type of font file it is. This probably needs to be in a ZIP file as github does not allow arbitrary attachments.

The font selection given in OpenSCAD is *not* actually selecting a specific font but leaves that decision to the fontconfig library which is trying to match the actual font. I suspect there's some different font matching going on causing the differences.

Try running `openscad --debug=fontcache`, this should show the actually selected font, e.g.:

```
FontCache: font = "Noto Emoji:style=Regular", lookup = "Noto Emoji:style=Regular"
FontCache: charmap = 0: platform = 0, encoding = 5
FontCache: charmap = 1: platform = 3, encoding = 10
FontCache: Successfully selected unicode charmap: Noto Color Emoji/Regular
FontCache: result = "Noto Color Emoji", style = "Regular"
WARNING: Can't set font size for font Noto Emoji:style=Regular 
```

microenh, 2024-10-22:
--debug=fontcache  sheds some light on the situation.

with Mac version 2024.10.20 the console shows this:

FontCache: font = "Noto Emoji:style=Regular", lookup = "Noto Emoji:style=Regular"
FontCache: charmap = 0: platform = 0, encoding = 4
FontCache: Successfully selected unicode charmap: Apple Color Emoji/Regular
FontCache: result = "Apple Color Emoji", style = "Regular"

with version 2021.01 (which works)

FontCache: font = "Noto Emoji:style=Regular", lookup = "Noto Emoji:style=Regular"
FontCache: charmap = 0: platform = 0, encoding = 3
FontCache: charmap = 1: platform = 0, encoding = 4
FontCache: charmap = 2: platform = 3, encoding = 1
FontCache: charmap = 3: platform = 3, encoding = 10
FontCache: Successfully selected unicode charmap: Noto Emoji/Regular
FontCache: result = "Noto Emoji", style = "Regular"

It appears that with 2024.10.20 the font selection is falling back to Apple Color Emoji which can't be rendered as it is color. With 2021.01, it doesn't fall back, but selects the correct (monochrome) font.

The font file I installed is attached.

[NotoEmoji-VariableFont_wght.ttf.zip](https://github.com/user-attachments/files/17481082/NotoEmoji-VariableFont_wght.ttf.zip)

kintel, 2024-10-22:
The selection probably comes directly from fontconfig. On macOS we do ship our own fontconfig configuration, and libraries have been upgraded since 2021.01. Perhaps something is odd about that.

A Blackbox way of looking for the regression could be to bisect OpenSCAD binary builds..

Persistenttiger, 2025-01-17:
I have also come across this error using a mac. Is there a work around?

drpeppershaker, 2025-01-28:
> I have also come across this error using a mac. Is there a work around?

I'm here with the same issue. I don't know when the change was implemented. But the following works perfectly in 2023.07.17

`linear_extrude(5)
fill()
text(text="🖤", size=15, font="Noto Emoji:style=Regular",halign="center",valign="center");`

I guess the current work around is to roll back to a version where Noto Emoji still works

jordanbrown0, 2025-01-28:
You don't say whether you're using the "NotoEmoji-VariableFont_wght" variation or the static "NotoEmoji-Bold" (et al) variation.  I don't know whether it matters, but it might.

On Windows 11, I'm seeing the opposite behavior:  I can use these fonts on a development snapshot, but not on 2021.01.

Does Help/Font List show this font?

What does Help/Library Info say about fontconfig in each of the two builds?  (I have 2.13.1 in 2021.01 and 2.14.2 in 2024.12.26.)

jordanbrown0, 2025-01-28:
It may also be relevant that with `--debug=fontcache` I get
```
Fontconfig error: Cannot load default config file: No such file: (null)
```
on the Windows (not OpenSCAD) console.

jordanbrown0, 2025-01-28:
... on both builds.

jordanbrown0, 2025-01-28:
Note:  the "set font size" referred to here is internal; it is not related to OpenSCAD's `size=` parameter.

drpeppershaker, 2025-01-28:
Bear with me, I'm a little bit new to this, so I apologize if I’m not providing exactly what you’re asking for. Please let me know if I can clarify anything.

MacOS 14.5
Apple M1 Max
[OpenSCAD](https://www.openscad.org/) version 2025.01.27 (git 11c07a0d6)

Comparing to the working version of an old snapshot I happened to grab a while back
[OpenSCAD](https://www.openscad.org/) version 2023.07.17 (git 44fdb2c46)

> You don't say whether you're using the "NotoEmoji-VariableFont_wght" variation or the static "NotoEmoji-Bold" (et al) variation. I don't know whether it matters, but it might.

Variable Font Weight is the one I have installed.
NotoEmoji-VariableFont_wght.ttf

> On Windows 11, I'm seeing the opposite behavior: I can use these fonts on a development snapshot, but not on 2021.01.

> Does Help/Font List show this font?
Yes, it shows the same in the 2023.07.17 as well as 2025.01.27

[image]

>What does Help/Library Info say about fontconfig in each of the two builds? (I have 2.13.1 in 2021.01 and 2.14.2 in 2024.12.26.)
It appears that both my development snapshots are using the same fontconfig(?)

**2025.01.27**
fontconfig version: 2.14.1

**2023.07.17**
fontconfig version: 2.14.1

jordanbrown0, 2025-01-28:
I suspect that this change was introduced by #5191 in June 2024.

These look like they are the key entries:

### 45-generic.conf

```
        <alias binding="same">
                <family>Apple Color Emoji</family> <!-- Apple -->
                <default><family>emoji</family></default>
        </alias>
```

```
        <alias binding="same">
                <family>Noto Emoji</family> <!-- Google -->
                <default><family>emoji</family></default>
        </alias>
```

### 60-generic.conf

```
        <!-- Prefer to match color emoji font. -->
        <match>
                <test name="lang">
                        <string>und-zsye</string>
                </test>
                <test qual="all" name="color" compare="not_eq">
                        <bool>true</bool>
                </test>
                <test qual="all" name="color" compare="not_eq">
                        <bool>false</bool>
                </test>
                <edit name="color" mode="append">
                        <bool>true</bool>
                </edit>
        </match>

        <!-- TODO
         ! Match on "color" and alias B&W ones first if no color is requested.
         ! That's "hard" because <alias> doesn't work in match and needs to be
         ! expanded to its non-sugar form.
         !-->
        <alias binding="same">
                <family>emoji</family>
                <prefer>
                        <!-- System fonts -->
                        <family>Noto Color Emoji</family> <!-- Google -->
                        <family>Apple Color Emoji</family> <!-- Apple -->
                        <family>Segoe UI Emoji</family> <!-- Microsoft -->
                        <family>Twitter Color Emoji</family> <!-- Twitter -->
                        <family>EmojiOne Mozilla</family> <!-- Mozilla -->
                        <!-- Third-Party fonts -->
                        <family>Emoji Two</family>
                        <family>JoyPixels</family>
                        <family>Emoji One</family>
                        <!-- Non-color -->
                        <family>Noto Emoji</family> <!-- Google -->
                        <family>Android Emoji</family> <!-- Google -->
                </prefer>
        </alias>
```

The "prefer to match color emoji" block in 60-generic is very suspicious.

You might try appending `:color=false` to your font name.  That might help, but I think it would be a bug if it's needed.  It seems to me that the font name should be the very highest priority matching parameter, but I'm not familiar with fontconfig and its matching rules.

kintel, 2025-01-28:
Note that we got those configs directly from fontconfig. Perhaps there are upstream issues in fontconfig which have been fixed since then, or are still present.

drpeppershaker, 2025-01-28:
`:color=false`

Yes that worked!
`font="Noto Emoji:style=Regular:color=false` gives the expected result.

jordanbrown0, 2025-01-28:
>Note that we got those configs directly from fontconfig. Perhaps there are upstream issues in fontconfig which have been fixed since then, or are still present.
Or their use cases are sufficiently different from ours that their configs are not appropriate for us.

jordanbrown0, 2025-01-29:
"How could anybody prefer black-and-white line-drawing emoji over color shaded emoji?"