openscad/openscad #5388: Insertion caret (cursor) disappears from editor on Mac OpenSCAD 2024 multiple versions

microenh, 2024-10-21:
MacOpenSCAD versions 2024.10.11, 2024.10.14 and 2024.10.20 the insertion caret (cursor) disappears in the editor window. I've not determined a repeatable way to cause it, it just disappears from time to time. Some menu options may make it reappear, but the best way to get it back is to shut down OS and restart.

Though the caret is not visible, the insertion point is still active and can be moved with the arrow keys.

Mac M2 Mini
OSX 14.7 Sonoma

kintel, 2024-10-21:
I've experienced the same, but I've been unable to reproduce in a repeatable fashion.

microenh, 2024-10-23:
Some more information on this. I have found that when the caret disappears, hiding and re-showing the editor window does not bring the caret back, but minimizing and restoring the entire OpenSCAD application does bring the caret back.

kintel, 2024-12-01:
Starting with 2024.12.01, we've upgraded to Qt6, which may influence this issue. If I don't hear any new reports in a while, let's consider it solved. If anyone still experiences this, please update this issue

closed 2026-10-08 by github-actions[bot]