openscad/openscad #5389: openscad-nightly linux 22/10/24 and before

BertrandSahel, 2024-10-23:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Code reproducing the issue**
If applicable, add trimmed down, self-contained code demonstrating the issue.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info (please complete the following information):**
 - OS: [e.g. Windows 10, macOS 12 Monterey, Ubuntu 22.04]
 - System: [e.g. Intel PC 64-bit, Raspberry PI 4]
 - OpenSCAD Version [e.g. 2021.01, 2023.04.16] and where it was installed from

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

**Additional context**
Add any other context about the problem here.

BertrandSahel, 2024-10-23:
impossible to resize the main window , staying in full screen . The only options are 'go to precedent screen'  and 'minimize'.

t-paul, 2024-10-23:
Please edit the main post and add the information needed, like specific Linux distribution and a screenshot. This sounds like an issue with the desktop system and not OpenSCAD itself. You could try removing the config file, which then should default to a standard window layout of OpenSCAD.
```
mv -iv ~/.config/OpenSCAD/OpenSCAD.conf ~/.config/OpenSCAD/OpenSCAD.conf.bak
```

BertrandSahel, 2024-10-23:
Thanks to t-paul: problem solved!

closed 2024-10-23 by BertrandSahel