openscad/openscad #5390: REGRESSION: Doesn't have an icon anymore

php4fan, 2024-10-26:
**Describe the bug**
I've had Openscad installed on Manjaro Linux for ages, installed from "Official Repositories (extra)".
It used to have an icon, but now it shows up with a generic Wayland icon (see screenshot below)
**To Reproduce**
Steps to reproduce the behavior:
1. Launch Openscad

**Expected behavior**
Should have its own icon in the Task manager

**Observed behavior**
(note BTW the "observed behavior" section was missing in this template)
Has a generic Wayland icon.

**Screenshots**

[image]

**Environment and Version info (please complete the following information):**
 - OS: Manjaro Linux
 - System: Intel PC 64-bit
 - OpenSCAD Version 2021.01 installed from Manjaro's official repositories (extra)

**Library & Graphics card information**
Operating System: Manjaro Linux 
KDE Plasma Version: 6.1.5
KDE Frameworks Version: 6.6.0
Qt Version: 6.7.2
Kernel Version: 6.6.54-2-MANJARO (64-bit)
Graphics Platform: Wayland
Processors: 12 × 12th Gen Intel® Core™ i7-1255U
Memory: 15.3 GiB of RAM
Graphics Processor: Mesa Intel® Graphics
Manufacturer: ASUSTeK COMPUTER INC.
Product Name: Vivobook_ASUSLaptop X1502ZA_F1502ZA
System Version: 1.0

t-paul, 2024-10-26:
Sorry, but there is no plan on updating the 2021.01 release for Wayland support. As this sounds like it's built directly by Manjaro, there is a theoretical option to back-port the change (I believe that is  #4862), but that's not in our control.

closed 2024-10-26 by t-paul

php4fan, 2024-10-26:
I don't understand. Is there a more recent release than 2021.01?

t-paul, 2024-10-26:
The is no later release yet. Right now there is a (quite usable) nightly build which will be turned into a release version at some point.

What I'm trying to say is that it's unlikely that an official 2021.01-fix version will be released.