openscad/openscad #5391: Errors apparently produced while exporting .stl

rjcarlson49, 2024-10-26:
**Describe the bug**
When the part described in the attached .scad file is exported to .stl, it produces a file with many errors. The errors are reported by PrusaSlicer. PS "repairs" the errors and performs a slice, but the results are completely screwed up.

**To Reproduce**
- Using the file attached, which requires BOSL2, export the part in .stl (F7). 
- Import the .stl to PrusaSlicer 2.8.1.
- Note the errors listed.
- Hit the icon to slice.
- Looking at the rendering of the gcode, you can see that the slice is incorrect.

**Expected behavior**
A coherent slice should have been produced and no errors should be detected by PS. Using a .obj export produces a coherent slice.

**Code reproducing the issue**
`````
include <BOSL2/std.scad>

// Number of Teeth
_n = 36;          // Number Of Teeth

// Inner Radius
_ir = 30;
// Outer Radius
_or = 50;
// Is the coupling conical?
_conic = 10;
// Tooth Profile Angle
_profile = 60; // [60, 90]
// Percentage of tooth height
_chamfer = 5; // Default 5%
_base = 1;

$fn = 180;
tiny = 1 / 1024;
superFine = 16*360;

hirth_concave();

_irRatio = _ir / _or;
_toothAngle = 360/_n;
_toothHeight = (sin(_toothAngle/2) * _or) / tan(_profile/2);
_chamferHeight = _chamfer * _toothHeight / 100;
_grooveAngle =  atan(((_toothHeight/2) + _conic) / _or);
_ridgeAngle = -atan(((_toothHeight/2) - _conic) / _or);
_innerCenterLine = (1 - _irRatio) * _conic;
_stackHeight   = _innerCenterLine + _toothHeight + 2*_base;
_tubeHeight = 2*_conic + 2*_base + 2*_toothHeight;

module hirth_concave() {
    zrot(_toothAngle/2)
    union() {
        difference() {
            intersection() {
                union() {
                    for (i = [0 : 360/_n : 359]) 
                        zrot(i)
                        _profileF();
                }
                tube(ir = _ir, or = _or, anchor = CENTER, l = _tubeHeight, orient = UP, $fn = superFine);
            }
            _chamferF();
            zcyl(r = _ir, h = _tubeHeight, $fn = 16*360, anchor = CENTER);
        }
        _baseF();
    }
} // hirth_concave

module _profileF() {
    IR = _ir * .5;
    OR = _or * 1.5;
    tI = [spherical_to_xyz(IR,   0,            _grooveAngle + 90),
          spherical_to_xyz(IR,  _toothAngle/2, _ridgeAngle + 90),
          spherical_to_xyz(IR, -_toothAngle/2, _ridgeAngle + 90)];

    tO = [spherical_to_xyz(OR,   0,            _grooveAngle + 90),
          spherical_to_xyz(OR,  _toothAngle/2, _ridgeAngle + 90),
          spherical_to_xyz(OR, -_toothAngle/2, _ridgeAngle + 90)];
    up(_conic)
    skin([tI, tO], slices = 0);
}

module _chamferF() {
    A = -_toothHeight/2 - .1;
    B = -_toothHeight/2 + _chamferHeight;
    
    pts = [[0, _conic], 
           [_or+tiny, A], 
           [_or+tiny, B]];

    rotate_extrude(angle = 360, $fn = superFine)
    polygon(pts);
}

module _baseF() {
    A = _base + _irRatio*_toothHeight/2 + _innerCenterLine;
    B = _irRatio * (_toothHeight/2 - _chamferHeight/2) + _innerCenterLine;
    C = _toothHeight/2 - _chamferHeight/2;
    pts = [
           [_ir, A],
           [_ir, B],
           [_or, C],
           [_or, A]
           ];

    rotate_extrude(angle = 360, $fn = superFine)
    polygon(pts);
}
`````
**Screenshots**
[image]

**Environment and Version info (please complete the following information):**
OpenSCAD 24.09.20
PrusaSlicer 2.8.1
MacOS 14.6.1
iMac M3

**Library & Graphics card information**
BOSL2 snapshot from 24.05.03

**Additional context**
When the part is exported to .obj no errors are reported and the slice is correct.

nophead, 2024-10-26:
You probably have points too close together to be represented with the floats in STL files uniquely, so they get merged when the file is read causing broken topology. OBJ files I think store topology rather than just a triangle soup.

rjcarlson49, 2024-10-27:
It turns out that In _profileF the polyhedron is constructed inside out. If the second and third points of tI and tO are swapped, the polyhedron is correct and the problem disappears. The bug is now that OpenSCAD does not detect the inside out polyhedron. One could also say that skin() contains a bug because it did not generate a correct polyhedron.

pca006132, 2024-10-27:
and re. stl, my recommendation would be to avoid stl and use something else like 3mf

jordanbrown0, 2025-10-24:
You would know better than I do what the possibilities are for detection and repair, and what the ambiguous cases are.  I would think of confirming that each edge is visited twice, once in each direction.  That should allow detecting flipped faces, and if you can get them all flipped the same way then if the polyhedron volume is negative you need to flip them all.  But you have to watch out for Klein bottles.

jordanbrown0, 2025-10-24:
Thrown-together purple is helpful, yes, but isn't always easy to see, and may not even be visible.

As for negative space, I'd need to understand better what you're trying to do before offering an opinion.  But I think an inside-out polyhedron isn't naturally antimatter, if that's what you're looking for.

I was curious whether I could come up with a reasonably efficient test in OpenSCAD language, and just for fun threw one together.

```
module polycheck(faces) {
    o = object([
        for (f = faces)
            let(n=len(f))
                for (i = [0:n-1])
                    let(j = (i+1)%n)
                        [ str(f[i], " ", f[j]), true ]
    ]);

    nedges = len([ for (f=faces) each f ]);
    assert(nedges == len(o));

    for (f = faces) let(n=len(f)) for (i = [0:n-1]) let(j = (i+1)%n)
        assert(o[str(f[j], " ", f[i])]);
}
```

I think that catches a lot of cases.

jordanbrown0, 2026-05-07:
See also #1560, but actual detection seems better.