openscad/openscad #5394: Previewing complains about missing shader files

dankamongmen, 2024-10-27:
**Describe the bug**
I press F5 to preview; this issue seems independent of what source I'm using. I get the following ERRORs in my console:

```
Saved backup file: /home/dank/.local/share/OpenSCAD/backups/unsaved-backup-uNircUCo.scad
Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
Geometries in cache: 1
Geometry cache size in bytes: 848
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized tree has 1 elements!
ERROR: Cannot open shader source file: 'Preview.vert'   <----
ERROR: Cannot open shader source file: 'Preview.frag'   <----
ERROR: Cannot open shader source file: 'Preview.vert'   <----
ERROR: Cannot open shader source file: 'Preview.frag'   <----
Compile and preview finished.
Total rendering time: 0:00:00.035
```

**To Reproduce**
Steps to reproduce the behavior:
1. Install OpenSCAD from current sources.
2. Preview a valid SCAD file.

**Expected behavior**
I don't expect to see these errors. I'm presumably also not seeing the intended effects.

**Code reproducing the issue**
Any valid SCAD will reproduce.

**Screenshots**
I think the provided output probably suffices for now.

**Environment and Version info (please complete the following information):**
 - OS: Debian Unstable
 - System: AMD64 + 2080Ti
 - OpenSCAD Version 2024.10.24 with Manifold enabled, installed from source

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.
```
[schwarzgerat](0) $ openscad --info
Could not initialize localization.
QObject::startTimer: Timers can only be used with threads started with QThread
OpenSCAD Version: 2024.10.24
System information: Linux 6.11.4nlb2 #116 SMP Thu Oct 17 19:23:44 EDT 2024 x86_64 Debian GNU/Linux trixie/sid 64 CPUs 251.57 GB RAM
User Agent: OpenSCAD/2024.10.24 (Linux x86_64; Debian GNU/Linux trixie/sid)
Compiler: GCC "14.2.0" 64bit
MinGW build: No
Debug build: No
Boost version: 1_83
Eigen version: 3.4.0
CGAL version, kernels: 6.0.1, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Qt version: 5.15.13
QScintilla version: 2.14.1
InputDrivers: 
GLib version: 2.82.2
lodepng version: 20230410
libzip version: 1.11.1
fontconfig version: 2.15.0
freetype version: 2.13.3
harfbuzz version: 10.0.1
cairo version: 1.18.2
lib3mf version: 1.8.1
Application Path: /home/dank
Documents Path: /home/dank/.local/share
User Documents Path: /home/dank
[... 170 lines cut ...]
  /usr/share/fonts/X11/encodings/large
  /usr/share/fonts/opentype/roboto/slab
  /usr/share/fonts/truetype/roboto/hinted
  /usr/share/fonts/truetype/roboto/slab
  /usr/share/fonts/truetype/roboto/unhinted
  /usr/share/fonts/truetype/roboto/unhinted/RobotoTTF

GL context creator: GLX (new)
GLX version: 1.4
PNG generator: lodepng

GLEW version: 2.2.0
OpenGL Version: 4.6.0 NVIDIA 545.23.06
GL Renderer: NVIDIA GeForce RTX 2080 Ti/PCIe/SSE2
GL Vendor: NVIDIA Corporation
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

[schwarzgerat](0) $ 
```

**Additional context**
I can look into this myself, but wanted to get the bug filed. Thanks for the great software!

dankamongmen, 2024-10-27:
I do have the specified shaders in `/usr/local/share/openscad/shaders`:
```
[schwarzgerat](0) $ sha256sum /usr/local/share/openscad/shaders/*
f01fb91fb137af9c5505fcc6bf5806d077eabea6b139fcc00bd5a96ada942b9b  /usr/local/share/openscad/shaders/MouseSelector.frag
15ee3f2e6e96ea77a1f966fb3894ca29440a192f7645fb95a462ca754cda587c  /usr/local/share/openscad/shaders/MouseSelector.vert
6fa61f09cd86c2cbdbb8e4f836b185901ff96b7a655b1846577eb4c0b20df9a6  /usr/local/share/openscad/shaders/Preview.frag
5d6d57697788a20ce57f7f6b3de732eef30a99eebec63f10fb8ff28dfb82171f  /usr/local/share/openscad/shaders/Preview.vert
[schwarzgerat](0) $ 
```

dankamongmen, 2024-10-27:
The error is coming from here:
```cpp
std::string Renderer::loadShaderSource(const std::string& name) {                                                                   
  std::string shaderPath = (PlatformUtils::resourcePath("shaders") / name).string();                                                
  std::ostringstream buffer;                                                                                                        
  std::ifstream f(shaderPath);                                                                                                      
  if (f.is_open()) {                                                                                                                
    buffer << f.rdbuf();                                                                                                            
  } else {                                                                                                                          
    LOG(message_group::UI_Error, "Cannot open shader source file: '%1$s'", shaderPath);                                             
  }                                                                                                                                 
  return buffer.str();                                                                                                              
}                                                                                                                                   
```

dankamongmen, 2024-10-27:
it looks like i'm not getting a `resourcePath("shaders")`

dankamongmen, 2024-10-27:
yeah, it looks like that subsystem might be broken. i'm also seeing this on cerr:

```
Error reading examples.json: examples.json: cannot open file
Error reading examples.json: examples.json: cannot open file
WARNING: GLView: unknown colorscheme DeepOcean
ERROR: Cannot open shader source file: 'MouseSelector.vert'
```

dankamongmen, 2024-10-27:
yeah i suspect this is not going to generally fly (from `src/platform/PlatformUtils.cc`):

```cpp
#else                                                                                                                               
  const char *searchpath[] = {                                                                                                      
    RESOURCE_FOLDER("../share/openscad"),                                                                                           
    RESOURCE_FOLDER("../../share/openscad"),                                                                                        
    "usr/share/openscad", // AppImage, no build suffix                                                                              
    ".",                                                                                                                            
    "..",                                                                                                                           
    "../..",                                                                                                                        
    nullptr                                                                                                                         
  };                                                                                                                                
#endif
```

dankamongmen, 2024-10-27:
let me see if i can't pull the install root out of CMake and add it to this list. i don't like the idea of e.g. hardcoding `/usr/local/share/openscad'.

dankamongmen, 2024-10-27:
`resourceBasePath()` is returning my home directory, which certainly isn't going to help anyone.

also interesting is

```cpp
#ifdef OPENSCAD_SUFFIX                                                                                                              
#define RESOURCE_FOLDER(path) path OPENSCAD_SUFFIX 
#else                                                                                                                               
#define RESOURCE_FOLDER(path) path                                                                                                  
#endif  
```

which is definitely not what i would expect (i'd think it would be a prefix, first off, but maybe this is doing something i don't quite understand).

dankamongmen, 2024-10-27:
verified that if i hack `lookupResourcesPath()` to return `/usr/local/share/openscad` the errors go away, my preview is white instead of OpenScad Yellow™, and standard out/err are clear. not a fix obviously but demonstrates we're on the right track.

t-paul, 2024-10-27:
Same issue as #5392. Where is the application binary located? In `/usr/local/bin`?

dankamongmen, 2024-10-27:
aye

```
[schwarzgerat](0) $ command -v openscad
/usr/local/bin/openscad
[schwarzgerat](0) $ 
```

dankamongmen, 2024-10-27:
I'm not sure this is the same as #5392; this install is *not* an AppImage.

t-paul, 2024-10-27:
Not identical problem, but same root cause: Lookup of resource folder fails.
See comments there... https://github.com/openscad/openscad/pull/5368#issuecomment-2439722255

Here the issue seems to be that
https://github.com/openscad/openscad/blob/master/src/openscad.cc#L665
is not doing path lookup so it fails when the application is called without an absolute path.

t-paul, 2024-10-27:
Some digging pointed me to (boost >= 1.61):
```
#include <boost/dll.hpp>
boost::dll::program_location().parent_path();
```
Promising at first but not working if the current folder has a file with the same name:
```
boost::filesystem::system_complete(argv[0]);
```

All from https://stackoverflow.com/questions/5694190/boost-filesystem-how-to-find-out-in-which-directory-your-executable-is which notably has the totally wrong answer selected as solution :disappointed:.

t-paul, 2024-10-27:
Ok, lets see how #5395 works across platforms.

dankamongmen, 2024-10-27:
what if we just exported `CMAKE_INSTALL_PREFIX` https://cmake.org/cmake/help/latest/variable/CMAKE_INSTALL_PREFIX.html and used that in `lookupResourcesPath()`? that works even if someone e.g. moves the binary after installing it.

t-paul, 2024-10-27:
`CMAKE_INSTALL_PREFIX` is only correct in a very small amount of cases, e.g. it does not work for AppImages. The goal is making the application independent from compile time constants by expecting the resource files relative to the actually running binary. Being able to move the binary while leaving the resource files in the old place does not seem a use case to target.

The extra lookup cases in PlatformUtils are used for dev setup so it's not needed to run `make install` when doing development.

closed 2024-10-28 by t-paul

dankamongmen, 2024-10-28:
i can confirm this is fixed for the case where `openscad` is installed to and run from `/usr/local/bin`. i can likewise confirm, as predicted, that the fix does not hold for `openscad` installed to `/usr/local/bin` but run from some other root:

```sh
[schwarzgerat](0) $ cp /usr/local/bin/openscad /tmp/
[schwarzgerat](0) $ /tmp/openscad 
Could not initialize localization (application path is '/tmp').
Error reading examples.json: examples.json: cannot open file
```

that's certainly a less usual case, though. thanks for the fix!

dankamongmen, 2024-10-29:
> I would not mind having `CMAKE_INSTALL_PREFIX`in addition to the current logic. Do you think that's a case that happens enough so implement it?

I can only state with authority that it affects 100% of my own builds =].

It wouldn't harm anything that I can think of, and it is the standard idiom for doing this kind of thing IMHO. If you're just annoyed at the idea of having to further interoperate with CMake (totally understandable), I'm happy to prepare the patch for your review.

t-paul, 2024-10-29:
A pull request would be nice. I don't see much spare time in my near future but a hopefully small patch should be ok to review and integrate. I think we need to watch out to not change behavior on Windows and MacOS as I believe only Linux currently has `make install` support in our `cmake` setup.

dankamongmen, 2024-10-29:
.nodnod, the functionality is already split up for the three os's in `LookupBasePath()` iirc so that ought be trivial

commit 7f05c629 (2024-10-28): Merge pull request #5395 from openscad/fix-resource-folder-lookup

Fix resource folder lookup (fixes #5394).