openscad/openscad #5396: Extract and export exactly that portion of OpenSCAD script code used when rendering current preview.

KurtPfeifle, 2024-10-28:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated while writing OpenSCAD scripts. I'm a beginner and still struggling to follow a systematic approach. My code often is convoluted. I use #, *, % and ! a lot for debugging and for previewing only parts of complex structures.

When these parts are rendering the preview correctly (i.e. just what I want to see right now), they use parts and snippets of my code which are distributed all over a chaotic script.

I'm missing a nice way to display which actual lines of code my current preview is displaying.

**Describe the solution you'd like**
I want to be able to "export" as a separate file exactly this part of my total .scad code which is used in the current preview, skipping all the un-used parts.

**Describe alternatives you've considered**
I make extensive use of #, %, * and ! in my code.

**Additional context**
I cannot give additional context. I hope my description is good enough to be understood by everybody who as ever code in OpenSCAD. Else, ask.

jordanbrown0, 2024-10-28:
So you would want to omit the components that are behind other things and so not visible?

The existing previewer is incapable of doing that, because it doesn't know what's in front and what's behind.  It generates all of the triangles for the entire model and throws them at the graphics processor to figure out what's in front.

Right-clicking on the model might help; it shows the stack trace that led to the creation of that particular point.  (But that won't tell you the chain of calculations that fed into that result.)

KurtPfeifle, 2024-10-29:
I only know that adding a '*' in front of a line that calls a rendered object makes it not render it in the preview. So the code extraction should skip all definitions for that object.

Also, sometimes many parameters are defined at the beginning of the file (as are modules used inside an object). The code extraction should only include those parameter and module definitions which are used to render the current preview.

Similarly, a '!' can make the object on a line be the only one rendered. Code extraction should only incorporate those parameter and module definitions which contribute to rendering the !-ed object.

jordanbrown0, 2024-10-30:
Unfortunately, that doesn't seem likely.

You can sort of get the `*` behavior by running the program with F5 or F6 and then using Design / Display AST.  (But you will lose all comments, "include" file structure, and "use" references.)

You can sort of get all three using Design / Display CSG Tree.  (But you lose all variables, all calculations, all modules, et cetera, retaining only the shapes and operations on them - you would even lose *which * transformation was used, since they are all turned into multmatrix(), )

Doing anything better than those would involve processing that the execution engine doesn't do and can't easily do.

One problem is that by the time that it could do *any* of what you're talking about, it's already forgotten everything about comments and formatting.  What it's got left is what you see in Display AST.

It could fairly easily tell which modules went unused in any particular execution, marking them as they are executed.  (But would you want to prune away modules that are called on the failing sides of `if`s?)  Tracking which modules *could* have been used is harder; it would require tracking the call tree between the modules, finding uncalled modules and pruning them, then finding the the modules that are *then* uncalled and pruning them, et cetera.  But the only way it would have to strip them out is to emit the stuff that's *not* them, and again by that time it's lost comments and some of the file structure.

Stripping out unused variables would be similar - it would have to run the program or track the dependency tree, find variables that went unused, prune them, then find variables that are *then* unused, prune them, et cetera.  Again, the only way it would be able to prune them is by emitting what it's got left, having lost comments and file structure.

Limiting to the `!` object hurts my head, because the ! object can be buried inside other operations.  It'd need to do something similar to the processes above, but then in addition replace all parent operations with some no-op operation.  (One might think that it could emit the variables that went into the "!" object as a single flat list, but they could have been constructed from multiple nested scopes with overlapping names.)

It could maybe retain comments in the AST, which would make the result less unappealing.

Net, it would be a lot of work and the result would likely not be very satisfying.

jordanbrown0, 2025-03-08:
Strongly related to #4254.

jordanbrown0, 2026-05-07:
Let's call this a duplicate of #4254.

closed 2026-05-07 by jordanbrown0