openscad/openscad #5397: Intersecting Circles become Unioned Circles if for loop is used

allTexas, 2024-10-29:
**Describe the bug**
It appears that intersection() is working properly on two overlapping circles as long as the circle positions are manually generated.  Switching to a for loop for their translation causes the intersection() command to perform a union() instead. 

**To Reproduce**
Steps to reproduce the behavior:
1. Create intersection of two overlapping circles
2. Now switch to using a for loop to generate the circles
3. Observe union is performed instead

**Expected behavior**
Intersection of two circles should result in only the overlap region

**Code reproducing the issue**
```
module fishHead() {
    rad= 20;
    offset= 6;

    intersection() 
    {
        translate([0,offset,0]) 
        circle(r = rad);
        translate([0,-offset,0]) 
        circle(r = rad);
    }  
}

module fishHead_forLoop() {
    rad= 20;
    offset= 6;

    intersection() 
    {
        for(i=[-offset,offset]){
            translate([0,i,0]) 
            circle(r = rad);
        }  
    }  
  
}

fishHead();

translate(v = [50,0,0]) 
fishHead_forLoop();

```

**Screenshots**
[image]

**Environment and Version info (please complete the following information):**
 - OS: Debian 12
 - System: AMD 7900x
 - OpenSCAD  version 2024.10.28.ai21036 (git 7f05c6294) appImage downloaded from release page
 - Also confirmed that it happens in OpenSCAD version 2024.06.04.ai19515 (git 6db5adc0c) appImage downloaded from releases page.  Identical result :  manually generated circles' intersection appears correct, and for loop generated circles` intersection is really a union. 

**Library & Graphics card information**
Library version nfo trimmed for brevity: 
OpenSCAD Version: 2024.10.28.ai21036 (git 7f05c6294)
System information: Linux 6.1.0-25-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.106-3 (2024-08-26) x86_64 Debian GNU/Linux 12 (bookworm) 24 CPUs 30.49 GB RAM
User Agent: OpenSCAD/2024.10.28.ai21036 (git 7f05c6294) (Linux x86_64; Debian GNU/Linux 12 (bookworm))
Compiler: GCC "9.4.0" 64bit
MinGW build: No
Debug build: No
Boost version: 1_71
Eigen version: 3.3.7
CGAL version, kernels: 5.5.1, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Qt version: 5.15.4
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.64.6
lodepng version: 20230410
libzip version: 1.5.1
fontconfig version: 2.13.1 (runtime: 2.14.1)
freetype version: 2.10.1 (runtime: 2.12.1)
harfbuzz version: 2.6.4 (runtime: 6.0.0)
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: roof, input-driver-dbus, lazy-union, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output
Application Path: /tmp/.mount_OpenSCRTuU37/usr/bin

GLEW version: 2.1.0
OpenGL Version: 4.6 (Compatibility Profile) Mesa 22.3.6
GL Renderer: AMD Radeon Graphics (gfx1036, LLVM 15.0.6, DRM 3.49, 6.1.0-25-amd64)
GL Vendor: AMD
RGBA(8880), depth(0), stencil(0)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Qt graphics widget: QOpenGLWidget
QSurfaceFormat: RGBA(8880), depth(0), stencil(0)

kintel, 2024-10-29:
`for` implicitly unions all objects. To use intersection, there is `intersection_for()`: https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Conditional_and_Iterator_Functions#Intersection_For_Loop

closed 2024-10-29 by kintel

allTexas, 2024-10-29:
Interesting, learned something new today.   Appreciate the helpful response!

allTexas, 2024-11-15:
Thank you!  That does make sense.  I  guess I had assumed in my head that a 'for' loop would be unrolled before it's contents executed, I didn't realize it was executed as an operation itself.  I learned something and am better equipped to build with this amazing tool.  Appreciate it!