openscad/openscad #5400: Various errors with the model

hsuonsivu, 2024-10-31:
**Describe the bug**

Then compiling the model (copypasted after the log), openscad display is corrupted, and when trying to compile it with F6, errors below appear, and the model is not rendered. Openscad version is what macos brew installs (apparently not the latest).

OpenSCAD 2023.03.17 
https://www.openscad.org/
 
Copyright (C) 2009-2022 The OpenSCAD Developers
This program is free software; you can redistribute it and/or modify it under the terms of the GNU General Public License as published by the Free Software Foundation; either version 2 of the License, or (at your option) any later version.
 
Loaded design '/Users/hsu/repo/hsuonsivu/3d-models/showertray.scad'.
Compiling design (CSG Tree generation)...
Compiling design (CSG Products generation)...
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[fast-csg] Performing safer but slower nef operation instead of corefinement because operands share some vertices. (can override with fast-csg-trust-corefinement)
[... cut ...]
  /System/Library/Fonts
  /Library/Fonts
  /Users/hsu/Library/Fonts
  /System/Library/Fonts/Supplemental
  /Library/Fonts/Microsoft

GLEW version: 2.2.0
OpenGL Version: 2.1 Metal - 88.1
GL Renderer: Apple M1 Max
GL Vendor: Apple
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
GL context creator: Cocoa / CGL
PNG generator: Core Foundation
OS info: Mac OS X Version 14.6.1 (Build 23G93) (arm64 kernel)
Machine: 64-bit

**Additional context**

hsuonsivu, 2024-10-31:
I installed 2024.10.28 (git 7f05c6294) (brew openscad@snapshot). It no more gives errors when rendering with F6, but display is still funky.

[image]

I made some changes to code (makes no difference on display):

// Copyright 2024 Heikki Suonsivu
// Licensed under Creative Commons CC-BY-NC-SA, see https://creativecommons.org/licenses/by-nc-sa/4.0/
// For commercial licensing, please contact directly, hsu-3d@suonsivu.net, +358 40 551 9679

//include <hsu.scad>

print=0;
pipes=1;

fnsmall=print?30:12;
fnlarge=print?90:30;

bardiameter=18;
distancefromwall=52+4.45-bardiameter;

piped=14;
pipeseparation=126+piped;
pipedepth=22-piped/2;

versiontext="Showertray V1.2";
textdepth=0.8;
textsize=7;

cornerd=10;
width=220;
height=20;
backcornerd=distancefromwall*2;//height*2;
topextension=5;
bardepth=distancefromwall-bardiameter/2;
barnarrowing=0.3;
depth=distancefromwall-bardiameter/2+height*2+20;
wall=2.0;

holed=12;
holesx=3;
holesy=5;
holexstart=pipedepth+piped; //distancefromwall; // backcornerd/2;
holeystart=bardiameter/2+cornerd/2;
holexend=depth-cornerd/2;
holeyend=width/2-cornerd/2;
holexstep=(holexend-holexstart)/holesx;
holeystep=(holeyend-holeystart)/holesy;

module curvedcylinder(h,diameter,curved,a,stretch) {
  rotate_extrude(convexity=10,angle=a) translate([curved,0,0]) {
    hull() {
      circle(d=diameter);
      translate([stretch,0,0]) circle(d=diameter);
    }
  }
}

module trayform(w) {
  difference() {
    intersection() {
      minkowski() {
	sphere(d=cornerd);
	difference() {
	  union() {
	    translate([cornerd/2+backcornerd/2+w,w-cornerd/2,cornerd/2+w]) cube([depth-backcornerd/2-cornerd/2-cornerd/2-w*2,width/2-w*2,height-cornerd/2-w+topextension]);
	    intersection() {
	      hull() {
		translate([backcornerd/2,0,height]) rotate([-90,0,0]) cylinder(d=backcornerd-cornerd-w*2,h=width/2-cornerd/2-w,$fn=fnlarge);
		if (backcornerd/2+topextension > height) {
		  translate([backcornerd/2,0,backcornerd/2+topextension]) rotate([-90,0,0]) cylinder(d=backcornerd-cornerd-w*2,h=width/2-cornerd/2-w);
		}
	      }
	      translate([cornerd/2,0,cornerd/2+w]) cube([backcornerd/2+cornerd/2,width/2-cornerd/2,height+cornerd/2+topextension]);
	    }
	  }

	  union() {
	    // Cut for bar
	    translate([distancefromwall+bardiameter/2,0,cornerd/2-0.01]) cylinder(h=height+cornerd/2+topextension+0.02,d=bardiameter+cornerd+w*2,$fn=fnlarge);
	    translate([distancefromwall+bardiameter/2,-cornerd/2,cornerd/2-0.01]) cube([depth-distancefromwall-bardiameter/2,bardiameter/2+cornerd+w-barnarrowing,height+topextension+0.02]);

	    if (pipes) {
	      // Cuts for pipes on the wall
	      translate([pipedepth,pipeseparation/2,height-0.01]) cylinder(h=cornerd+topextension+0.02+w,d=piped+cornerd+w*2,$fn=fnlarge);
	      intersection() {
		translate([pipedepth+distancefromwall,pipeseparation/2,height]) rotate([-90,0,180]) curvedcylinder(height+cornerd/2+topextension+0.02,piped+cornerd+w*2,distancefromwall,90,piped);
		translate([pipedepth-piped/2-w,pipeseparation/2-piped/2-cornerd/2-w,0]) cube([distancefromwall,piped+cornerd+w*2,height+cornerd/2-w]);
	      }
	      translate([w-0.01,pipeseparation/2-piped/2-cornerd/2-w,w-0.01]) cube([pipedepth,piped+cornerd+w*2,height+topextension+cornerd+1+0.02]);
	    }
	  }
	}
      }

      // Cut off outside extras
      if (w==0) {
	cube([depth,width/2,height+topextension]);
      }
    }

    // Make holes for draining and lightness
    if (w==0) {
      if ((holexstep>holed) && (holeystep>holed)) {
	for (x=[holexstart:holexstep:holexend-1]) {
	  for (y=[holeystart:holeystep:holeyend-1]) {
	    translate([x+holexstep/2,y+holeystep/2,-0.1]) cylinder(h=height+topextension+1,d=holed);
	  }
	}
      }

      translate([distancefromwall-wall-1,-0.01,height-0.01]) cube([bardiameter/2+wall+1,bardiameter/2+0.02,topextension+0.02]);
    }
  }
}

module oneside() {
  difference() {
    trayform(0);
    trayform(wall);
  }
}

module showertray() {
  difference() {
    union() {
      oneside();
      mirror([0,1,0]) oneside();
      translate([pipedepth,0,wall-0.01]) rotate([0,0,90]) linear_extrude(height=textdepth+1) text(versiontext, size=textsize, valign="center",halign="center",font="Liberation Sans:style=Bold");

      difference() {
	translate([0.1,0,depth-height-(pipeseparation-piped-cornerd)*0.5]) rotate([0,90,0]) scale([0.5,1,1]) cylinder(h=wall,d=pipeseparation-piped-cornerd);
	translate([-0.1,-pipeseparation/2,-pipeseparation/2-0.1]) cube([wall+0.3,pipeseparation,height+pipeseparation/2]);
      }
    }
  }
}

intersection() {
  showertray();
  //      cube([100,width/2,100]);
}

hsuonsivu, 2024-10-31:
PS. Is there a way to attach scad files as files, copy-pasting seems to confuse github?

t-paul, 2024-10-31:
You can either enclose in triple back quotes ` ``` ` (I updated your post) or name the file `*.scad.txt` as github only allows a fixed list of extensions.

t-paul, 2024-10-31:
The display issue is a convexity problem, add `convexity=2` to the `minkowski()` call.

jordanbrown0, 2024-10-31:
Also, general advice:  when asking for help with a problem or reporting a possible bug, try to reduce the example to the minimum that will demonstrate the problem.  It doesn't have to look anything like your original model; the point is to make it so that the reader can spend less time trying to understand your model and more time trying to understand the problem.

As a side benefit, the process of chopping the example down to the minimum might well show that the problem is in your model.

hsuonsivu, 2024-11-02:
minkowski(convexity=2) fixed this, thanks.