openscad/openscad #5405: No preview/render without saving.

UBaer21, 2024-11-02:
**Describe the bug**
unsaved file fail on AST generation

**To Reproduce**
new↦ render/preview
```
Parsing design (AST generation)...
ERROR: Parser error: file access denied
ERROR: Compilation failed! (no top level object found)
```
 

**Expected behavior**
no error

**Code reproducing the issue**
none  

**Screenshots**
[image]

**Environment and Version info (please complete the following information):**
 - OS:  Windows 10
 - OpenSCAD Version 2024.10.31

thehans, 2024-11-03:
Can confirm on linux mint 22 also.

jordanbrown0, 2024-11-03:
Not on 2024.10.28 on Windows 11.

thehans, 2024-11-03:
Right, this is due to changes from #5399 which was merged on Oct 31, and involved migrating from `boost::filesystem` to `std::filesystem`.  
Looking into a fix now.

thehans, 2024-11-03:
Demo of differences for reference https://godbolt.org/z/Tz7bWPvv7

PR #5407 (2024-11-03): Fix difference between boost and std library treatment of empty path in filesystem::absolute
Fixes #5405
closed 2024-11-03 by kintel