openscad/openscad #5406: Undocking windows doesn't "stick"; they redock on restart

jordanbrown0, 2024-11-03:
**Describe the bug**
When I undock windows and restart, they're back in their original positions.

**To Reproduce**
Steps to reproduce the behavior:
* Delete the OpenSCAD registry subtree to return to install defaults.
* Start OpenSCAD.
* Edit / Preferences / Enable undocking of Editor and Console to separate windows
* Drag Editor, Console, Error log, Animation, Font List to separate locations.
* Quit
* Start OpenSCAD
* Observe that they have all moved back to their original positions.

**Expected behavior**
Expected them to stay in new positions.

**Environment and Version info (please complete the following information):**
 - OS: Windows 11
 - System: Intel PC 64-bit
 - OpenSCAD Version 2024.10.28 from download page

**Additional context**
I haven't checked whether OpenSCAD installation populates the Registry.  Perhaps it does, and running OpenSCAD without Registry data repopulates it wrong.

I restored my original (ancient-roots) OpenSCAD registry subtree.  I redocked Editor and Console back into their original locations, and quit, and restarted, and they went back to their undocked locations.  This suggests that undocking and redocking aren't saving state.

jordanbrown0, 2024-11-04:
Tidbit:  This command may be helpful in exercising this issue.
```
C:\Users\Jordan>reg copy HKCU\Software\OpenSCAD.save HKCU\Software\OpenSCAD /s
The operation completed successfully.
```

kintel, 2024-12-06:
To add to your observations: On macOS, the floating window doesn't redock on restart, it gets hidden.

jordanbrown0, 2026-05-07:
This particular problem appears to have been fixed, but there are still problems with stickiness of window rearrangements.

I'm going to close this one out; current problems can be covered by #6817.