openscad/openscad #5410: handling of globals in local scopes

yahbluez, 2024-11-04:
foo = 0;
bar = 1;
//(1) 
module foobar() {echo(foo,bar);} 
foobar(bar=2, foo=3);

This will produce an error because bar and foo are not declared as variables for foobar.

This will work:

//(2) 
module foobar(foo=foo, bar=bar) {echo(foo,bar);} 

But this forces to copy all used and maybe changed vars to listed in the definition in the name=name manner.

My idea is what would happen if the use of a global name while calling a module would just act as in example (2) and make a local copy with the new value?

I don't think that this may get in conflict with the openscad syntax in any way but would enhance the reuse of modules a lot.

Crazy or useful? 

best regards
yahbluez

closed 2024-11-04 by t-paul

yahbluez, 2024-11-04:
I know that, and my example (1) showed that i know that.
The point is, that sometimes an over wright is useful.
In this cases a declaration is needed, as shown in example (2).

So that leads to the idea to allow over wright of globals just by naming them while calling a module
and not each by each in the declaration.

Just an idea to make openscad more smooth.

t-paul, 2024-11-04:
> I know that, and my example (1) showed that i know that.

The example passes the parameters in addition which is a different scenario and therefor creates the warnings.

I'd advise against using `$` variables unless it's **really** clear how they work. I'm sure @UBaer21 knows it very well, but for people new to OpenSCAD it's better to start without. Specifically the example will not behave the same when imported into a different file when calling `foobar()` without parameters.

I'm still not seeing what the goal is. I think that if globals are defined, they should be used as they are. That's the whole point of defining them globally. If I change them in one place, I do they impact all the places. If you need to overwrite things, have a different parameter controlling the behavior (e.g. add an extra offset, scale by a specific factor) and more importantly *explain to someone reading the code what's going on*.

Do you have a specific use case showing how this is improving the code? For judging this kind of feature, some a little bit more explaining examples are needed than just the technical minimal case.

UBaer21, 2024-11-04:
@yahbluez When creating a module it is quite useful if you define every variable within the module, else as t-paul indicated it will become very hard to use, understand or debug. So you can use that module without any problems in other code / libraries. There is also the assert() function/module to ensure you don't end up with undef parameters.
While the $ special variables just don't cause a warning - you can work with warnings as these are not errors (don't activate "stop at first warning") - but you should maintain code so no warnings emerge.
Using $ within a module (without define them in the module) will make it even harder to debug.
Also take into consideration that documentation (e.g. Doxygen) will read what is defined for a module so editors using that will show you what parameters you can use. Like the help bubbles.