openscad/openscad #5415: Please add option to use Render instead of Preview

threed-factory-store, 2024-11-08:
**Is your feature request related to a problem? Please describe.**
My models are simple enough that there isn't a big difference in the time it takes to Preview vs Render.  However, Preview leaves many visual artifacts so I usually choose to Render, meaning a usable view takes a keypress and twice as much time as it needs to. 

**Describe the solution you'd like**
OpenScad would automatically Render my model without the Preview step.

**Describe alternatives you've considered**
None.

**Additional context**
None.

jordanbrown0, 2024-11-08:
Preview artifacts usually come in two forms:  Z-fighting and convexity issues.  Both are easy to avoid.

https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#What_are_those_strange_flickering_artifacts_in_the_preview?
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_are_some_parts_(e.g._holes)_of_the_model_not_rendered_correctly?

And less than those, the camera needs to be completely outside the model, including outside negative objects from difference() and objects that are intersected away.

https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_does_difference_(or_intersection)_sometimes_not_work_in_preview?

threed-factory-store, 2024-11-08:
This is great, thanks!

I already have a "fudge" value to get rid of the flickering artifacts. 

I'm using Solid Python and have wrapped my model in a render() call.

Thanks for your help!

closed 2024-11-08 by threed-factory-store

DerGolo, 2025-01-12:
I am new to OpenSCAD and would love to always automatically render when changing a custom variable. I mean using a render instead of a preview. Rendering on my system takes just a few milliseconds to adding an option to use full render instead of preview would be a great thing.