openscad/openscad #5419: Unable to access files containing non-English letters in the path

coder2999, 2024-11-10:
OpenSCAD is unable to access files containing non-English characters in the path from version 2024.10.28.

Any version of OpenSCAD is unable to use Ctrl+Click to open "include" or "use" files if the paths of these files contain non-English characters.

hzeller, 2024-11-10:
What operating system are you on ?

thehans, 2024-11-11:
Please paste your "Library Info" (from the Help toolbar, or using `openscad --info` option from command line).

Could you also provide an example file name and exactly how are you opening it (e.g. from File -> Open File dialog, or command line, etc)

coder2999, 2024-11-11:
I tested OPENSCAD on Windows 10 and 11 (English), version 2021.01 and version 2024.11.11.

For example, OPENSCAD cannot open "c:\пицца.scad", but it can open "c:\pizza.scad".

thehans, 2024-11-11:
Duplicate of #113

closed 2024-11-11 by thehans