openscad/openscad #5421: macOS: Sign and notarize the latest release: OpenSCAD-2021.01

kintel, 2024-11-11:
The current latest release (OpenSCAD-2021.01) is not signed and notarized.
An attempt to sign and notarize the current binary was done as part of #880, but resulted in a binary which only worked on Intel.

The missing piece is: Why does the signed and notarized 2021.01 binary crash on Apple Silicon, and is there a way of fixing this?

ElhemEnohpi, 2024-11-22:
I don't know why the signed and notarized version crashes. I do know that if you just sign it, with an ad-hoc signature, you can at least get it to run then, after using the "Open Anyway" button in System Preferences / Security, instead of getting the weird "Finder does not have permission to open (null)" error.  The user can do that on their own, with:

`codesign --deep -f -s - /Applications/OpenSCAD.app`

But that means they have to somehow figure out that they must do that, maybe by coming here. Maybe doing that on the release version, or making a note about it, would at least be an improvement, until the notarization can be figured out?

kintel, 2024-11-23:
The official 2021.01 binaries have now been replace with ad-hoc signed binaries.

ElhemEnohpi, 2024-11-23:
Sounds good. But I just tried downloading it - from https://github.com/openscad/openscad/releases/download/openscad-2021.01/OpenSCAD-2021.01.dmg - and it still doesn't have a code signature. The modification date is still January 2021. Tried emptying my browser cache too.

kintel, 2024-11-23:
oh, we have binaries hosted on github too? I've updated those as well now.

ElhemEnohpi, 2024-11-24:
Ok, it works now, with "Open Anyway". I also tried to install it with Homebrew, but that now fails with a SHA256 mismatch. I guess the cask code has to be updated for that.

kintel, 2024-11-24:
Yeah, I guess homebrew would need to update the sha.

gingerbeardman, 2024-11-24:
I filed a PR: https://github.com/Homebrew/homebrew-cask/pull/193120

sciencejoan, 2024-11-24:
I tried opening the 2021.01 from the openscad.org/downloads page on a brand new M4 iMac and got the error "Apple could not verify “OpenSCAD-2021.01.app” is free of malware that may harm your Mac or compromise your privacy." The only options were "Done" or "Move to trash." On the bright side, the 2024.11.01 (git 5dd397d7e) opens perfectly fine on the M4 and is blindingly fast.

gingerbeardman, 2024-11-24:
> I tried opening the 2021.01 from the openscad.org/downloads page on a brand new M4 iMac and got the error "Apple could not verify “OpenSCAD-2021.01.app” is free of malware that may harm your Mac or compromise your privacy." The only options were "Done" or "Move to trash." 

As expected. You still have to right-click > open *twice* and then you'll see:

[image]

sciencejoan, 2024-11-24:
I don't get that message unless I open the package contents with a right click, and then go to the folder  Contents >MacOS > OpenSCAD and in turn click on that.   (No amount of right-clicking gives the message you show from the icon on the install disk image, but if I go to the security settings anyway it does give me the option there - without telling me about it. Argh.)  As a side note, we are authors of books that use OpenSCAD to teach math, so my purpose here is to document for our readers what the minimum path is....pretty much we're just telling them to use the 2024 build until this gets more straightforward.

gingerbeardman, 2024-11-24:
> I filed a PR: [Homebrew/homebrew-cask#193120](https://github.com/Homebrew/homebrew-cask/pull/193120)

this has been approved :)

nophead, 2024-11-24:
Why on earth does anybody use Macs these days? I certainly never will.

curliq, 2024-12-06:
This fixed it for me:

Open system settings -> Privacy and Security -> Scroll all the way down -> there's a "Allow OpenSCAD to be opened" button

ElhemEnohpi, 2024-12-07:
@curliq yes, we know that, thanks. However, it only became possible to do that two weeks ago, when the binaries were updated with a partial fix (ad-hoc signature). Soon there should be a full fix (notarization), so that you won't even have to do that, and just answer "Ok" to the usual "this was downloaded from the Internet. Open it?". That's the point of this issue, to figure out the steps to get notarization to work.

copecog, 2025-01-26:
> Why on earth does anybody use Macs these days? I certainly never will.
> […](#)

I concur, I don't know why anyone would use anything besides OpenBSD!

haty1, 2025-03-01:
It is now March 2025 and I just installed OpenSCAD via Homebrew and am still getting this error message when I try and open the app. Seems it is still a problem

kintel, 2025-03-01:
It's still an issue for the 2021.01 release: Signing the existing binary someone causes it to crash on Intel. We could try to dig out an old Mac and rebuild it from source and see if that helps, but it's a lot of effort for uncertain gain.

This is fixed in development builds since December though, so the issue only affect the 2021.01 release.

pkernevez, 2025-06-15:
Do you plan to to release a signed and notarized version of the OpenSCAD ?
The idea is not to by pass the OS protection by changing the security settings.
I already signed application for Mx silicon processor and it works fine (and your are able to sign both Intel & Silicon from a silicon mac (and I suppose we should be able to do the inverse).

gingerbeardman, 2025-06-15:
The issue is that the "official" build is so old people don't know there are development nightlies.

kintel, 2025-06-16:
We're working on a new official release, but it will still take a while as our dev team is tiny. Hoping for end of 2025.

If anyone has experience in this area, help would be appreciated.

pkernevez, 2025-06-16:
> The issue is that the "official" build is so old people don't know there are development nightlies.

It was my case. I use the last nightly build and it works fine.
And to be transparent, I used OpenScad 4 years ago (the last time I needed to do 3D printing) and it worked fine. 
But when I saw that they wasn't any release since 4 years, no perfect support for Mac Silicon and no notarized binary, I start looking for an alternative.
Without this post I didn't found the last build.

You may add a link on the download page to the nightly build.

There isn't a nightly build stable enough to be promoted as a new release ?

I only worked on packaging a java app with notarization. I don't think I could help you.

jordanbrown0, 2025-06-16:
The downloads page has a section for development snapshots. It's even on the left sidebar.

https://openscad.org/downloads.html#snapshots

gingerbeardman, 2025-06-16:
I think it could good to signpost it from right next to the current/official download.

"newer versions are work in progress and available here <link>"

Development snapshots or nightlies are known by developers, but mean nothing to the average user.

jcheatham, 2025-07-04:
I agree with @gingerbeardman , even as a developer I tried the first option, then brew, and was inclined to give up thinking the project was dead... but I refused to give up hope and found this thread.  Development snapshots at the top would have helped me for sure; OR a blurb in the macOS section saying "We know the official version has signing issues, check out the latest snapshots instead" would have helped :)