openscad/openscad #5425: problem with path after openscad-2024.10.28-x86-64

fmarionpoll, 2024-11-12:
**Describe the bug**
since versions after 2024.10.28, I cannot work on files which using other files located on a vpn drive. It is fine if the files are on a local drive. When I launch a scad file on the vpn, I obtain the following error:
Compilation aborted by exception: filesystem error: cannot make canonical path: No such file or directory [/]

**To Reproduce**
Steps to reproduce the behavior:
1. create 2 openscad files:
file 1 called "dummy1.scad":
module create_cube()
{
    cube([10, 10, 10], center=true);
}
create_cube();

file2 called "dummy2.scad":
use <dummy1.scad>
create_cube();
---
2. save them in a directory called "dummy" on your local computer and on a vpn drive.

3. Behavior: when these 2 files are in a directory on my desktop (windows 11) - no error. If "dummy" is on a vpn drive, double clicking on file dummy2.scad fires an error.  However, opening "dummy1.scad" works, even if it is located on the vpn drive. The vpn drive has an address like "\\disk_drive_from_my_university.fr\MultiCAFE\docs-outils-3D\3D models\OpenSCAD\dummy"

**Bug found in all nightly versions after 2024.10.28**
Version 2024.10.28 works find. All others - including 11.11 are firing an error.

**Environment and Version info (please complete the following information):**
 - OS: Windows 11 pro, English version
 - System: Intel PC Dell
 - OpenSCAD (Nightly) Version after version 2024.10.28 (git 7f05c6294)

kintel, 2024-11-12:
@thehans could be another regression.

antfurn, 2025-02-16:
Oh, the _real_ error is that I had used an upper-case 'D' for diameter rather than lower-case 'd', in the cylinder() call.
Very miss-leading error msg (and I thought that either case was supported - I maybe misremembering though).

kintel, 2025-02-16:
That's very odd. Is this trivially reproducible?

antfurn, 2025-02-16:
Any trivial error in the file (that's saved on the network mount) produces the same miss leading error.
screen shot, unsaved file / local file / map network  --all correct error report.
                    network path file -- Very wrong error message [bottom right]

[image]

antfurn, 2025-02-16:
This was never a problem on nightly builds from early last year, I was running OpenSCAD-2024.01.26-x86-64.

jordanbrown0, 2025-03-18:
Easily reproduced.  It looks like the problem has nothing to do with VPNs; rather, it looks like it has to do with [UNC](https://en.wikipedia.org/wiki/Path_(computing)#Universal_Naming_Convention) paths.

* Create a directory.
* Share it (properties/share).
* Start OpenSCAD. (Demonstrated with 2025.03.14.)
* Type a file with an error (say, one line containing "x").
* File/Save.
  * Navigate to that directory via its network path (\\\\computer\share\dir).
  * Save the file there.
* F5.
* Observe error.

>OpenSCAD 2025.03.14 
>https://www.openscad.org/
> 
>Copyright (C) 2009-2024 The OpenSCAD Developers
>This program is free software; you can redistribute it and/or modify it under the terms of the GNU General Public License as >published by the Free Software Foundation; either version 2 of the License, or (at your option) any later version.
> 
>Saved design '//jordan/users/Jordan/Downloads/tmp/bad.scad'.
> 
>Parsing design (AST generation)...
>ERROR: Compilation aborted by exception: filesystem error: cannot make canonical path: No such file or directory [/jordan]

jordanbrown0, 2025-03-18:
[ I should note that my testing below is all MSYS2, not the production MXE. ]

The exception is from fs_uncomplete, where the arguments are
* /jordan/users/jordan/users/Jordan/Downloads/tmp/bad.scad
* an empty path
when it calls fs::relative(/jordan/users/jordan/users/Jordan/Downloads/tmp/bad.scad, .).

But that's a bad path; the original starts with *two* slashes.  It's no real surprise that something trying to process it gets into trouble.

It looks like the bad string comes from parser.y, where it calls lexer_set_parser_sourcefile().  fs::path::generic_string() turns a good path into a bad one.
```
src/core/parser.y 810 parse filepath=\\jordan\users\jordan\users\Jordan\Downloads\tmp\bad.scad
src/core/parser.y 814 parse parser_sourcefile=/jordan/users/jordan/users/Jordan/Downloads/tmp/bad.scad
```
(pay no attention to the line numbers; they reflect additional diagnostics added)

One step might be to get rid of that call to generic_string().  However, doing that doesn't fix the problem; when fs_uncomplete is called with \\jordan\users\jordan\users\Jordan\Downloads\tmp\bad.scad.

One possible problem:  if I'm understanding this correctly, it's expected to produce a path relative to "." that leads to the specified file.  But I am not sure that there *is* a relative path from "." to a UNC path; the top level of the UNC namespace does not appear in the top of the normal hierarchy.

It might be reasonable to catch the exception from fs::relative and just use the original path if it fails.

kintel, 2025-03-18:
Regarding building relative file paths: How does this work across drive letter boundaries on Windows?

I guess MSYS2 perhaps circumvents this by building a global virtual filesystem on top of the drive letter system, and other platforms all have aglobal filesystem namespace.

jordanbrown0, 2025-03-18:
That's a good question.  I don't think it can; I think it would have to yield an absolute path.

kintel, 2025-03-18:
> That's a good question. I don't think it can; I think it would have to yield an absolute path.

Yeah, makes sense. Do you know if that works as expected today? 

It kind of feels like `filesystem::relative()` should understand network share paths on Windows , but perhaps that's not part of the spec..

jordanbrown0, 2025-03-18:
I can't do as deep analysis on MXE builds, and there's no particular reason to believe that MSYS2 is the same, but...

With an MXE build, include paths with drive letters seem to lead to bad paths in error messages.

For two files on the same drive, including with a drive letter:
```
ERROR: Parser error: syntax error in file , line 
```
With the drive letter removed:
```
ERROR: Parser error: syntax error in file ../../../../users/jordan/downloads/tmp/bar.scad, line 2
```

anoblet, 2025-04-07:
I just ran in to this error as well.

jordanbrown0, 2026-08-27:
The problem is with constructing relative paths across non-UNIXy scenarios like crossing drives or UNC paths. It is not related to path length.

roel-v, 2026-08-31:
I ran into this issue too today, not on a UNC path, but on "Cloud File", a file on Dropbox. This issue apparently started happening in 2024 when switching from boost::filesystem to std::filesystem, where MinGW libstdc++ didn't properly support UNC paths (and presumably other "irregular" path types) yet. The problem is that the throwing version of canonical() is used in src/core/parsersettings.cc, without actually catching the exception when something goes wrong. So one solution is to use the non-throwing version, and when canonicalization fails, just use the passed in version as-is (as it's already gone through check_valid()/exists() already anyway), or use a small wrapper function that catches the exception and then does the same. Other solutions exist of course, like going back to boost's filesystem, use Qt filesystem wrappers etc. But then it becomes a case of 'the devil you know', as it's unknown what sort of issues we'd run into in similar corner cases.

I have a small test case that exercises this failure mode as well as a patch that fixes it, if someone agrees that this is the cause and the fix approach is directionally solid. I can make a PR.

jordanbrown0, 2026-09-02:
It's been a while since I looked at this, but yes, it's a problem with the "uncomplete" function that attempts to derive a relative path given an absolute path.  The problem is, I believe, that for some paths A and B, there is simply no relative path from parent(A) to B.

In UNIX-land, there is always a relative path from any point in the file system to any other point.  You might have to `..` all the way to the root and climb back down again, but you can do it.

In Windows, not so.  There is no parent namespace that includes all of the drives, that you can `..` up into.  If you want a relative path to `D:\FOO` from `C:\BAR`, sorry, doesn't exist.  Similarly for UNC paths, and quite possibly other types.

As you say, it should probably catch the exception and just use the absolute path.

roel-v, 2026-09-02:
The problem is that gcc's implementation of std::filesystem does not handle UNC paths. In that respect, the switch to std::filesystem from boost::filesystem was a regression. But I guess that ship has sailed. Also a build with msvc (i.e, using msvc's std::filesystem) works fine. 

Regardless, a patch is here: https://github.com/openscad/openscad/pull/7004 .

jordanbrown0, 2026-11-01:
There have been other problems with "uncomplete" discovered.

In addition to the cases where it misbehaves because there isn't a relative path, it can also yield silly paths in some cases.

In the case I observed, my MSYS2 home folder was set to /c/users/jordan, when the actual path is /c/Users/jordan.  Windows is case-insensitive, so they yield the same result, but it seems that fs::relative decides they are different and so in some cases would generate relative paths that climbed all the way up to /c and then back down again.

Another user, on a Linux system, observed a similar misbehavior when operating in a folder reached by a symbolic link.  I don't remember the details, but the effect was similar:  a path that climbed way up and then way down again.  I'm not sure that it actually reached the right place.

But all of this led to thinking about what the **right** answer is.  fs_uncomplete() is used primarily (or exclusively?) in error messages.  The intent, I believe, is to take paths that have been made absolute, and may be unwieldy, and simplify them.  The problem is in figuring out what a relative path should be relative **to**.  If you encounter an error in a library, say in `\Program Files\OpenSCAD\libraries\MCAD`, a relative path from `\Users\jordan\OneDrive\Documents\OpenSCAD\MyProject` is going to have an awful lot of `..` in it, and yield a path that's a lot less obvious than the absolute path.  Also, a number of the calls (most?  all?) ask for a path relative to the current folder, and in a GUI environment where the application is started from some kind of menu or icon shortcut the current folder is not necessarily obvious or useful.  (In Windows, it's the installation folder.)

I think that we need to make fs_uncomplete behave better when a relative path can't be found, but I think we also need to look more carefully at what the best way is to report file paths.  I think a lot of the right answer is dependent on how the user referred to the file.  For the main file in the GUI I think it's fine to just use the file's leaf name.  For the main file in the CLI, I think it should use the exact name that the user typed in the command.  For a library file found by include<> or use<> by following the library path, I think it should be an absolute path.  For a file found relative to the main file, I think a relative path from the main file's folder is appropriate.  But what if the problem reference is in a file other than the main file?  Suppose your main file says `include <somelib.scad>`, that gets found in `[Documents]/OpenSCAD/libraries`, and that file says `import("nonexist.svg")`?  It's looking for the file in the same folder as the library.  Relative to current directory?  Useless.  Relative to main file?  Probably useless.  Absolute?  Maybe.  Relative to the library?  Maybe, but then you need to ensure that the error tells you where **that** is.

Alexb715, 2026-11-06:
I'd like to take the `fs_uncomplete()` part of this, as a separate PR from #7004 (which covers the `fs::canonical()` calls in `find_valid_path()`).

The change I have in mind is the one suggested above: when `fs::relative()` fails or there is no relative path (another drive, a UNC path), `fs_uncomplete()` returns the original path instead of throwing or returning an empty path. It doesn't try to answer the bigger question of what paths should be relative to.

kalj, 2026-11-07:
I ran master, #7004, #7151, #7187 and all three combined through a Windows CI harness (MSYS2/UCRT64 build, real UNC paths via \\localhost\C$ and an SMB share, a mapped drive and a cross-drive file): https://github.com/kalj/openscad/actions/runs/42398458989

- #7187 fixes the empty file name in messages for mapped-drive / other-drive files (`in file , line 3` -> `in file Z:/error.scad, line 3`) and the crashes when the working directory is a UNC path.
- With all three PRs combined, command-line runs on a UNC file still abort as soon as evaluation starts (use, include, or a plain warning): `filesystem error: cannot set current path: No such file or directory`. That's `fs::current_path(fparent)` in do_export() (openscad.cc), which none of the PRs touch. Syntax-error files don't hit it because parsing fails first.
- Messages show the UNC prefix being dropped (`\\localhost\unctest\error.scad` printed as `localhost/unctest/error.scad`), and with a UNC working directory use<>/include<> look for `/localhost/unctest/dummy1.scad` and fail - so the path mangling Jordan mentioned is real and still unaddressed.

kalj, 2026-11-07:
Pinned down the root cause with a small std::filesystem probe built with the same MSYS2/UCRT64 toolchain (https://github.com/kalj/openscad/actions/runs/42411205117, fs-diagnostics job): MinGW's std::filesystem::path doesn't parse UNC root names. For `\\localhost\unctest\use.scad` it reports root_name='' and is_absolute()=false, so:
- fs::absolute() returns `C:\localhost\unctest\use.scad` (treated as a folder on the current drive)
- generic_string() collapses the prefix to `/localhost/unctest/use.scad`
- canonical(), relative() and current_path(parent) all fail with "No such file or directory"

exists() on the original string works; local and mapped-drive (Z:\) paths are fine at every step. This matches the regression from boost::filesystem to std::filesystem. The existing PRs prevent crashes, but for UNC files to actually work, paths starting with \\ or // need to bypass absolute()/canonical()/generic_string() (e.g. small UNC-aware helpers in fileutils used by do_export(), parser.y, find_valid_path() and fs_uncomplete()). Happy to work on that if this approach sounds right.

kalj, 2026-11-07:
With the three PRs plus #7194, nothing crashes in any tested scenario.

Please confirm if I should start working on UNC-aware path helpers.

kintel, 2026-11-13:
I'll leave it to @jordanbrown0 to comment on Windows-specific issues as he's currently best suited to evaluate that, but apart from that I'd say your explorations are valuable.

kalj, 2026-11-14:
I've opened #7222 as a draft proposal for the remaining `use<>`/`include<>` part: a few UNC-aware path helpers that work around MinGW's `std::filesystem` not recognizing `\\server\share` as a root. With it, use/include from UNC paths passes on a Windows runner, and nothing else changes.