openscad/openscad #5427: Bug openscad with iMac sequoia 15.1

Hannes689, 2024-11-13:
I had OpenScad on a Mac book Pro Monterey12.7.6 . After passing to a iMac sequoia 15.1 it was impossible to use OpenScad. I tried to download a new version  but it was impossible. Finder indicated that it does not work with this message : message : Finder App is not allowed to open. Help, please.

kintel, 2024-11-13:
Fixed by #880 - please use the development snapshots for now.

closed 2024-11-13 by kintel

Hannes689, 2024-11-13:
Thanks for the abundant information. It does not solve my problem. I
have used  OpenScad on Macbook laptop Monterey12.7.6 without any problems
until I transferred towards a new Imac Apple Sequoia 15.1. When I tried to
open Openscad it didn't work . I tried to reinstall Openscad unsuccessfully
with an error message from Finder (cf screen capture below).
I want to use OpenScad on my new iMac but I need simple information, do you
any simple solution please ? Thanks [image: Capture d’écran 2024-11-13 à
14.09.11.png]

thehans, 2024-11-13:
You can find development snapshots towards the bottom of the downloads page here: https://openscad.org/downloads.html#snapshots