openscad/openscad #5428: won't run in gui mode with termux-x11

nnolan-wbd, 2024-11-13:
**Describe the bug**
can't run in termux-x11
hitting openscad.cc line 990
"    LOG("Requested GUI mode but can't open display!\n");"

before blaming termux; consider:
fluxbox is working
firefox --no-xshm  works
aterm works
glxgears works
display is :0.0 

**To Reproduce**
install termux and termux-x11 on a docked android device.
go through termux-x11 setup instructions. fluxbox may save you some setup time over their xfce instructions.
ensure display is correct and exported
ensure x11 and mesa apps work  
pkg install openscad
run openscad

**Environment and Version info (please complete the following information):**
 - OS: [Android 10, termux & termux-x11]
 - System: SD865, adreno 650 android 10 device
 - OpenSCAD Version 2023.11.19 (provided by termux repo)
 
**Library & Graphics card information**
`
Could not initialize localization.

OpenSCAD Version: 2023.11.19
System information: Linux 4.19.81-perf #1 SMP PREEMPT Thu Sep 26 18:38:00 HKT 2024 aarch64 8 CPUs 5.55 GB RAM
User Agent: OpenSCAD/2023.11.19 (Linux aarch64)
Compiler: Clang "17.0.2 (https://android.googlesource.com/toolchain/llvm-project d9f89f4d16663d5012e5c09495f3b30ece3d2362)"
MinGW build: No
Debug build: No
Boost version: 1_83
Eigen version: 3.4.0
CGAL version, kernels: 5.6, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: <not enabled>
Qt version: Qt disabled - Commandline Test Version
GLib version: 2.78.1
lodepng version: 20180910
libzip version: 1.10.1
fontconfig version: 2.14.2 (runtime: 2.15.0)
freetype version: 2.13.2 (runtime: 2.13.3)
harfbuzz version: 7.3.0 (runtime: 10.1.0)
cairo version: 1.18.0 (runtime: 1.18.2)
lib3mf version: (not enabled)
Application Path: /data/data/com.termux/files/home
Documents Path: /data/data/com.termux/files/home/.local/share
User Documents Path: /data/data/com.termux/files/home
Resource Path: /data/data/com.termux/files/home
User Library Path: /data/data/com.termux/files/home/.local/share/OpenSCAD/libraries
User Config Path: 
Backup Path: /data/data/com.termux/files/home/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /data/data/com.termux/files/home/.local/share/OpenSCAD/libraries
  /data/data/com.termux/files/home

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  /system/fonts
  /data/data/com.termux/files/usr/share/fonts
  /data/data/com.termux/files/home/.local/share/fonts
  /usr/local/share/fonts
  /usr/share/fonts
  /data/data/com.termux/files/home/.fonts
  /data/data/com.termux/files/usr/share/fonts/100dpi
  /data/data/com.termux/files/usr/share/fonts/75dpi
  /data/data/com.termux/files/usr/share/fonts/TTF
  /data/data/com.termux/files/usr/share/fonts/cyrillic
  /data/data/com.termux/files/usr/share/fonts/encodings
  /data/data/com.termux/files/usr/share/fonts/misc
  /data/data/com.termux/files/usr/share/fonts/util
  /data/data/com.termux/files/usr/share/fonts/encodings/large

GL Renderer: NULLGL Glew
OS info: Unix
Machine: 64-bit

$ 
`

t-paul, 2024-11-13:
>  Qt version: Qt disabled - Commandline Test Version

When built without GUI, there is no GUI.

t-paul, 2024-11-13:
This seems to be intentional, so this needs to be raised @ termux if there are other options.

https://github.com/termux/termux-packages/blob/7725b09f46936e043baf98139996527ce0031f28/packages/openscad/build.sh#L2

closed 2024-11-13 by t-paul

jordanbrown0, 2024-11-13:
If it's built with no GUI and you ask for a GUI, it seems like the error message could be a lot clearer.

nnolan-wbd, 2024-11-15:
Okay, Thanks for pointing that out. I will raise this with Termux.
 if your team thinks it would be helpful, please also update the message as previously suggested.

t-paul, 2024-11-15:
None of the official builds use that, so it's not going to get a very high priority. Termux seems to build the 2021.01 release version, that will not be updated anymore but a PR for future releases would be welcome.