openscad/openscad #5429: Unable to invoke Wireframe view.

boring-pilot, 2024-11-14:
For unknown reasons, the wireframe view does not come up.  F11 and menu invocation does nothing, the surfaces view remains.
Running Openscad 2021.01windows on a Windows 11 Pro, v23h2, Dell xps17 9720.
Have not tried system resets, etc. yet.

boring-pilot, 2024-11-14:
Problem persisted after reset, but it appears to be limited to just one design file (so far).  It is a fairly simple file, so I donʻt know why the wireframe view fails.  File SwimNoodleEnd.scad text below...
//----------------------------------------------
//swim noodle end piece, v2

rout = 100/2;
rin = 65/2;
rdif =rout-rin;
w = 10;
thmin = 1.8;
scfx = .7;
scfy = 1.5;

res = 180;

rotate_extrude(angle = 360, convexity = 4, $fn=res)
translate([rin,0,0])
difference() {
union() {
    square([rdif,w]);
    translate([rdif,thmin,0]) circle(thmin,$fn=res/5);
    translate([0,w,0]) circle(thmin,$fn=res/5);
}
    translate([thmin+rdif*scfx,w,0]) scale([scfx,w/(scfy*rdif),0]) circle(rdif,$fn=res);
    translate([-thmin,w-thmin,0]) square([thmin,2*thmin]);
}
//-----------------------------------

kintel, 2024-11-14:
Closing as duplicate of #2031

closed 2024-11-14 by kintel

boring-pilot, 2024-11-14:
Read Issue #2031, and modifed the source to create the same object in a more-complex fashion, and now the Wire-frame view works!  This was non-obvious......

kintel, 2024-11-15:
Yeah, it needs to be fixed. It's a blocking bug for the next release.